cleanflight/cleanflight #1036: Understanding master mix table in Cleanflight

tedbyers, 2015-06-20:
I am having difficulty understanding the master mix table in cleanflight.

In MultiWii, it was dirt simple to add new models. Consider the following:

    #elif defined( OCTOFLATX )
    motor[0] = PIDMIX(+1 ,-1/2,+1); //MIDFRONT_L
    motor[1] = PIDMIX(-1/2,-1 ,+1); //FRONT_R
    motor[2] = PIDMIX(-1 ,+1/2,+1); //MIDREAR_R
    motor[3] = PIDMIX(+1/2,+1 ,+1); //REAR_L
    motor[4] = PIDMIX(+1/2,-1 ,-1); //FRONT_L
    motor[5] = PIDMIX(-1 ,-1/2,-1); //MIDFRONT_R
    motor[6] = PIDMIX(-1/2,+1 ,-1); //REAR_R
    motor[7] = PIDMIX(+1 ,+1/2,-1); //MIDREAR_L
    #elif defined( OCTOX8 )
    motor[0] = PIDMIX(-1,+1,-1); //REAR_R
    motor[1] = PIDMIX(-1,-1,+1); //FRONT_R
    motor[2] = PIDMIX(+1,+1,+1); //REAR_L
    motor[3] = PIDMIX(+1,-1,-1); //FRONT_L
    motor[4] = PIDMIX(-1,+1,+1); //UNDER_REAR_R
    motor[5] = PIDMIX(-1,-1,-1); //UNDER_FRONT_R
    motor[6] = PIDMIX(+1,+1,-1); //UNDER_REAR_L
    motor[7] = PIDMIX(+1,-1,+1); //UNDER_FRONT_L
    #elif defined( HEX6H )
    motor[0] = PIDMIX(-1,+1,-1); //REAR_R
    motor[1] = PIDMIX(-1,-1,+1); //FRONT_R
    motor[2] = PIDMIX(+1,+1,+1); //REAR_L
    motor[3] = PIDMIX(+1,-1,-1); //FRONT_L
    motor[4] = PIDMIX(0 ,0 ,0); //RIGHT
    motor[5] = PIDMIX(0 ,0 ,0); //LEFT
    #elif defined( OCTO_H )
    motor[4] = PIDMIX(+1 , -1 ,-1); //FRONT_L
    motor[0] = PIDMIX(+1 ,-1/2,+1); //MIDFRONT_L
    motor[7] = PIDMIX(+1 ,+1/2,-1); //MIDREAR_L
    motor[3] = PIDMIX(+1 ,+1 ,+1); //REAR_L
    motor[1] = PIDMIX(-1 ,-1 ,+1); //FRONT_R
    motor[5] = PIDMIX(-1 ,-1/2,-1); //MIDFRONT_R
    motor[2] = PIDMIX(-1 ,+1/2,+1); //MIDREAR_R
    motor[6] = PIDMIX(-1 ,+1 ,-1); //REAR_R
    #elif defined( VTAIL4 )
    motor[0] = PIDMIX(+0,+1, +1); //REAR_R
    motor[2] = PIDMIX(+0,+1, -1); //REAR_L
    motor[1] = PIDMIX(-1, -1, +0); //FRONT_R
    motor[3] = PIDMIX(+1, -1, -0); //FRONT_L
    #elif defined( QUADX )
    motor[0] = PIDMIX(-1,+1,-1); //REAR_R
    motor[2] = PIDMIX(+1,+1,+1); //REAR_L
    motor[1] = PIDMIX(-1,-1,+1); //FRONT_R
    motor[3] = PIDMIX(+1,-1,-1); //FRONT_L
    /*
    observe that VTAIL compared to QUADX, roll is set to 0 and yaw is reversed, which implies the following
    */
    #elif defined( VTAIL6H )
    motor[1] = PIDMIX(-1,-1,+1); //FRONT_R
    motor[3] = PIDMIX(+1,-1,-1); //FRONT_L
    motor[4] = PIDMIX(0 ,0 ,0); //RIGHT
    motor[5] = PIDMIX(0 ,0 ,0); //LEFT
    motor[0] = PIDMIX(0,+1,+1); //REAR_R
    motor[2] = PIDMIX(0,+1,-1); //REAR_L
    #elif defined( VTAIL8H )
    motor[4] = PIDMIX(+1 , -1 ,-1); //FRONT_L
    motor[0] = PIDMIX(+1 ,-1/2,+1); //MIDFRONT_L
    motor[7] = PIDMIX(+1 ,+1/2,-1); //MIDREAR_L
    motor[1] = PIDMIX(-1 ,-1 ,+1); //FRONT_R
    motor[5] = PIDMIX(-1 ,-1/2,-1); //MIDFRONT_R
    motor[2] = PIDMIX(-1 ,+1/2,+1); //MIDREAR_R
    motor[6] = PIDMIX(0 ,+1 ,+1); //REAR_R
    motor[3] = PIDMIX(0 ,+1 ,-1); //REAR_L

If you're familiar with the MultiWii code, you'll recognize that the first few mixes are extracted from the MultiWii master mix table. The Octocopter H-frame, and the V-tail hex and V-tail octocopter mixes are new (so new they're untested, and testing will have to wait until I finish making the airframes and adding the required electronics and motors). Alas, the new FC I am after, and awaiting most impatiently, runs cleanflight so I must achieve the same thing in the cleanflight code.

I have found mixer.h and mixer.c, but I do not see the relation between the code there and the MultiWii code. I found, for example MIXER_HEX6H, but have not found where it is used, or how it relates to what is done for HEX6H in the MultiWii code.

If it is the case that Cleanflight is a complete rewrite of MultiWii, what is the Cleanflight way to achieve what I did inthe MultiWii code?

Can someone please enlighten me on how one can add new mixes to the Cleanflight code? 

Thanks. 

Ted

borisbstyle, 2015-06-21:
cleanflight supports custom mixer.
So not the ones from the dropdown menu, but you can add them in the cli. There are manuals for that.
It is also easier than multiwii ;)

tedbyers, 2015-06-21:
I was not impressed with multiwii coding style (I have been coding in C/C++ for decades), but I thought maybe what I saw was the result of a compromise associated with writing firmware (something I did decades ago, but not since).  I like how the code is laid out in Cleanflight much better, and the use of makefiles.  I just now have to figure out what development tools I need install on Ubuntu and/or Windows, and what the process is for installing the binary on a FC board.  Not to worry, at some point I'll find that documentation.....

They're not presently in the dropdown menu, but rather than adding these in the cli, I was looking at adding them to that menu, so I don't have to add them each time I change things about.  Or is it the case that adding a new mix in the cli will store the mix so that it is available in the dropdown menu the next time it is needed?

I have been poking around the code a bit, and learned a couple things.  First, I added the following in mixer.c:

static const motorMixer_t mixerVTailHex6H[] = {
    { 1.0f,  0.0f,  1.0f, -1.0f },     // REAR_R
    { 1.0f, -1.0f, -1.0f,  1.0f },     // FRONT_R
    { 1.0f,  0.0f,  1.0f, -1.0f },     // REAR_L
    { 1.0f,  1.0f, -1.0f, -1.0f },     // FRONT_L
    { 1.0f,  0.0f,  0.0f,  0.0f },     // RIGHT
    { 1.0f,  0.0f,  0.0f,  0.0f },     // LEFT
};

static const motorMixer_t mixerOctoH[] = {
    { 1.0f,  1.0f, -0.5f,  1.0f},  //MIDFRONT_L
    { 1.0f, -1.0f, -1.0f,  1.0f},  //FRONT_R
    { 1.0f, -1.0f,  0.5f,  1.0f},  //MIDREAR_R
    { 1.0f,  1.0f,  1.0f,  1.0f},  //REAR_L
    { 1.0f,  1.0f, -1.0f, -1.0f},  //FRONT_L
    { 1.0f, -1.0f, -0.5f, -1.0f},  //MIDFRONT_R
    { 1.0f, -1.0f,  1.0f, -1.0f},  //REAR_R
    { 1.0f,  1.0f,  0.5f, -1.0f},  //MIDREAR_L
};

static const motorMixer_t mixerVTailOctoH[] = {
    { 1.0f,  1.0f, -0.5f,  1.0f},  //MIDFRONT_L
    { 1.0f, -1.0f, -1.0f,  1.0f},  //FRONT_R
    { 1.0f, -1.0f,  0.5f,  1.0f},  //MIDREAR_R
    { 1.0f,  0.0f,  1.0f,  1.0f},  //REAR_L
    { 1.0f,  1.0f, -1.0f, -1.0f},  //FRONT_L
    { 1.0f, -1.0f, -0.5f, -1.0f},  //MIDFRONT_R
    { 1.0f,  0.0f,  1.0f, -1.0f},  //REAR_R
    { 1.0f,  1.0f,  0.5f, -1.0f},  //MIDREAR_L
};

And then I extended some code in mixer.h as follows:
    MIXER_CUSTOM = 23,
    MIXER_VTAIL_H_HEX = 24,
    MIXER_H_OCTO = 25,
    MIXER_VTAIL_H_OCTO = 26
} mixerMode_e;

Is that sufficient to have these new mixes always accessible?  If not, what did I miss?

On a slightly different note, I have a design for a V-Tail tetrakaidecacopter (that is a MR copter with 14 motors, made by altering my V-Tail H-Frame octo to have pairs of coaxial motors on the front three pairs of arms; the same concept I used to make a decacopter from the V-tail, H-frame hexacopter).  Alas, in mixer.h, I see "#define MAX_SUPPORTED_MOTORS 12"  Can the maximum number of supported motors be increased?  I would assume that is not just a matter of changing that define, so what else would I need to refactor in order to add support for my tetrakaidecacopter?  Apart from this coding question, are there any FC boards that can be bought that support more than 8 motors?  If so, where can they be found?

As for why I want to try this; I am wanting to conduct experiments to determine the efficacy of melding different existing airframe designs to see if the benefits of each can somehow be realized in a single airframe (e.g. can we combine the extra power coaxial designs like the Y6 and X8 with the improved yaw authority of a V-tail, or the extra power of flat hexacopters, albeit on an H-frame, with the improved yaw authority of a V-tail).  And of course, these experiments will be less costly if I can reuse components each in the next build (hence I'd like to build the mixes I need into the firmware instead of having to add it dynamically each time).

tedbyers, 2015-06-21:
That closing was an accidental clicking on the wrong button.  :-(  I sought only to submit the note I wrote.

borisbstyle, 2015-06-21:
In cleanflight go to mixer and define the motors

Type:
mixer custom
Now define all motors according to manual. No need to changing the code

joshuabardwell, 2015-06-21:
@borisbstyle he wants to add a new model to the pull-down menu so he doesn't have to copy over the mix each time.

Personally, I agree that the best solution is to copy/paste the cmix commands. Just save them to text files. No need to fear the CLI.

borisbstyle, 2015-06-21:
I see. I misunderstood that. I think he needs to add it in the configurator as well if he wants the pulldown :D.
He also needs to add it in serial_cli.c as there it can be selected in cli.

hydra, 2015-06-22:
@tedbyers Hi ted, glad to see you found the code and the issue tracker.

Yes, having defaults for the models you mention sounds OK.  Should only take a few bytes of flash, however I do question the value of it at this time since no-one has asked for it before now.

Are there some new commercial frames that are now available?

Is there an issue with just using the existing functionality or do the models require other changes than just mixer defaults and names?

tedbyers, 2015-06-22:
@borisbstyle I will take a look at serial_cli.h and serial_cli.c, but what is this configurator?  What file(s) ought I examine, and in what directory are they located?

@hydra I am not looking at distribution anything at this time as there certainly isn't demand for it, or new commercial airframes, YET (to the best of my knowledge).  You see, I am experimenting with new ideas for UAVs.  As far as I know at present, nothing more is required than adding a mix for each configuration I am testing.  I would like to be able to add them, in code I have on my system, and the deploy that on a couple FC boards, so that I can guarantee that each test with a given configuration is using identical code, and and identical mix, each time I test that configuration even if I have deployed that FC board on a dozen different configurations between each test of a that configuration.  If it is made a question of simply selecting the mix for a given model from a menu, then the risk of human error in entering the mix for  it is reduced, or even eliminated.  In point of fact, I have learned of additional options for the MR copters of interest to me that more than double the number of mixes I need for the suite of possible combinations of options for these models.

borisbstyle, 2015-06-22:
I used to use custom mixers on my spider hexas in the past and I never found it hard to copy the settings in the cli each time.

This were the sets commands to the cli I used for example. I never found it a big deal to do it like this
mixer custom
cmix 1 1 -0.44 0.93 0.84 
cmix 2 1 -0.56 -1.0 0.83
cmix 3 1 0.44 1.0 -0.83
cmix 4 1 0.56 -0.93 -0.84
cmix 5 1 -1 0 -1
cmix 6 1 1 0 1
save

tedbyers, 2015-06-24:
Thanks borisbstyle.

That is appreciated.

I will investigate further.

closed 2015-06-24 by tedbyers