cleanflight/cleanflight #1037: Blackbox samplerate alters pid-calculations

henn1001, 2015-06-20:
I did a lot of testing/logging with the blackbox the last few days to test out the d-filter.
https://github.com/cleanflight/cleanflight/pull/1028
I also posted a few logs and pictures over there because i noticed a strange behavior related to the blackbox logging rate.

So i did a few more tests with complete stock settings on pid1 and only a few changes to make it flyable for me (ppm/oneshot etc.). I attached the cli dump.

cc3d OPBL
zmr250 3S tm-mn2204-2300
sn20A blheli oneshot

https://www.dropbox.com/sh/rtr2suhux3mzger/AABGFaJnsQS3Yq7zXP-D0Kjba?dl=0

I tried various combinations and added a few in the link above. I first thought the problem was related to the d-term filter because the difference between
[image]
and
[image]
was my first finding. (blackbox rate 1/1 vs 1/2) Its introducing a lot of P noise.
My understanding of the rate was that with 1/2 it only logs every second iteration/looptime.

But as you can see here:
[image]

[image]

Rate 1/1 vs 1/10
As expected the data is sparser but also the amplitude of P and especialy D is going all over the place ( time critical parameter).

So idk what to make out of all this.. is this intended? Is it screwing with the looptime? In my eyes it makes tuning with anything beside a rate 1/1 pointless. 

Would be cool if anyone can confirm this.

borisbstyle, 2015-06-20:
@thenickdude can answer this.

You can also try to export the file as csv and generate gpaphs by yourself?

henn1001, 2015-06-20:
Didn´t know i could do that - i´ll analyse it with matlab :)

joshuabardwell, 2015-06-20:
@borisbstyle I wonder whether this could be related to the extreme noise that appeared to be in my graphs. I run BB at 1/2 rate because I figure that at 1600 looptime, that still gives me plenty of sample rate to tune out problems, and occasionally I have dropped frames at 1/1.

borisbstyle, 2015-06-20:
Hacking blackbox for debugging is also possible. replace for example P and I with cycletime and previous cycletime ;)

borisbstyle, 2015-06-20:
@joshuabardwelll
I dont know. I run the same 1/2. Try and compare.

joshuabardwell, 2015-06-20:
Dang. Nevermind then. It's apples to apples. Your graphs were much less noisy than mine.

henn1001, 2015-06-20:
Yes please can someone else compare logging 1/1 vs 1/2 or 1/10. Hoverdata in your room/garden is enough

thenickdude, 2015-06-21:
Blackbox does effectively log the looptime, as it logs the starting time of every loop iteration.

If you run logging rate = 1/2 and your looptime is set so short that it's shorter than your FC can actually achieve while logging, you would definitely see looptime jitter of the same length of a blackbox logging iteration, which is distinctly non-zero, especially when logging to onboard flash. Increasing looptime to something your FC can achieve should cure that.

henn1001, 2015-06-21:
Well i was running a looptime at 2000us. At 1/1 it looked fine to. But how can 1/2 be worse than that?

I also came to the conclusion that only logging via the onboard dataflash seems to be affected by this because i never saw corrupted frames in my testings? Even at looptime 1300 and 1/1. Or maybe the onboard flash can handle this? idk.

thenickdude, 2015-06-21:
If there's no execution time left over in the loop during a logging iteration (because the looptime is too short), then in the iterations where the Blackbox doesn't log, the looptime will end up  faster since it doesn't have that logging work to do. That said, 2000 looptime should be achievable. I'll try it here too. 

The dataflash can basically keep up with every possible logging rate so data is never dropped (in my testing).

henn1001, 2015-06-21:
But then it would show up in the logs (timespamps) right? I had a look at them and didn´t find any skipped frames. They should show up as 4000us @looptime 2000 with 1/1 and as 6000 or 8000 @looptime 2000 with 1/2.

borisbstyle, 2015-06-21:
@thenickdude
I think you should save previous looptimes when not logging to be able to project the logging more accurate when not logged.

thenickdude, 2015-06-21:
That would require potentially many skipped looptimes to be recorded (logging rate goes down to 1/32) and would only really be useful for diagnosing this specific problem, so I don't think it would be a net gain for most users. Useful for debugging this issue though.

thenickdude, 2015-06-21:
Looks like it is not an actual flight problem but rather an artifact of the way that the viewer is smoothing the data. Here's log 2 (logging rate 1/1) with default smoothing:

[image]

And no smoothing:

[image]

Log 3 (logging rate 1/2) with default smoothing:

[image]

And no smoothing:

[image]

Notice that the logging rate 1/2 log's acc and gyro look the same with and without smoothing applied. But the logging rate 1/1 log looks much more similar to the 1/2 log once its smoothing has been removed. 

This is because the smoothing time radius for gyro and acc data in the web viewer is 3ms. Your 1/1 logging rate samples are 2ms apart (within the smoothing radius) and the 1/2 logging rate samples are 4ms apart (outside the radius) and don't get smoothed.

Allowing the user to configure the smoothing radius is a feature I hope to add in the viewer. I'm not sure if using a more accurate smoothing filter is feasible on JavaScript.

borisbstyle, 2015-06-21:
That explains why I was not seeing the problem as I run looptime 1200 and stay within 3ms :)

But the viewer could automatically detect what smoothing should use based on the looptime intervals or average intervals? But ability to turn it off would also be nice.

henn1001, 2015-06-21:
Wow that makes a lot of sense, thank you!! Yeah would be cool if we can enable/disable/change the smoothing in the viewer :+1:

fedorcomander, 2015-06-22:
@thenickdude What do you think to add option to switch blackbox on and off on rc switch????

For example, i have small 8mb flash, and i wanna run it with 1:1 denom. I am not interested in hovering, but some things like punchouts i wanna log. Adding option to have switch 1000 - OFF, 1500 - ON, 2000 (WHEN NOT ARMED) erase flash would be nice. What do you think?

borisbstyle, 2015-06-22:
@fedorcomander 
+1

fedorcomander, 2015-06-22:
Usually when you want something to log its already full ;) Story of my life ;)

joshuabardwell, 2015-06-22:
IMO, if you're going to do this, the right way to do it is by adding adjustment functions. That way the user could configure the switch and aux channel however they liked.

fedorcomander, 2015-06-22:
Not really. I was thinking more about RC modes. The adjustments are something else :)

thenickdude, 2015-06-22:
I think there is an issue open for a flight toggle switch already. Anyway it is off-topic for this issue, which appears solved, closing.

closed 2015-06-22 by thenickdude