cleanflight/cleanflight #1040: BMP180 can spoof MS5611 detection code

SteveAmor, 2015-06-22:
While initialising a board with a BMP180 baro connected, some BMP180's can pass the MS5611 CRC test resulting in the false detection of an MS5611 baro.

Alternative firmware test for a BMP085/BMP180 before testing for an MS5611, to get around this issue.

A better solution would be to make the MS5611 detection code more robust so it doesn't matter which way round the detection tests are performed.

stronnag, 2015-06-22:
The I2C board from http://www.ebay.co.uk/itm/111424501317  (BMP180 + HMC5883)  does this consistently. The workaround of reversing the order of BMP085 / MS5611 tests (as legacy firmware)   results in correct detection (both for this board and for MS5611 on standard Naze). As noted above, this expedient is hack, addressing the symptom rather than the cause.

ledvinap, 2015-06-22:
Can someone check values that are read from BMP085 in MS5611 detection routine?
Is is quite probable that first 5 values are 0x0000/0xffff, this can be used to fail detection (at least C1 must be nonzero for calculations to make sense)

stronnag, 2015-06-22:
For the above GY 652 device, the following are read consistently.
```
index  dec   hex
0	-23148	a594
1	 18441	4809
2	-17881	ba27
3	 31297	7a41
4	-14319	c811
5	  9229	240d
6	 -1019	fc05
7	-14557	c723
```

SteveAmor, 2015-06-22:
@stronnag to save me time, how did you obtain those values?  Although, I am not able to provide my readings for a few days. 

I intend to write (have a go a writing) a unit test for ms5611 crc method as soon as I can.

stronnag, 2015-06-22:
I used debug[0..3] (twice)  to hold the ms5611_c[0..3 and 4...7] values immediately before the  ms5611_crc() call, and inspected them in the configurator. Repeated multiple times to validate consistency. Then tested by setting debug[0...3] to sentinel values in the same place, and verifying that the sentinel values were seen in the configurator.

ledvinap, 2015-06-22:
Hmm ... I checked on AfroMini (bmp180) : {0xa594, 0x4219, 0x2d27, 0x6252, 0xa65a, 0x2192, 0xfb40, 0xc643} (and the rest 0x8414 0x6337 0x45d0 0x1973 0x0031 0x8000 0xd1f6 0x0a11)

Is seems that there is 16 words of eeprom implemented, but only last 11 are documented

Reading ms5611, i get: {0x58, 0xcfca, 0xd2b3, 0x7dca, 0x7503, 0x7b3a, 0x6bbc, 0xa582, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0}

So my ms5611 returns zero for addresses that are not documented (upper eight bytes). 

Testing these bytes is probably quite safe, all Meas-spec devices I checked (very quickly) do use 8x16-bit configuration ...

@SteveAmor : CRC calculation probably is OK, but CRC is only 4bit, so 6% of random configurations will have matching CRC ...

SteveAmor, 2015-06-23:
@stronnag Just to confirm that your BMP180 data does have a correct MS5611 PROM CRC!!  I just checked with my ms5611_crc_test Unit Test.  Likewise, @ledvinap ms5611 passed crc check.

ledvinap, 2015-06-24:
So ... what about reading word after ms5611 eeprom space and failing detection if it is nonzero?
And modifying detection so that this test is used only for baro autodetection, not when baro is selected explicitly?

stronnag, 2015-06-24:
Putting
```
   if (ms5611_prom(PROM_NB) != 0)
        return false;
```
before reading the other coefficients does the right thing for my BMP180 as well as MS5611 on full naze32 and flip32+ boards.

@SteveAmor , will you do the PR with your unit test?

SteveAmor, 2015-06-24:
I'm really not that keen on using undocumented addresses as a solution.

ledvinap, 2015-06-24:
@stronnag : This test should trigger only when baro autodetection is selected - user can override detection if necessary ..
@SteveAmor : Would be great to have something better. But this is real problem and documentation does not help here much ..

SteveAmor, 2015-06-24:
@stronnag trying to detect a bmp085/180 before trying to detect an ms5611 is also a solution. It's a question of which solution is better.  I couldn't find a #define for baro autodectection. 

Changing the order of detection (with a comment in the code) saves code space and doesn't rely on undocumented addresses - now or in the future.

SteveAmor, 2015-06-24:
Guys, I've gone for the "test for a BMP085 before MS5611 approach" and raised PR #1052.

closed 2015-06-24 by SteveAmor