cleanflight/cleanflight #1043: LuxFloat, OneShot, causing motors to overheat, shorter flightime and could completed PID tuning vs. other PID ctrls

directgumby, 2015-06-23:
Hello,

I was trying to do a PID tuning on my 450, with 800Kv motors running 10x4.5 props, Afro 20A ESCs with BlHelli OneShot setup, along with my Naze32 board today.  When doing this I had what seemed to be very odd behavior, the motors would be super hot, the battery would be draining faster and the battery was also very hot, the ESCs were fine.

I was adjusting up to this PID point and then it was having trouble lifting off... so I swapped over to Mode 5 and started to tune that - much better on the motors, ESCs, and flight time was back to where I expected it, nothing over hot - much, much better.

Here are the settings for the LuxFloat setup profile settings:

ONESHOT123 enabled
set looptime = 1500

set pid_controller = 2
set p_pitch = 40
set i_pitch = 30
set d_pitch = 23
set p_roll = 40
set i_roll = 30
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_pitchf =  4.100
set i_pitchf =  0.310
set d_pitchf =  0.031
set p_rollf =  4.100
set i_rollf =  0.310
set d_rollf =  0.031
set p_yawf =  8.000
set i_yawf =  0.500
set d_yawf =  0.050

Let me know if you need any other info from me to help.

Thanks!

Arakon, 2015-06-23:
Those PIDs are very high for luxfloat. Chances are you simply ended up having lots of tiny oscillations, which will of course cause things to heat up.

borisbstyle, 2015-06-23:
Your PID's look way too high indeed. In my case I anded on all my quadcopters 3S and 4S ones with value between 1.0 and 2.0

But there is a slight true in this story that luxfloat seems more prone to micro vibrations than other pid controllers. This PR might improve that
https://github.com/cleanflight/cleanflight/pull/1028

hydra, 2015-06-28:
Closing as mis-configuration / support

borisbstyle, 2015-07-09:
New defaults for luxfloat and further improvements
https://github.com/cleanflight/cleanflight/pull/1103

closed 2017-02-25 by hydra