cleanflight/cleanflight #1048: LED enhancement with VBAT

Fvk1960, 2015-06-23:
I like to propose a LED option to display the VBAT status. LEDS that changes from Green over yellow to RED indication battery voltage and at last flashing RED at minimum voltage.
To reduce battery consumption a dim function on all LED linear to batteryvoltage and or option to control brightness.
Thanks

hydra, 2015-06-28:
Do you mean LED or LEDSTRIP?

dkisselev, 2015-08-11:
I'm pretty sure LEDSTRIP.

Seconding the request though, being able to program a few nodes on the strip to show battery status would be pretty useful.

Fvk1960, 2015-08-11:
Yes the ledstrip function.

andymac4182, 2015-10-10:
It would be great to have this like Ring Thrust State. Where you could set the min and max or auto detected and it would show the battery going flat and changing colour as it did.

closed 2017-02-25 by hydra