cleanflight/cleanflight #1051: gimbal_flags=4 doesn't work

joshuabardwell, 2015-06-24:
I've observed that gimbal_flags=4 doesn't seem to work on some post-1.9 versions of the code. I have tested:

* 1.8.1 - **Works**
* Custom 1.8.1 from Boris B dated a month ago or something like that - **Works**
* 1.9.0 stable release - **Works**
* Custom 1.9 from Boris B dated June 18 - **Doesn't work**
* Custom 1.9 from Boris B downloaded today - **Doesn't work**

To reproduce this issue, I flash the firmware with full chip erase, then from the default config, enter:

'map RETA1234
'feature rx_ppm
'set gimbal_flags=4

I then flip the Aux switch and see if my LEDs and beeper respond correctly.

I know some changes have been going on with custom mapping of outputs, so I'm not sure whether this is related to that shuffling or whether it is unintentional and a bug. I searched for closed issues that seemed like they would cause this and didn't find any--but I don't know the codebase well enough to be confident I didn't miss it.

borisbstyle, 2015-06-24:
My build was from the current master. 
Should try the current master as well to verify.

But i think some other change lately is affecting this

borisbstyle, 2015-06-24:
maybe this change could be responsible

https://github.com/cleanflight/cleanflight/commit/7db626747d5e4d179b87f5935148cdea4cdf9ea6

joshuabardwell, 2015-06-24:
>Should try the current master as well to verify.

I'm not currently set up to compile, but maybe this will be the motivation.

joshuabardwell, 2015-06-24:
Just tested based on Boris's latest build, compiled this afternoon from the current master. Same problem is present. So it seems like something in the master has changed or broken gimbal_flags=4.

borisbstyle, 2015-06-25:
@hydra 
I saw you did a lot of changes regarding servos. Can you check this one?

joshuabardwell, 2015-06-25:
There was also that change that pushed Aux1 and Aux2 over to motor headers 5 and 6 when feature ledstrip is enabled. It could be related.

ethomas997, 2015-06-25:
I believe it happened with commit 889b14d; I noticed the issue on my tricopter, not sure about other mixes.  I tagged this comment on the commit:

@hydra After applying this with a tricopter mix, the servo outputs enabled via "gimbal_flags=4" got shifted -- the ch6 output became a ch7 output. Adding a "servoIndex++;" statement to the end of the "case MIXER_TRI:" block in "mixer.c" restored the functionality. I don't know if this is the "correct" fix, but it worked.

--ET

joshuabardwell, 2015-06-26:
I'm not sure that just shifting the outputs is the problem. I'm not 100% sure, but I'm 99% sure that I tested all of RC5-8. None of them drove a servo, and also I put a volt meter on them and they were all at 0 volts, whereas I would expect to measure a few hundred mV if there was a PWM signal on them.

ethomas997, 2015-06-26:
What mix were you using (tri, quad,...)?

joshuabardwell, 2015-06-26:
I can provide a dump. It's a custom mix.

https://gist.github.com/joshuabardwell/65038aff895da88c0ea6

joshuabardwell, 2015-06-26:
I did a full erase before each flash, so it would have been the default QUADX mix. Like I said, the only commands I entered before testing were:

    map RETA1234 
    feature rx_ppm
    set gimbal_flags=4

ethomas997, 2015-06-26:
Right now I'm only setup to test TRI mix and CH6 output, and for that configuration the fix I mentioned worked.  Pretty sure that commit (889b14d) is the culprit, as it makes significant changes to the 'forwardAuxChannelsToServos()' function.

--ET

joshuabardwell, 2015-06-26:
I'm going to test this now.

SteveAmor, 2015-06-26:
I'll do a ```git bisect``` over the next 24 hours to find the commit that introduced this.

joshuabardwell, 2015-06-26:
@SteveAmor thanks. I discovered this issue the hard way, after crashing in some bushes and not being able to use my buzzer to find the copter (I did find it, eventually). They'll get my #1028 when they pry it from my cold, dead hands, but gimbal_flags not working is still a concern. It'd be a real pisser if I was using a tilt servo or gimbal.

hydra, 2015-06-26:
It's because some code was moved.

https://github.com/cleanflight/cleanflight/commit/889b14df5f6e3a14e3c50660469c869cd526b906#diff-95e70d63cf1e155a276b2ad2a8e7a947L713

moved to

https://github.com/cleanflight/cleanflight/commit/889b14df5f6e3a14e3c50660469c869cd526b906#diff-95e70d63cf1e155a276b2ad2a8e7a947R514

when gimbal flags is set without using a mixer that uses servos the code will not be run, e.g. for a quad.

https://github.com/cleanflight/cleanflight/blob/889b14df5f6e3a14e3c50660469c869cd526b906/src/main/flight/mixer.c#L453

I will fix this, not sure how yet.

hydra, 2015-06-26:
Simply deleting the `if (useServos)` check is enough.  Doing a bit more testing...

hydra, 2015-06-26:
@ethomas997 With regards to the tricopter mix, i'm guessing that before the change there was a spare outpin next to where you plug in the tricopter tail servo.  The change mentioned *should* allow the unused pin to be used.  This is change important for targets with limited outputs and re-mappable inputs.

closed 2015-06-26 by hydra

joshuabardwell, 2015-06-26:
@borisbstyle could you let me know when you next put a build up (that includes this commit)?

borisbstyle, 2015-06-26:
@joshuabardwell
I am off to field to test my latest changes and if that goes well will put it up later on.

joshuabardwell, 2015-06-26:
Can confirm that the latest build fixes the gimbal_flags issue for me.

ethomas997, 2015-06-26:
@hydra On my little tricopter I have a wire soldered on to the pad for a CH6 tilt-servo output, so a hard-coded shift to the positions of those outputs would be undesirable.  (I am currently traveling and probably won't be able to test this for a few days.)

--ET

imagebuff, 2015-07-19:
Running Boris 1.10.0 Jul 16 2015 00:29:22 it seems the CLI refuses to even recognize gimbal_flags.  I have a cam tilt that was making use of gimbal_flags=4 with CF 1.9 (early version).

set gimbal_flags=4 
Invalid name

joshuabardwell, 2015-07-19:
Addressed on your issue #1141. This is a feature change, not a bug.

commit 06086016 (2015-06-26): Ensure that all AUX channels are forwarded to any available unused servo
output.

Fixes #1051.