cleanflight/cleanflight #1053: Sync control loop to IMU data

atomiclama, 2015-06-25:
I'd like to work on syncing the IMU data to the control loop.
I can't find any information if people are working on this or not.
I have seen some code in the IMU to handle the data ready and tie it into an interrupt but that's it. 

So I'd like this to be a discussion place and also let you know I intend working on this unless someone else is well under way with it. Don't want to duplicate effort.

What's the consensus ?

ledvinap, 2015-06-25:
I have partial, quick-and-dirty, but flyable  implementation .. 

https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/accgyro_mpu6050.c#L470

atomiclama, 2015-06-25:
Thanks I'll have a look.
Did you notice any improvement?

ledvinap, 2015-06-25:
I don't know, I didn't compare it yet ...

hydra, 2015-06-28:
@borisbstyle will likely be interested in this.

Yes, its something that has been on the cards for a while now.

Be aware that not all boards out there have the MPU_INT lines wired up so this *must* be an optional feature, even for boards like the SPRacingF3 that do have it wired up.

Perhaps for some boards we could default it to being on.

digitalentity, 2015-06-28:
Maybe a better approach would be to make control loop wait for EOC interrupt (like it is done in TauLabs). Once data is ready, process it and act on it.

ProDrone, 2015-06-28:
@digitalentity Yes! To do it right, we actually need to sync the control loop to the IMU data-ready, make all calculations and then sync the moment the output is written with a timer to make the whole system deterministic. With respect to PID control there is no use in a faster loop time. Other activities may see improved performance with a faster loop time.

ledvinap, 2015-06-28:
@hydra: With FIFO it is possible to check FIFO length and thus determine when new data was written. But reading this information is a bit slow over I2C (AD+W RA AD+R DAT DAT), about 12us (35us to read gyro + accmeasurements)

borisbstyle, 2015-06-28:
@hydra Yes I was indeed interrested in checking this and was already looking to ledvinaps experiments with this one.
I just think it's not a major win against what it is now when running low looptimes. If i2c wasn't the limitation in this one it could have been usefull probably.
Still something I would like to see if it can help.

borisbstyle, 2015-06-29:
@ledvinap 
Would you be able make a PR with integrating your FIFO approach to the current master?
I think a simple approach would be fine to start with where the loop always waits till the new data is retreived without filtering.
Well the pt1 algorythm will still be applied to it eventually.

ledvinap, 2015-06-29:
IMO looptime only ~10x quad natural frequency is neccessary (~100 - 200Hz), there will be improvement with faster looptime, but not that much.
IMU sampling is possibly one reason why short looptime work better in practice (and why 256_NOLPF2 helps) (but I may be wrong)

Waiting for new data is quite easy, just reading INT_STATUS should do the trick (possibly set MPU6050 samplerate divider to that data are read out fast enough).
Filtering is important if full 1kHz readout is impossible - it will reduce jitter and aliasing ...

atomiclama, 2015-06-29:
This is what I plan to try out.

Use the data ready int to kick of a read of the IMU this will be done within ISR DMA, need to look at the FIFO, once all the required data is read in I'll set a flag indicating new data ready.
In the main loop at the start of the loop it will wait until this flag is set before starting the main loop. I'll implement some timeout incase the flag does not get set for some reason.

This is basically how the products work in my job where we have a 500us cycle time synchronised to the current measurement used in SLVC control. The main improvement we saw when synchronised was improved current waveforms.

This may take me some time due to other commitments and having no quads to fly at the mo, they are all in bits. But I'm keen to give it ago.
I have a CC3D and a Naze32 to try on. 

Cheers

ProDrone, 2015-06-29:
Syncing to sensor read is one (important) thing, but the processing of these readings may take a different time on each cycle. So to be deterministic (and have no jitter) you need to time the output to the actuators (ESCs) as well. Measuring the longest processing time and use that for write-timing is a good method.

borisbstyle, 2015-06-29:
@ProDrone 
Well right now it is also different time each cyle. This would not be issue if you do it on simillar way like ledvinap did in his local. He actually collects a few measurements first....than applies filter to it and writes out to the motors. That slows down the actual refresh rate to the motors, but makes it more accurate and realtime.

ledvinap, 2015-06-29:
@atomiclama : You are overengeneering it. Using DMA with I2C will require rewrite of I2C driver.
With cleanflight loop() structure all that is necessary for test it to check data_ready bit and enter imu/pid/mixer code conditionally.
With FIFO, you can just check for new data on each loop and read data available (reading is done as soon as possible, so there is less delay where processing in entered), then enter imu/pid/mixer if enough samples are ready.

ledvinap, 2015-06-29:
@ProDrone : IMO processing IMU data synchronously will be much more important than syncing output.

atomiclama, 2015-06-30:
@ledvinap : yeah was thinking more how I do things at work rather than what is easiest or best for cleanflight.  cheers

bdoiron74, 2015-07-15:
I believe you can read the data ready status bit from an MPU register (INT pin not necessary) and adjust the MPU sample rate to an achievable looptime (1kHz, 500Hz, 250Hz...). I haven't looked closely, but I think the internal MPU filter operates on it's native 8kHz gyro sample rate, so you'd want to pick a filter cutoff to match the output rate. Also, the cycletime used in PID calculations should be exactly the MPU sample rate (or division of, if manually combining higher rate samples), not the jittery time between MPU data reads.

borisbstyle, 2015-08-04:
This is my implementation of it. 
https://github.com/cleanflight/cleanflight/pull/1186

atomiclama, 2015-08-05:
@borisbstyle yes been following with interest. Hats off to you sir for the work you are doing, wish I had the time to spend on CF.

closed 2015-08-05 by atomiclama

borisbstyle, 2015-08-05:
@atomiclama Thanks. Well you brought me to the idea :D. It was something on todo list for a while and now in holiday periods managed to get some time.

atomiclama, 2015-08-05:
Going to have to find something else to look at in my spare time now. ;-)

borisbstyle, 2015-08-05:
@atomiclama Haha
BTW I do need some help with other less popular targets, which are not familliar to me. Not sure if someone can help? There is sparky etc.....

atomiclama, 2015-08-05:
@borisbstyle I'm running CC3D, if there is anything left to do when I get back from my Hols, I'll help out with testing or other.