cleanflight/cleanflight #1054: GPIO on Flip32+

sirachacopter, 2015-06-25:
All,

Is it possible to control the GPIO pins PA15 and PB5 with the Flip32+/Naze32 flavor of Cleanflight? I see that PA15 can be used for a buzzer circuit for CC3D (https://github.com/cleanflight/clean...docs/Buzzer.md). 

I have a Flip32+ and from schematics it looks like these pins aren't brought out to any connectors. 

Annotated Flip32+ schematic: http://static.rcgroups.net/forums/attachments/5/9/1/7/2/0/a8000444-198-image.jpg

Board, showing unconnected pins:
http://static.rcgroups.net/forums/attachments/5/9/1/7/2/0/a8000446-70-image-07228374.jpg

But assuming I can solder wire on, is there built-in support for using these with, say, a relay to control a single-color 12V LED strip? Or would code need to be customized for this?

Or can I somehow use the RGB LED support to control single color LED strips? 

Thanks,
Sirachacopter

joshuabardwell, 2015-06-25:
This may not be exactly what you're looking for, but you can get an RC-activated relay/switch that plugs into a receiver channel and turns on when the channel goes above 1500 (or whatever). Then use gimbal_flags=4 to pass through your Aux channels to RC5-8 and use an Aux channel to control the strip.

Since you're considering using the ledstrip feature, I assume you have pins RC5-8 free.

If you are using any of the unofficial releases past 1.9.0, gimbal_flags=4 appears to not work, however. You should be fine if you are using 1.9.0 or before.

sirachacopter, 2015-06-25:
Wow that's pretty cool ... did not know about the radio controlled relay
option.

I am going to see if there are schematics online for them. Maybe I already
have the parts needed for this setup.

However, intuitively it seems like the FC would be removed from this
configuration: if the relay is plugged into a receiver channel, what does
the FC have to do with this?

I'm probably confused.

Thanks

joshuabardwell, 2015-06-25:
Here's an example. In order to build your own, you will need something to detect the timing of the pulses.

http://hobbyking.com/hobbyking/store/__8833__Turnigy_Receiver_Controlled_Switch.html

The FC is removed from the configuration. The assumption is that you are using serial receiver or CCPM so that you don't have individual PWM channels being output by the receiver. If you are using PWM, then plug the switch direction into the receiver and forget about gimbal_flags=4..

sirachacopter, 2015-06-25:
Ah makes sense now. Indeed I am using PWM.

While this post is still open, have you ever heard of anyone using the
Naze/Flip32+ GPIO with CleanFlight?

Thanks for your help.

joshuabardwell, 2015-06-25:
BTW, I don't know enough about LED strips to know whether there is an easier way to control a single-color strip with an RGB controller. Maybe there is.

Not sure what you mean by controlling the GPIO with Cleanflight. All of the GPIO pins on the board are used for various things, but that's not what you mean...

sirachacopter, 2015-06-25:
There are two pins mentioned in the Naze32 manual:

PA15: 5V tolerant GPIO
PB5: 3.3V tolerant GPIO

The CC3D flavor of Cleanflight uses PA15, but the documentation doesn't
mention any use for these pins on the Naze flavor of Cleanflight.

I was wondering if there was built in control of these that isn't mentioned
in the documentation.

Sirachacopter

joshuabardwell, 2015-06-26:
I don't know about that.

PiManASM, 2016-04-08:
I'm also wondering about this,is there a way to have the radio (FrSky Taranis on SBus) toggle logic high/ low on the GPIO pins? 

(I see that this is a year old)

closed 2017-02-25 by hydra