cleanflight/cleanflight #1056: Add SPEKTRUM_BIND to SPRACINGF3

giryan, 2015-06-26:
from a brief look at [src/main/target/SPRACINGF3/target.h](https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/src/main/target/SPRACINGF3/target.h)
it appears that the spektrum binding define hasn't been set.
I can't build or test right now, but I think that this ought to be the change? 

#define SPEKTRUM_BIND
#define BIND_PORT  UART3_GPIO
#define BIND_PIN   UART3_RX_PIN

giryan, 2015-06-26:
I don't really want to put this in as a pull request yet, but this is the change I have so far:
https://github.com/giryan/cleanflight/commit/af4eec604395c2b1d904564a063f7645796f3d5a

hydra, 2015-06-28:
Its done, maybe you used an old build. I tested it yesterday.

closed 2015-06-28 by hydra