cleanflight/cleanflight #1059: SpracingF3 Bootloader Failed when flashing

borisbstyle, 2015-06-26:
Cant seem to get to flash mine. The board works fine for the rest. I can connect and change things.
[image]

fly4props, 2015-06-27:
I'm in the same boat, The board works but trying to upload new firm ware doesnt work. I am getting the same bootloader failed era.

fly4props, 2015-06-27:
BorisBstyle try checking the box that says Manual Baud Rate. I forgot to check it even if the baudrate was set. Hope that helps you.

borisbstyle, 2015-06-27:
Thanks. What baudrate did you select?

borisbstyle, 2015-06-27:
i think this issue should be moved to the configurator if this is true than. It should select the proper settings automatically

fly4props, 2015-06-27:
256000 Boris. Mine was selected just the check box had to be checked.

borisbstyle, 2015-06-27:
Yes it works :D

closed 2015-06-27 by borisbstyle

fly4props, 2015-06-27:
great!!!!

Trussroads, 2015-06-27:
> rubs hands together in anticipation!< :D

@hydra Does this mean you have you shipped the first batch then?