cleanflight/cleanflight #1072: Command max_angle_inclination

dubik3e, 2015-06-30:
Hello,

Changes of this command do not make any changes. I change it from standard 500 to 850 without any quad behaves same - same angle of pitch and roll. Of course I checked by "get max_angle_inclination"  value and save it. My setup is CF 1.9, CC3D, quad 250 with luxfloat controler and I'm flying in angle mode.

ctzsnooze, 2015-07-20:
I can confirm this is not working in LuxFloat and have identified the cause, at least in that PIDC, and any others that use this code:

    errorAngle = (constrain(rcCommand[axis], -((int) max_angle_inclination),
                    +max_angle_inclination) - inclination.raw[axis] + angleTrim->raw[axis]) / 10.0f;

This code constrains rcCommand within the max_angle_inclination values, then generates the error(used later in the code) based on the difference between that value and the actual inclination of the quad.

In the GUI max_angle_inclination has the range 100 to 900, so the intent is to be able to set this angle anywhere between 10 degrees and 90 degrees.

It looks fine doesn't it?  It does work - please confirm this - *below* 50 degrees, for example when max_angle_inclination = 200 it will really limit to 20 degrees.

But it won't work above 50 degrees.  

The reason is simple: rcCommand does not go above 500!  

So we can limit rcCommand to 800 or 900 but the constrained value cannot go above 500 - hence the maximum effective angle cannot be more than 50 degrees.  That's the cause of this problem

:-)

I don't know the best way to fix this... need help from coders and testers - shouldn't be too difficult.

Cheers - C

thenickdude, 2015-07-20:
If that's the case then you can increase rc_rate to achieve higher tilt angles. It's a multiplier that scales up rcCommand above 500. 

Setting it to 180 should allow 90 degree tilts. However I wouldn't be surprised if the quad flies crazy at exactly 90 degrees from gimbal lock.

ctzsnooze, 2015-07-20:
Good idea for testing!   :-)

But it should work without a hack like that, which also changes many other things...  :-)

It's an easy thing to reproduce so its a big thumbs up for whoever can make the best solution...

In the code above, I tried simply multiplying rcCommand by 2.0f, but it didn't solve the problem, or maybe I didn't do what I thought I did....   so much for my programming skills  :-(  ha ha

ctzsnooze, 2015-07-20:
PS I think it flies without any problem around vertical.  I have a small Horizon modification that also reduces levelling as the absolute angle of the quad increases... allows controlled inverted flight in Horizon... So I would predict no instability at all around 90 degrees from gimbal lock in level modes.

borisbstyle, 2015-07-23:
I thought Horizon was invented for these purposes?

ctzsnooze, 2015-07-24:
@borisbstyle - Horizon mode has no angle limit, so the quad can be flipped with big stick inputs.  It allows flips and rolls etc but also provides for some levelling as well.

Level mode however has an angle lock or angle limit.  This is intended to stop the absolute angle of the quad exceeding the value specified in the CLI for max_angle_inclination.  As a result, in level mode it is impossible to flip or loop or get any angle on the quad in excess of max_angle_inclination.

The default value for max_angle_inclination is 500 (50 degrees).  

The OP has found that if he wants the maximum angle to be (say) 70 degrees, and he sets the CLI value to 700 (allowable range in CLI includes up to 900), the quad still stops at 50 degrees.  

So the intent here is to fix the code so that max_angle_inclination works properly when set above 500.  At the moment it doesn't because max_angle_inclination is only a limit on the stick angle, and the stick angle itself can't exceed 500, as far as I can tell, so setting max_angle_inclination above 500 actually does nothing.

If you could help out with the best way to fix it I'd be grateful.

bdoiron74, 2015-07-24:
@ctzsnooze : It flies with without any problem around vertical in acro and 'kind of' with the original horizon, but I'm pretty sure it'll go a little crazy in angle mode if you set a 90 pitch limit and stuck it there where the roll inclination has lost all meaning... plug in to the configurator and watch the artificial horizon with the quad pointed nose down - roll flips all over the place due to the discontinuity. The leveling code should ignore the roll inclination more and more as it approaches +/-90 degrees pitch. It sounds like your new horizon code does exactly that (but for both pitch and roll). As the roll approaches 90 degrees, pitch level corrections also stop making sense because they're applied to the pitch axis, but at roll==90, correcting the pitch inclination would actually be a yaw rotation... I don't see any proper earth frame -> body frame corrections in this code for any of the auto level implementations.

The errorAngle calculations should probably apply a cosine of the opposite inclination axis to zero out when they're trying to do the wrong thing.

iforce2d, 2015-07-24:
fwiw max_angle_inclination angles above 50 degrees work fine with PID 1. OP says he is using luxfloat so perhaps this is just a luxfloat thing.

borisbstyle, 2015-07-24:
I also think it's a luxfloat thing as I asked some guys who I know from racing who fly level and they have all angles above 50degrees without issues on other pid controllers.

@ctzsnooze or others
Luxfloat has always been categorized as an experimental pid controller. I am now busy with improving it and finalize defaults and pieces of code. Please make what suggestions on level mode you would like to see in this PR https://github.com/cleanflight/cleanflight/pull/1103. I think the current rate/acro mode is fine, but I am not sure how the most would like to have the level/horizon defaults.

bdoiron74, 2015-07-24:
What I was saying about the auto level error signal generation and why it would be a problem at roll or pitch == 90deg applies to all of the PIDs.

borisbstyle, 2015-07-24:
I just think level mode is not designed for 90 degree angle. Once you go 90 degree you are looking straight to the ground and it can level  to all sides. Does it work on 89 degree? 
If that''s the case a limit should be placed on the max_inclination to 89degree or lower.

I think if autolevel mode should be flown on such a extreme angles you have to consider switching to horizon mode. That's where it's for invented.

bdoiron74, 2015-07-24:
Absolutely true. 89 might not be ok either .. you have to leave room for overshoot.

I think I may port what I did over on my APM tree sometime to illustrate the improvements that can be made to the auto level code using body frame transformations.

borisbstyle, 2015-07-24:
@bdoiron74 I really do wonder if it is needed for angle mode to go that far? What's the point of selflevel if you want to use extreme angles?
I would say that the level mode should not need more than let's say 60-70 degrees and even that is much.

dubik3e, 2015-07-24:
I not agree. In level/angle mode should be limit more than 50. Maybe not 90 but 80 should be enought. If someone don't want he will not change it and leave it on 50deg..

bdoiron74, 2015-07-24:
In horizon mode, auto level can come into play at any orientation whenever you let off the sticks, so you very well could be at 90 deg pitch. Which is why the current implementation(s) can leave the quad pointing in random direction depending on where you let off. Done right, the quad maintains it's heading throughout the level correction. Also, I have other improvements in that tree that ensure you can do I nice slow for circle by dealing with the fact that the auto level roll direction slows your commanded direction up to roll=180, then adds to it when it flips sign to -180. 

So mostly unrelated to the max_angle_inclination setting, but some applies if you want to allow extreme angles in angle mode.

borisbstyle, 2015-07-24:
Exactly my point! Use Horizon when you want to go to extreme sticks! That is really what you need. I don't think there is any craft out there what can fly constantly on angle of 90 degree or even 70. You will start losing altitude from that point and you never can maintain that angle for long. I do agree that more than 50 should be supported anyway.

borisbstyle, 2015-07-24:
@bdoiron74 That's a different situation what you describe there. Level mode also works when you go inverted in rate/acro mode and quickly flip to self level. You will still recover, but in your case you are removing the sticks from the angle. In my opinion the stick input should represent an angle. When you go left 90degree it could continue moving left due to leveling while you are moving stick from left to right. That is against the leveling principles if you ask me.

borisbstyle, 2015-07-24:
@ctzsnooze The best fix for luxfloat limitation is to remove degrees from the actual stick input, but add a percentage calculation of the maximum angle.
If max angle is 70 than 500 should be 100% of 70.

bdoiron74, 2015-07-24:
@borisbstyle - yes, in angle mode the pitch/roll rcCommand should represent a desired orientation. And the craft should take the shortest path to get there - ideally maintaining the same heading (assuming no commanded yaw).

ctzsnooze, 2015-07-24:
I agree with the concept that in level the stick angle should represent an angle. With the default max_angle of  500, full stick (500) should result in an angle of 50 degrees.  If the user changes max_angle to 700, then full stick (500) should result in an angle of 70 degrees. 

And definitely in Level mode, if activated at any angle of the quad, whether inverted or anything, it should re-level to the current angle of the stick.  So if you have gotten into trouble in Acro and are flipping fast and falling out of the sky,  you can hit level and throttle up and soon you are level and heading up no matter what the angle of the quad when level is activated.

And it would be good to retain heading if that is possible.

I am sure a maximum of 800 (80 degrees) would be OK.

If we can fix the level code in Lux to do Level Mode well, that would be really great.  Thanks for all your input - I'm sorry it's not an exciting topic but for many people it is very useful.

dubik3e, 2015-07-24:
@ctzsnooze Agree in 100%.

jaapp, 2015-07-26:
@borisbstyle I wanted to comment on your remark: "**Use Horizon when you want to go to extreme sticks!** That is really what you need. I don't think there is any craft out there what can fly constantly on angle of 90 degree or even 70. You will start losing altitude from that point really quickly and you never can maintain that angle for long."

I disagree with the part I bolded out. There is a big difference between Angle mode and Horizon mode which will make people prefer one over the other.

In Angle mode the multirotor's attitude is always linked to the absolute stick position of the controller. Which means that you can never make a flip, but also that you never need to mentally keep track of the attitude (which is one of the hardest aspects of this hobby and the biggest benefit of the flight mode). It makes perfect sense wanting to fly 90deg (or 359 deg) even if that means losing altitude.

In Horizon mode there is this transition between Angle mode and Rate mode depending on the amount of stick input. It means that at a certain amount of stick input the attitude is no longer linked to the stick. Which gives you the freedom of doing flips, but it makes determining the attitude tricky. It's not quite Rate mode and not quite Angle mode either.

For me personally Horizon mode was confusing. I didn't learn how to fly Rate(I had to fly Rate mode for that :) ) and I crashed often because it was unclear where the craft was heading. As a beginner I flew Angle mode and only momentarily switched to Horizon if I wanted to make a cool flip.

So there is a good reason wanting to fly Angle mode with at an extreme angle, instead of using Horizon mode.

borisbstyle, 2015-07-29:
@ctzsnooze 
The reason that it didn't work for you is probably because you were modifying it in the wrong part of the code from what I can see in your remarks.

bdoiron74, 2015-08-03:
How would you feel about a controller that didn't suffer from gimbal lock - allowing arbitrary roll/pitch endpoints for the stick in angle mode? So far it's just a conceptual model in python, but the next job is to shove it into cleanflight for experimentation. 

Here's a video - note the pilot inputs up top. [Quaternion Position Controller Simulation](https://youtu.be/3tcacDUp7qo)

digitalentity, 2015-08-03:
@bdoiron74 
It funny, but just yesterday I modified CF to use  Sebastian Madgwick's MARG (quaternion-based sensor fusion). All of the PID-controllers still use euler angles, but results are much better in terms of a more locked in position estimation. But it adds a HUGE overhead to looptime (about 400us on F1 target) and also consume alot of memory (~4k even with #1107).

digitalentity, 2015-08-03:
@bdoiron74 
Just tried to fly it - much worse than plain old complementary filter. HORIZON mode is very slow to react, copter feels mushy. Probably it's due to incorrect *beta* used. If you wish to play with it and get it working - feel fry to grab a copy of https://github.com/digitalentity/cleanflight/tree/quaternion-imu-for-nav

bdoiron74, 2015-08-03:
I'm going to try and implement what I mocked up in python for that simulation. quat_imu -> quat_pos_control -> rate PIDs. The position controller should lock in all flight modes and turn unachievable pilot commands into lowest torque alternatives. It obviously all depends on a good attitude estimate in flight, so filtering out actual acceleration will be important.

ctzsnooze, 2015-08-03:
I figured out - with Boris's help - a modification to Lux's level mode that:

a) doubles the stick responsiveness, which makes it usable for normal pilots, and 

b) allows max_angle_inclination to work properly, i.e. if you set max_angle to 700, it will go to 70; without this mod, setting max_angle to 700 you are still stuck at 50 degrees.

Very, very simple modification; around lines 155 in pid.c, just multiply rcCommand[axis] by two in both the ifdef GPS and else parts of the errorAngle calculation.

     #ifdef GPS
            errorAngle = (constrain( (rcCommand[axis]*2) + GPS_angle[axis], -((int) max_angle_inclination),
                    +max_angle_inclination) - inclination.raw[axis] + angleTrim->raw[axis]) / 10.0f; // 16 bits is ok here
    #else
            errorAngle = (constrain( (rcCommand[axis]*2), -((int) max_angle_inclination),
                    +max_angle_inclination) - inclination.raw[axis] + angleTrim->raw[axis]) / 10.0f; // 16 bits is ok here

This should be put into the code base asap.

MasterZap, 2015-08-18:
No, I think max inclination should be what you have at max stick. Having RC "rates" affecting this seems broken... it's not a rate, it's an angle.... So the 0-500 (or 0-whatever) the sticks give you should be scale d to 0-700 for a 70 degree setting. It's that simple, really...??

ctzsnooze, 2015-08-19:
MasterZap - no code has ever worked the way you describe, not back in MultiWii days either.  By that I mean, if you set max_angle to 250, stick sensitivity did not change, but the angle limit was 25 degrees. The stick sensitivity was always constant.  
The idea behind max_angle was never to reduce stick sensitivity - that is done with RC rates or in the Tx - but to stop the beginner flipping the quad and to limit the speed.
I suggest you try a limit of 250 (25 degrees), then adjust your sticks so that it requires full stick throw to get that 25 degrees.  You will find the quad un-flyable. It isn't good for beginners like that.

borisbstyle, 2015-08-19:
I remember in my beginner days on openpilot it was pretty much a nice
setting. There you always use a full stick in angle mode. The more advanced
you get on sticks the more sensitivity you could add by increasing max
angle.

borisbstyle, 2015-08-19:
btw that change is not trivial. You need to add an extra variable like
AngleRcCommand or something which is scaled from the rccommand and max
angle.

If 25 configured for example (ratio 1:2) full stick of 500 is divided by 2
in that case and for 25degrees.

borisbstyle, 2015-08-19:
or better a dynamic multiplier new variable:

float angleMultiplier;

angleMultiplier = max_angle / 500;

Now just replace your 2 by this calculated multiplier

ctzsnooze, 2015-08-19:
Well go fly both ideas everyone... Boris likes the low-sensitivity at the same with the low angle limit, I prefer the sensitivity fixed.  If at the end of the day we don't agree, just some people like one way and others like the other way, I guess we are looking at a new CLI value.  But please anyone who hasn't tried this idea please do so and provide feedback based on their experience in flight.

borisbstyle, 2015-08-19:
None of the methods would be fine for me :).
I fly rate mode and 10% of my stick input represents a roll rate of 12
degrees/sec. But on my small acro X quad it is more like 20deg/sec.

There really is no need to have those exactly scaled as most people use
expos etc. It is never linear.
I believe mw ancestors are the only ones not using the full stick
resolution.

In your case you are manipulating rccommand anyway. 10 means 20 and 25
means 50 etc.

ctzsnooze, 2015-08-19:
Ha ha true.  But if you do my simple change, you can switch from rate mode to level mode and probably fly it.  Without my change it is almost impossible to fly with max_angle = 500, it would be even worse if you needed all 100% stick to get 50 degree angle.

it is not possible to get a greater angle without manipulating rcCommand, because rcCommand itself is limited to 500 and that's the cause of the problem in Level mode.

borisbstyle, 2015-08-19:
Also the accuracy of angle modification is reduced in this code.
You now go from 0.1 degrees resolution to 0.2degrees.
A cameraman who plays with 1 degree differences for fine filming now has
more difficulty getting the right shot

MasterZap, 2015-08-19:
For sure the 2x is better than nothing - angle mode in LuxFloat is now kinda unflyable. However, I'm firmly in the camp that it should really work "like openpilot", which means, the "max angle" is really the angle at max stick, period. 

Having a max that makes big stick motions do nothing more than small motions makes very little sense to me..... having a "rate" multiplier affect an angle makes equeally little sense to me....

That's really what I liked about OpenPilot. Stuff was in real units. The rates in MultiWii/CF are random numbers. In OpenPilot it was 600 degrees/second.....

/Z

bdoiron74, 2015-08-19:
+1 to real units. I think it's pretty typical for angle mode to be stick == angle. How fast it gets from one angle to another is how fast you move the stick. Not sensitive enough at center, or too sensitive adjust expo. Separating AngleLimit[x,y] + AngleExpo[x,y] for angle mode and RateLimit[x,y] + RateExpo[x,y] for rate mode would be better.

borisbstyle, 2015-08-19:
for angle mode I agree, but openpilot was too static.
Configuring 600deg/sec meant you need to fiddle with your tx to get some
good expo.
This uses radians per second. 1 radion is 360deg/sec, but gets screwed up
by 2 different rate numbers.

MasterZap, 2015-08-22:
Yeah so what do the two rates do? Someone said "add up", someone said "multiply"... and it seems to depend on the PID controller chosen? Eww. I dunno? ANyway, consider this a vote for real units :)

closed 2017-02-25 by hydra