cleanflight/cleanflight #1075: Support for statically tilted rotors and cameras

dylnhdsn, 2015-06-30:
I'm making a feature request for a configuration value for camera angle and/or motor tilt to make control inputs match perspective for FPV flight of multirotors.

I've been messing around with different tilting styles to provide a fast forward level flight. I've found that tilting all the motor's in the same direction and applying an angle offset makes a multirotor fly as if the camera was tilted at the same degree.

However, the inputs in the case of tilted motors or tilted camera don't align with the perspective. if you yaw, you see a bit off roll and if you roll you see a bit of yaw (I think). I think most people have learned to account for the control differences, but I believe the FC could account for the perspective change and have the controls more closely match the perspective.

I'll will probably try to hack on this myself if no one is interested in doing it, but I am not a C developer, so it will take awhile for me to ramp up. :(

joshuabardwell, 2015-06-30:
Any time you have motors that aren't perpendicular to the intended direction of movement, you will get input coupling. For example, I had an A-tail quad whose arms were in an X configuration, and as a result, the rear motors were tilted slightly forward, in addition to tilting inward. When I yawed, I also got pushed forward, as if I had pitched forward. 

IMO, the first step to fixing this is to adjust your cmix as best as you can. For example, on my A-tail, I removed roll authority entirely from the rear motors to avoid roll/yaw coupling. On a quad with tilted motors, this won't be an option, since there aren't any non-tilted motors at all. The second step will be properly tuning the PID loop. If you are getting unwanted yaw when you roll, in theory, a perfectly tuned PID loop will detect and compensate for that. In reality, it's best to get rid of coupling via mechanical methods so that the PID loop has less work to do, but that's not always possible. If I had unwanted yaw when I rolled, I would look at yaw I gain to see if it was doing its job.

The real challenge, I think, would be dealing with translational effects--e.g. if you yaw, you translate forward instead of rotating around in a circle. I can't think of any possible way to avoid this if all motors are tilted the same direction, since there is simply nothing that can counteract the forward-thrusting vector that occurs when you spin up a tilted motor.

I guess what I'm wondering is what you intend to do that can't already be accomplished by the PID loop (unwanted coupling of axes) and/or that can't be accomplished at all (unwanted translational effects).

dylnhdsn, 2015-06-30:
Understand the case of your A tail. I had something similar with the front motors level and the back tilted 35 degrees. In that case, the PID controller handled the differences perfectly fine.

My case is a bit different. I have all motors tilted in the same direction and I've also set the pitch offset to match the motor tilt.  It works exactly as I expect it to in the situation with some coupling of roll and yaw from the perspective of my camera which is sitting level in the craft.

However, I believe it would be possible to decouple yaw and roll if the FC knew about the tilting and compensate by mixing input to make the craft act as though the FC and the camera are oriented the same.

I'm just arm-chairing here, but I believe it to be possible for the FC to do this because I can get what looks like a yaw from the camera's perspective by combining inputs. I'm doing it unconsciously and I can't currently describe what inputs I'm providing to do this. I plan on testing it and setting up mixes in my transmitter to achieve the effect that I'm talking about.

joshuabardwell, 2015-06-30:
>However, I believe it would be possible to decouple yaw and roll if the FC knew about the tilting and compensate by mixing input to make the craft act as though the FC and the camera are oriented the same.

I think I see what you're getting at. Not coupling because of the mixing of thrust vectors due to motor tilt, but coupling due to the gyro being aligned differently than the frame. But is that really coupling, or is the gyro just doing its job? You have told the FC that the board is tilted ten degrees when it really isn't. I think that the coupling you're seeing is no different than the coupling that you would see if the frame was tilted ten degrees forward. In other words, it's not actually coupling relative to the orientation axes of the flight controller, but it appears to be coupling relative to the horizon (or other external references).

If it's not immediately obvious what I mean, imagine you're pitched 45 degrees forward and then you yaw left 90 degrees. The yaw motion will occur around the Z axis of the FC, which is pitched 45 degrees forward. As a result, the copter will be rolled 45 degrees right after a 90 degree left yaw. This is 100% intentional and is how acro mode is supposed to fly; it's the only flight mode that's really possible based only on the gyro output.

If you add accelerometer input to the equation, you can implement an auto-coordinated yaw, in which the copter maintains its pitch and roll attitude while yawing around an imaginary Z axis that is perpendicular to the ground, rather than perpendicular to the FC. I think that APM's "drift mode" is similar to this, but not exactly. APM's drift mode attempts to auto-coordinate yaw, by adding an equivalent amount of roll. What I'm proposing is a mode where roll and pitch are maintained during yaw--in other words, pitch north 45 degrees, then yaw left 90 degrees, and be pitched west 45 degrees.

borisbstyle, 2015-06-30:
I always fly with tilted camera and also tilted motors.
I have 35degree cam tilt and 10degree motor tilt. I don't experience any problems you meantion there. I think I got used to it :D

fedorcomander, 2015-07-01:
@borisbstyle Do you do any config changes for titled motors, or its not required in rate mode at all?

borisbstyle, 2015-07-01:
@fedorcomander 
Yes I have my USB on the left side (90 degree yaw)....so I have to add -
[image]
10degree roll

fedorcomander, 2015-07-01:
@borisbstyle Thanks... I thought its only relevant for autolevel modes.

borisbstyle, 2015-07-01:
@fedorcomander 
Your gyro needs to know how to control the yaw axis for example :D. You now changed the board it's position to the frame.

joshuabardwell, 2015-07-01:
Hang on a sec. If I have a camera with 10 degrees up-tilt, and I lie to the FC and tell it that it's tilted back 10 degrees, will that cause the copter to roll and yaw about the camera axis instead of about the frame axis? I could just go out and test-fly, but it's raining all this week here.

Does the FC tilt offset setting interact with acc trim, or are they separate?

fedorcomander, 2015-07-01:
@borisbstyle -10 does the trick. Yaw is crisp now, like tilt is not even there ;)

joshuabardwell, 2015-07-01:
Just thinking this through... you still have to coordinate yaw, right? I mean, let's say you've got 10 degrees of up-tilt, but you're flying forward at 20 degrees of pitch. So that's effectively 10 degrees of tilt that you have to coordinate.

fedorcomander, 2015-07-01:
it just makes board yaw axis in the same plane as props. thats it. its enough.

joshuabardwell, 2015-07-01:
>it just makes board yaw axis in the same plane as props. thats it. its enough.

I'm more concerned about roll axis being coplanar with the camera.

borisbstyle, 2015-07-01:
@joshuabardwell
It is telling the fc the position related to the frame. When tilting motors you actually change the position of the frame.  This sounds a bit weird, but you have to think away the "old" frame and only think about a virtual frame which is in line of the motors. The thrust vector stays the same, which is vertical.
When you would enter this command without tilting motors it will also change thrust vectors and that can upset the fc and make your flight experience worse.

@fedorcomander
Also move the cg just a little bit forward with tilted motors and you  will be completaly in balance ;)
Tilted motors can make the rear motor work a bit harder. In my measurements just 5-8%. Moving cg tiny bit forward makes this unnoticable.
I love the acceleration I get from tilted motors :).

joshuabardwell, 2015-07-01:
To clarify, here's why I'm concerned about roll axis most. You're always going to be at different pitch angles depending on how fast you're flying. So you're always going to have to coordinate some amount of roll with your yaw. It seems like changing the FC's pitch orientation will change the zero-point where the quad requires no roll input to maintain a flat yaw turn, but any other amount of pitch will still require roll input to maintain a flat yaw turn, so it doesn't seem like there is a huge amount of benefit there. It just seems like a matter of taste as to where you want the zero point to be.

On the other hand, coordinating roll between the camera and the FC ... right now, let's say I do a move where I punch up, then pitch forward to point the camera straight down, then roll 180 degrees and pitch back as I descend so that I end up pointing the other direction from where I started. If I perform the 180 degree roll when the camera appears to be pointing straight down, in actuality, the frame's roll axis is pitched 10 degrees off the vertical. So when I finish the roll, my camera is looking 20 degrees off vertical. That's kind of okay since that's the direction I was going anyway, but it is a real challenge if I wanted to roll less than 180 degrees, in which case yaw input is required. If the copter rolled about the camera axis, a lot of this would sort itself out.

joshuabardwell, 2015-07-01:
>When you would enter this command without tilting motors it will also change thrust vectors and that can upset the fc and make your flight experience worse.

Ah well... here I thought I had found a solution. I guess I will just have to be a better pilot.

fedorcomander, 2015-07-01:
@borisbstyle: i noticed that 2 and 4 worked a bit harder indeed. not sure i can push lipo any further but thanks for the tip.

borisbstyle, 2015-07-01:
@joshuabardwell
No thats not the case with tilted. Different axis height actually doesn't make any difference. The motors will all travel the same distances on roll and pitch axis. 
I dont correct anything. I thini I even cant tell any difference anymore.
I actually started to fly tilted just to get more cam angle, but surprisingly I also gained more acceleration as well :). I can go full throttle low to the ground if I want without going up.

borisbstyle, 2015-07-01:
@fedorcomander
mine is just a little bit more forward. maybe 0.5-1cm and the rear motors work as hard as front.

joshuabardwell, 2015-07-02:
https://www.youtube.com/watch?v=uwF1vJ3j1U0

fedorcomander, 2015-07-02:
May be u need a craft with a gimbal ;) You are misaligning the board and trust vector of your craft. Bad idea imho.

borisbstyle, 2015-07-02:
Pitch and roll are actually not affected in rate/acro mode. No matter how tilted is or whatever it will feel the same. 
When you yaw now the FC will think that it's yawing improperly as the results of the motor reaction will not match the expectations. The motors still produce the same yaw directions.
I think it will fly, but it feel weird.

joshuabardwell, 2015-07-02:
@fedorcomander This is all hypothesis because it's raining buckets and I can't fly, but I don't think it matters at all what direction the FC is pointing. We can rotate the FC board 90 degrees to the left and nobody is concerned about the fact that we are swapping pitch and roll. When you tilt the FC board towards the front of the copter, you start mixing yaw into roll movements, but as long as the copter has adequate control authority, this shouldn't be a problem.

The idea that the thrust vector is misaligned... Since we haven't changed the layout of the motors, the thrust vector should still exactly the same regardless of how the FC is oriented. All we are doing is changing the relationship between the stick inputs and the pitch/roll/yaw outputs that are commanded by the FC. Even in an extreme case, where the board was mounted 90 degrees on its side... now the roll stick actually causes yaw, and the yaw stick actually causes roll. So, @borisbstyle the pilot's expectations will be subverted, but there is no reason the copter should fly badly. The copter should fly exactly the same as it did before, because the fundamental means of creating thrust has not changed at all. All that has changed is the direction that the vectors in the FC's internal matrix are pointing have rotated.

>I think it will fly, but it feel weird.

If it makes for better video and easier to keep the camera centered on the flight-path during extreme moves, I will get used to it. Once this weather clears, I will report back.

dylnhdsn, 2015-07-02:
@joshuabardwell I think you really understand what I was getting at when I made this issue. I just want the FC to compensate for the effect that you described in your video. I too believe that the FC should have the control authority to do this. I think the work is in defining the formula that would mix appropriately.

Your solution will probably work, but like @borisbstyle it will fly weird. It would be like adding 12 degree motor tilts and not setting that tilt configuration. You end up loosing yaw authority and start washing out on high speed turns.

I have a interim solution that works for my current needs. I've left the tilt configuration at 0 and I still have my motors tilted. The difference is that my front are tilted slightly less then my back to give me better yaw authority. Sadly, I believe a good amount of energy is wasted by motor spinning up when they don't need to, but at least some of that energy is providing me thrust forward.

joshuabardwell, 2015-07-02:
>It would be like adding 12 degree motor tilts and not setting that tilt configuration. You end up loosing yaw authority and start washing out on high speed turns.

This is all hypothesis until the weather lets me get a bird in the air, but I do love hashing things out in my head, so ... I really don't think that is the case. The key factor is that the motors are not tilted. _The actual thrust vectors that are being created by the motors have not changed._ All that has changed is the internal reference point that the gyro uses for its x/y/z axis. Imagine that you take a flat quadcopter and convert it to a V-tail. Now the thrust vectors being created by the rear motors include both vertical and horizontal components. So when the flight controller commands yaw, you will also get roll, and vice versa. In my case, none of that is going on. All that is happening is that, when the pilot commands 100% roll, the flight controller will actually output 90% roll + 5% pitch + 5% yaw, or something like that (I just made that up--don't over-analyze the numbers).

I anticipate that the change will be similar to the first time I up-tilted my camera. When the camera is level, the view from the goggles is mapped one-to-one with the FC's outputs. When the camera is up-tilted, the view from the camera diverges from the FC's outputs, and you need to put in different amounts of yaw and roll to coordinate your turns. If the copter is perfectly flat, you use 100% yaw and 0% roll to turn. If the copter was pitched 90 degrees forward so it was looking straight at the ground, you use 100% roll and 0% yaw to make turns. For the first few flights, I was "spinning out" on my turns because the copter looked like it was less tilted than it actually was, and I wasn't inputting enough roll. Then I got used to it and it was fine. So I anticipate that changing the board's tilt like this will make the copter fly more like if the camera wasn't up-tilted. But that's just a guess.

dylnhdsn, 2015-07-02:
The reason for the loss of yaw authority is simple. If you do as you describe, when it tries to yaw it will spin up opposing motors but they are now pointed slightly backwards. The back motors tilted this way will provide the proper yaw direction, however, the front motors will provide the wrong yaw. At some point, this overcomes the yaw authority provided by the torque of the motors and your yaw authority is basically gone.

The key to understanding this is to realize that you are now getting yaw from the angled thrust vector AND some from the torque of the motors. It's actually kind of nice to put some small motor tilts on the back motors because they provide more yaw authority and make yaw very crisp and precise.

joshuabardwell, 2015-07-02:
>If you do as you describe, when it tries to yaw it will spin up opposing motors but they are now pointed slightly backwards

Why would the motors be pointed backwards? The relationship between the motors' thrust angle and the camera's view angle hasn't changed at all. So my visual feedback as to what the copter is doing hasn't changed. Nothing has changed except the relationship between the stick inputs and the FC's outputs.

Imagine that I took a mode 2 transmitter and switched it to mode 4 (so that yaw and roll were swapped). This is exactly the same as if I mounted the FC board vertically and/or put in 90 degrees of roll adjustment in software. The motor angles haven't changed; the degree of coupling between axes hasn't changed. All that has changed is which stick I move to accomplish which axis of rotation.

dylnhdsn, 2015-07-02:
The FC is combining the tilt configuration with the motor mix, but your motors aren't in a flat configuration if it thinks that when it's tilted 12 degrees forward, it level. This is the equivalent to tilting the motors 12 degrees and not setting the tilt configuration. Would you agree?

If you do, visualize a tilted motor. It's thrust vector can be used to provide control authority. In the case that they all are tilted in the same direction, some of those motors are tilted in a way that takes the natural control authority away.

joshuabardwell, 2015-07-02:
The FC has no concept of level because I'm flying in acro mode.

joshuabardwell, 2015-07-03:
I was able to sneak in one test flight tonight as it was getting dark. There was only a tiny difference in how the flight felt; I had no problem coordinating turns. The rear end of the copter tended to lag the roll slightly at the beginning, and then overshoot slightly at the end, with a yaw-jump sounding bounce of the motor. My interpretation is that the FC is trying to do what it's being asked to do, but the yaw authority is insufficient to keep up even with the 12 degrees of mixing of yaw with roll. Possibly, the situation would work better with a V-tail or tricopter that had more yaw authority.

Another effect of the change was that yaw-turns were completely flat, whereas normally if I start from a centered horizon, a yaw turn requires roll coordination, since the copter is pitched forward when the horizon is flat. That's pretty nice, but not worth giving up having crisp rolls for.

Public link to BB log is here: https://dl.dropboxusercontent.com/u/23394347/LOG00069.TXT

MauroMombelli, 2015-07-07:
you can see the "yaw-roll" correction we are developing for dynamic tilting arm to see how we fix the input coupling

joshuabardwell, 2015-07-07:
@MauroMombelli link?

MauroMombelli, 2015-07-07:
see #691

ducktaperules, 2015-11-04:
i dont know if people are still looking at this but if inputs could be decoupled from the FC orientation and aligned to the cam orientation then i think this could be a useful feature. it would make switching between cam angles easier when you don't have to account for the yaw roll mixing manually. 
Moreover It could also enable modes where (within a certain angle range) yaw is always aligned with the horizon (regardless of FC tilt or cam angle) and roll would not have to be compensated at all whilst flying forward and cornering.

closed 2017-02-25 by hydra