cleanflight/cleanflight #1076: Airplane config with CC3d atom

Pedroalbuquerque, 2015-06-30:
I have load cleanflgight 1.9 to a CC3D Atom board an to a flip32 10Dof with airplane configuration.

on cleanflight GUI the radio input is correct (maps correctly radio stick movements)
Motor 1 is fine
Servo 1 - motor
servo 2 - elevator
servo 3 - rudder
servo 4 and 5 - aileron

The CC3D output are all massed up, 
out 1 has motor control
out 2 have a pwm signal but not moving
out 3 has ailerons
out 4 also airlerons
out 5 no signal (flat/constant low voltage)
out 6 no signal (flat)

Yet the same configuration on Flip board words just fine
out 1 - motor
out 2 - constante PWN but not mapped to any radio control
out 3 - aileron
out 4 - aileron
out 5 - rudder
out 6 - elevator

sppnk, 2015-09-03:
I am flying well two planes with 1.8.1 and CC3D. 

I tryied 1.9.0, and a experimental 1.10 flying wing, but I am not able to configure them, nor by the Chrome GUI, neither through the CLI.

I dont know if "smix" or "servo" commands have to be used for this (as GUI hasnt been fixed).

In 1.8.1 everything goes well.

Pedroalbuquerque, 2015-09-04:
Thanks for your comment sppnk, I will try with 1.8.1.

sppnk, 2015-09-04:
Hi Pedro, if you are flying airplanes should try https://github.com/digitalentity/cleanflight, so he is working in new navigation and RTH features for airplanes with GPS.

I am not able to test the last .bin for my CC3D, because these weird issues with 1.9 and 1.10. It is not possible for now use the configurator, but I have spent many hours trying to configure my airplane with the CLI and I couldnt, so I went back to 1.8.1.

I hope the problems could soon be solved, because the new NAV-REWRITE code from @digitalentity promises amazing things...

digitalentity, 2015-09-05:
@Pedroalbuquerque indeed my goal is to bring CleanFlight's navigation to a new level. However airplane support is not ready yet (working on it slowly). I mostly fly multicopters and aim to get things working on them first. If interested - keep an eye on this issue: https://github.com/digitalentity/cleanflight/issues/1

Pedroalbuquerque, 2015-09-06:
thanks,
my comment/note was about CC3D board output with simple acro and stabilizenot navigationMy test was ment to allow very basic plane stabilization, not navigation, but CC3D board code must have a bug, as I can make the code work perfectly on a naze board
so the mapping to output pins on the CC3d board code must have some bug
thansk for you reply anyway
Pedro Albuquerque

Date: Sat, 5 Sep 2015 16:12:55 -0700

sppnk, 2015-09-07:
@Pedroalbuquerq, in fact the mapping of the servo outputs in 1.10 for CC3D is buggy. Take a look to this (not working yet) dirty PR #1290.

sppnk, 2015-09-17:
Hi, I have setup airplane on CC3D through smix and mmix (CLI commands):

You have to use for now "mixer customairplane", like this:

mixer customairplane
mmix reset
mmix load airplane

smix reset
smix 0 2 0 100 0 0 100 0
smix 1 3 0 100 0 0 100 0
smix 2 4 2 100 0 0 100 0
smix 3 5 1 100 0 0 100 0

The smix command should be like this 

|COMMAND|RULE #|SERVO INDEX|SOURCE|RATE|SPEED|MIN|MAX|BOX|
|--------|-----------|-------|----------|----------|----------|-------------|----------|------------|-----------|

So, the matching of the outputs in CC3D is now:

|CC3D ATOM OUTPUT|CLI SMIX SERVO INDEX|
|------------|------------------|
|1|N.A. (is motor connection)|
|2|N.A. (not connecter or motor 2 AFAIK)|
|3|2 (is AIL)| 
|4|3 (AIL)|
|5|4 (RUD)|
|6|6(ELE)|

After the merge of PR #1283 I am now able to use servo command like this, so I can now reverse servos to match my airplane setup.

servo 3 1000 2000 1500 90 90 -100 -1 (reverse right aileron)
servo 5 1000 2000 1500 90 90 -100 -1 (reverse elevator)

Hope It helps. 

Please report if you or anybody is succesfull with CC3D and airplane. Thx.

digitalentity, 2015-10-22:
There is a PR with the fix #1425

closed 2017-02-25 by hydra

Pedroalbuquerque, 2017-04-12:
thanks sppnk, I will have a try

sppnk, 2017-04-18:
If you want to fly planes then go to the [inavflight code](https://github.com/iNavFlight/inav). It has had a huge advance since this topic was opnened.