cleanflight/cleanflight #1077: Update documentation for selective Software Filtering (Gyro, Dterm, Pterm)

digitalentity, 2015-07-01:
We now have cool software filtering #1028, but it does not have any docs on how it works and how to fine-tune it.

joshuabardwell, 2015-07-01:
What is the process for writing up documentation? I responded to a volunteer request for writing up PID tuning documentation over on the RCGroups forum. I asked for an offline email to help define parameters (formatting, style guide, etc...) and didn't get any response. I have professional technical writing experience, but I don't want to just dive in and put a lot of hours into writing when I'm sure there are guidelines in place that I won't be following.

borisbstyle, 2015-07-01:
It is wiki style on github. I can help you work with git if you want? 
Just add section advanced tuning to current pid tuning document or make another file and you can do a pr request.

if you have git installed:
fork a branch here through git web or clone it from git shell.
Than type in assuming you want to make a new branch forked from master called docu tuning:
git checkout -b docu_tuning

now add or change .md file in your repository publish and push it online and you can make a pr.

But manual can also be found here on git somewhere.

joshuabardwell, 2015-07-02:
Ok, I have downloaded Git desktop for Windows and cloned the repo. I can't sync changes, presumably because I don't have write access to the repo. So after I am done modifying the local files, I would submit a pull request?

borisbstyle, 2015-07-02:
Modify files..make a commit....publish those and it will be uploaded online.
You can make a pr after that.

joshuabardwell, 2015-07-02:
I think I see. I have to give the commit a summary and description before the "commit to master" button becomes available.

Should I commit only when I have changes worth sharing, or should I sort of commit as I go?

joshuabardwell, 2015-07-02:
@borisbstyle @hydra should the soft filter tuning be part of the PID tuning page, or a separate page?

borisbstyle, 2015-07-02:
you better should make a nee branch instead of changing master.
On that way you can always keep the master in sync in the future with the
original master.
If you are using desktop version. Click right mouse button on cleanflight
to het git cli. There you can make a new branch
git checkout -b <name of your branch>

that forks the master and you can change things without touching the master
itself

joshuabardwell, 2015-07-02:
Ok, please pardon my dumb: I have forked CF and cloned to my desktop. Is that the same as what you are talking about? 

Now that I am modifying my own fork, I can sync changes back to GitHub. But that's not the same as a commit.

Is there a 101 for this stuff that I should read?

borisbstyle, 2015-07-02:
I think there is a lot of git documentation around.

Quick explaination:
When you fork cleanflight there are different branches in there. There is
not just one version in there, but more and you can switch in between. Like
you already saw I also have a local branch which is yet another fork from
my master and my master is what I always sync with real cleanflight master.
So you have to create your own branch within your forked cleanflight.

joshuabardwell, 2015-07-02:
Okay, I have forked, cloned to local, then modified PID Tuning.md, then synced, but I don't see my changes in the online version of the file.

https://github.com/joshuabardwell/cleanflight/commits/master/docs/PID%20tuning.md

I need to commit before I will see them? Is that right?

borisbstyle, 2015-07-02:
yes you need to commit indeed.

jasonyandell, 2015-07-02:
@joshuabardwell first you commit locally.  Then push to your branch on GitHub.  Then create a Pull Request.  Then it should show up in this repo!

joshuabardwell, 2015-07-02:
Ohh, I see. Once I did the commit, then Sync showed "1 unsynced". I sync'ed and then it showed up in my branch.

Thanks, everybody.

joshuabardwell, 2015-07-03:
Okay, I have committed changes to my branch, which you all are probably clever enough to find without me even linking to it. I wrote the text with the assumption that 1.9.1 would include the recent PID-related changes that have been rolled into master, so I will wait until 1.9.1 is dropped to make the pull request. Until then, anybody who has comments is welcome to make them. Or... I don't know, branch it yourself and fork pull push commit put your thing down flip it and reverse it. Github.

borisbstyle, 2015-07-03:
[quote]Yaw Rate

In PID Controllers 0 and 5, it acts as a PID reduction as explained above. In PID Controllers 1-4, it acts as a stick sensitivity multiplier (linearly additive to RC rate), as explained above.[/quote]
This part isnt entirely correct as RC rate doesn't affect yaw rate at all. Yaw rate is actually more less the only yaw rate parameter.

joshuabardwell, 2015-07-03:
>This part isnt entirely correct as RC rate doesn't affect yaw rate at all. Yaw rate is actually more less the only yaw rate parameter.

I actually wondered about that, but that text was from a previous rev. so I just left it. I'll change it. Thanks.

borisbstyle, 2015-07-03:
I know, but time to correct it :)

closed 2017-02-25 by hydra