cleanflight/cleanflight #1083: emf_avoidance = 1 messes up virtual USB port on CC3D

Arakon, 2015-07-04:
I'm not even sure this is a bug or simply unavoidable with the hardware.
As soon as emf_avoidance = 1 is set on the CC3D, the USB port stops working (windows complains about an unrecognized device). Connecting via serial is still possible just fine, so the issue is seemingly in the timing of the USB port.
I know that strictly, the feature is not needed, but a few extra Mhz could help, especially with Boris' new filtering.

hydra, 2015-07-07:
the USB clock initialisation would also have to be changed to match the system clock.  Please investigate if you can.

Arakon, 2015-07-07:
Sorry, I'm no coder.. I can't really touch this, I'm afraid.

ProDrone, 2015-07-07:
The USB needs a clock of 48 MHz and has a prescaler that can only divide by 1 or 1.5. The overclocked frequency cannot be divide to have the required 48 MHz. So the USB part is also overclocked and that causes problems. So IMHO there is no solution for this.

thenickdude, 2015-07-08:
In that case we should update the docs for emf avoidance to mention this drawback.

closed 2017-02-25 by hydra