cleanflight/cleanflight #1085: gyro_lpf should be a profile- instead of a master variable

ProDrone, 2015-07-06:
Reason: `gyro_lpf` is now used as a integral part of the selective software filtering, for which all other parameters belong to a profile. I think it is dangerous when switching to a profile that has default settings for the selective software filters, after gyro_lpf has been set to 256.

borisbstyle, 2015-07-07:
I am not sure about this one. I still see this as a general parameter and the soft filters as flexible parameters. But either way it could be fine.

borisbstyle, 2015-07-07:
https://github.com/cleanflight/cleanflight/pull/1093

ProDrone, 2015-07-07:
It looks like you have given it some more thought and agree :grinning:

borisbstyle, 2015-07-07:
@ProDrone I don't know. Been thinking about it for a while and some others asked me that. I think it's great to show them the code in case they want to use it for testing with filters.

ProDrone, 2015-07-07:
I guess people who ask for this also think about what happens when they switch to a profile with defaults and `gyro_lpf = 256` We have to wait and see what @hydra thinks...

Arakon, 2015-07-08:
I'd actually like to see it as a part of the profile too. I'm still messing around a lot with the PIDs and filters, and trying to run a profile with filters off is not doable unless I also have my cellphone and a wiring mess with me to switch the gyro_lpf from 256 to 42.

digitalentity, 2015-07-08:
@Arakon, gyro_lpf is used *only once* when initializing gyro on boot. Imagine switching a profile and forgetting to reboot so new gyro_lpf would be applied. Unless gyro could be re-initialized without reboot (and since we can switch profiles in flight this shoud take a fraction of a second) moving gyro_lpf to profile is a bad idea IMO.

ledvinap, 2015-07-08:
@digitalentity : It's quite trivial to write LPF switch function and add it to gyro driver ...

digitalentity, 2015-07-08:
@ledvinap Yes it is. It also means that PR to move gyro_lpf to profile must also add this functionality for each gyro driver.

borisbstyle, 2015-07-08:
@digitalentity 
Once you move gyro_lpf to profile you might as well move a lots of other things or even entire config.
Think about looptime, acc filtering parameters and many more. 

My thought about profile settings is that it needs to be minimalized as much as possible. I do agree it's nice for testing things in the field, but the main "product" is to be used and not so much for testing purposes.

ProDrone, 2015-07-08:
I would prefer to group everything that changes (flight) behavior into a profile (and have a copy_profile function). Things that are bound to your hardware belong in master.

More on topic: @borisbstyle don't you think it is useful for tuning to be able to switch between profiles with different gyro_lpf values?

Note: Originally i wanted to prevent the possibility to switch to a non-working profile, but on second thought i will take that back, since users are always capable of configuring such a profile.

ctzsnooze, 2015-07-08:
Main reason is that with low loop times and the filters, we want 256 all the time, not going to use anything else in any profile.

hydra, 2015-07-10:
yes i think it would be useful for testing different gyro settings in the field.  but really how difficult is it to do it in the field anyway.  no very.  i find my self agreeing with @borisbstyle and @ctzsnooze.

i also think that there is merit in being able to change the gyro filter while in-flight - think autotune.

To this end, i think i would accept code which can change the filter of the gyro on a profile switch as long as the solution is simple and is implemented for all gyro drivers.

a unit test would be nice too.

ProDrone, 2015-07-19:
The in-flight autotune option may be nice, but will add to code and EEPROM size for functionality that may be used only a few times. Also not sure how the gyro's behaves when switching the LPF in-flight.

I think it is best to close this.

closed 2015-07-19 by ProDrone