cleanflight/cleanflight #1096: The buzzer sounds mean.

yuan0226, 2015-07-08:
Hi
    I know cleanflight code buzzer different names represent different meanings, but I want to know what the meaning of specific representatives, to write the appropriate documents or forms to me?         Thank you.

thenickdude, 2015-07-08:
On the "CLI" tab of the Configurator, type "play_sound" lots of times. It will play each of the sounds so you can hear what each one sounds like. The list of sounds is here:

https://github.com/cleanflight/cleanflight/blob/master/docs/Buzzer.md

I don't think there is currently written documentation about what each beep sounds like.

yuan0226, 2015-07-08:
I have read this document, this document I can only know from the text called on behalf of each of the buzzer corresponding meaning, not from the correct discrimination between different sounds mean.

Pierre-A, 2015-09-11:
Buzzer sound meanings are here :
https://github.com/cleanflight/cleanflight/blob/master/src/main/io/beeper.c
and sure should be added to the Buzzer.md document.

closed 2017-02-25 by hydra