cleanflight/cleanflight #1097: uint32_t for sonar variable unneeded

bk79, 2015-07-08:
Why do we have the need of using a uint32 when the maximum allowable sonar range is not exceeding 300 400 cm?

hydra, 2015-07-08:
it's quite common to use datatypes that are effecient for the target platform, though I would prefer the compiler to sort that out with code generation.  Submit a PR that fixes it please.

ledvinap, 2015-07-08:
320m range (int16) is not that much for distance sensor ...

bk79, 2015-07-08:
@ledvinap 300 CM is the acceptable limit for valid ranges on HC-SR04 out of that sensor has no meaning that's why it's only used for low altitude control where baro is almost useless with a precision of 0.25m, if you want higher range move for laser sensor

ledvinap, 2015-07-08:
@bk79 : Yes, but when I add laser/radar/etc it will be much easier to use existing framework - and that means same return type as for sonar ...

bk79, 2015-07-08:
@ledvinap uint16_t can go up to 65535 cm I don't know what is the scope of measuring in cm over 50m...probably we're talking about two different scope: sonar for me is used for precison landing , baro switches over after that

ledvinap, 2015-07-08:
uint16_t won't allow -1 as N/A (spacial value can solve this ... but why)

In my view of CF, sonar is simply relatively accurate distance measurement sensor (that may need tilt compensation). Laser rangefinder may have identical interface.

Resolution is important if you need derivative - Z speed in this case.

Also barometer has cm as output units ...

bk79, 2015-07-09:
0 is also a not valid value for any range finder won't have 0 values, agreed that resolution is very important...but really 65535 cm means 655.35 M any laser finder I've seen is no going above 300m that means 30000cm also a int16 would suit. IMO int32 is a waste of resource, but feel free to close the issue if you think it's worth

bk79, 2015-07-26:
Closing the issue I've converted i2c Nav board sonar reading into 32bit as well BTW I've created a smooth code reading for sonar that should clean a lot of noise

closed 2015-07-26 by bk79