cleanflight/cleanflight #1099: Invert Buzzer functionality CLI option

ctzsnooze, 2015-07-09:
I don't know how to author a pull request but this is what I'd like to achieve....

When using the Buzzer pin to drive large clearly visible LEDs for visibility in low light, the intent is the opposite of when that buzzer pin is used to make a noise.  At night you want the inverse behaviour = lights on normally, flashing off when there's a problem.

I figured out the changes required and would be very grateful if someone could help author a pull request to make them happen:

Around lines 130, change the default for beeper on/off:

    // Beeper off = 0 Beeper on = 1
    //static uint8_t beeperIsOn = 0;  **** default
    static uint8_t beeperIsOn = 1;

Around 310:

    // BEEP_ON; **** default
    BEEP_OFF;

And likewise a bit later around 325:

    //BEEP_OFF;  **** default
    BEEP_ON;  

These are the only points where the code needs reversal, so if say BUZZER_INVERT = 1 in the CLI, the opposite behaviour would apply.

This would not need to be profile based.

Some beeper sound sequences are best changed a little when using them with LEDs.  They were designed for buzzers, with reasonably long 'on' times to ensure the beeps are heard.  But for LEDs we only need brief flashes off to communicate (say) low battery states, otherwise we can't see anything in the dark for the critical alarm, which must buzz all the time!  :-)

A more thorough solution, matching the LED orientation of the reversed state, would also change some of the buzzer sound patterns.  For instance for low and critical battery I did this:

    // low battery beep
    static const uint8_t beep_lowBatteryBeep[] = {
    // 25, 50, BEEPER_COMMAND_STOP   ***default
    100, 4, BEEPER_COMMAND_STOP

(a single very brief flash off every second when low)

    // critical battery beep
    static const uint8_t beep_critBatteryBeep[] = {
    //50, 2, BEEPER_COMMAND_STOP  ***default
    26, 4, 10, 4, BEEPER_COMMAND_STOP

(two very brief flashes when battery is critical, repeated 3 times every second)

Obviously these patterns could change when the user selected BUZZER_INVERT = 1 in the CLI

thenickdude, 2015-07-09:
drivers/sound_beeper.c actually implements a beeper inversion switch already, you could probably tack on to that system.

ctzsnooze, 2015-07-09:
Thanks Nick... I looked in sound_beeper.cI couldn't find anything in the CLI to let me set the isInverted beeperHardware(config), i.e. at line 69:

    initBeeperHardware(config);
    if (config->isInverted)
        systemBeepPtr = beepInverted;
    else
        systemBeepPtr = beepNormal;
    BEEP_OFF;

Nor could I see anywhere in io/beeper.c where beepInverted or beepNormal did anything.

Do you mean that the beepInverted / beepNormal options should somehow be set in the CLI, and then used in io/beeper.c should respond to them?

Pierre-A, 2015-09-12:
**@ctzsnooze** : this is fixed in main.c (see lines 268-290) depending on the target and the buzzer output driver through the BEEPER_INVERTED parameter. 
Personally, I prefer adding an external transistor and have buzzer **and** lights simultaneously activated. So, lights are continuously on and flash when buzzer is activated.

ctzsnooze, 2015-09-13:
Thanks Pierre - now I get it! 

So... if the relevant target.h file has no line #define BEEPER_INVERTED, then the beeper won't be inverted.  So for Naze, which doesn't have the line, I just add that line and the beeper will be inverted.  That's so simple.  I just didn't know how to do it.

Yes if I had a buzzer I'd do exactly what you do.  Makes so much sense.

Thanks so much!  - C

Pierre-A, 2015-09-13:
Yes for the first part but the Naze board is a specific case as seen in lines 281-286 of main.c. 
For newer revision (5 and above) the *isInverted* parameter is *true* due to a NPN driving transistor (2N3904) on the buzzer output (previously was OD for a 2N3906). 
So, you should only fix it to *false* instead of *true* on line 285 and it should be fine.

Pierre-A, 2015-09-14:
**ctzsnooze**, I tested the buzzer output inversion and it works fine on a Naze rev5. As I said, just fix the *isInverted* parameter to *false* on line 285 in *main.c*.

ctzsnooze, 2015-09-17:
Thanks very helpful!
PS - I personally changed the beeper tones for low battery, finding the standard tones a bit confusing.  I think in the end I did a double flash/beep for low and a run of four fast flashes for critical.

Pierre-A, 2015-09-17:
You're welcome !
I'm completing the file *docs/buzzer.md* for a better description of the beeper tones (PR#1312).

handsomejackuk, 2015-11-10:
can anyone confirm is this command changeable through CLI... i have tried some pnp transistors with no luck, and have loads of npn.. just curious if this can be done.. i have cc3d...

ctzsnooze, 2015-11-11:
Regrettably, as far as I know, there is no CLI option to invert the buzzer functionality - as far as I know,  a custom code build is needed.

It would be a really useful feature, I think.

To drive LEDs from buzzer, I use an IRLML2502 SMD FET, ground the source pin and connect the gate with a tiny wire to the CPU side of the buzzer pin; a current limiting resistor might be nice.  Connect the positive side of your LEDs  to battery positive, then the negative wire from the LED's via a current limiting resistor to the drain on the FET.  Test by temporarily touching the gate to say the LED drive pin, your LEDs should flash on when the main LEDs flash on.

Pierre-A, 2015-11-12:
//**@ctzsnooze**, 
//**@handsomejackuk**, just in case you were nor aware of this last PR #1482

Pierre-A, 2016-01-02:
**@ctzsnooze**, **@handsomejackuk**, just in case ..
I closed PR #1482 and #1607 and rebased the initial proposal in new PR #1645.

paddytfm, 2016-02-22:
Hi everybody
I have tried to invert beeper state using set beeper_inverted in last naze32 2016 version but the cdt is not recognised. How to make it work?
Any reply would be appreciated
Paddy

closed 2017-02-25 by hydra