cleanflight/cleanflight #1100: Default Luxfloat PID's way too high

borisbstyle, 2015-07-09:
I think this should definately be addressed. The default PID's are just very high on any setup I tried at the moment. This is actually even a potential danger as most setups can even jump in the air due to extremally overtuned float pids.

joshuabardwell, 2015-07-09:
Seconded. Given how bad Lux acts when P is too high, it should definitely err on the side of being a bit low. Every Luxfloat tune I've seen seems to have P between 1.0 and 2.5.

borisbstyle, 2015-07-09:
https://github.com/cleanflight/cleanflight/pull/1103

swilkens, 2015-07-14:
For what it's worth, I'm flying on 3.1 P.

borisbstyle, 2015-07-14:
@swilkens 
Do you have any footage of your flights on 3.1? I was able to go as high as that on pitch, but once I go full throttle it will oscillate.
I tested it even on my very underpowered little test quad. That's where you expect to get the highest proportional values.

swilkens, 2015-07-14:
@borisbstyle 
No videos, it's a pure LoS. I'll see about producing some logs and get back to this.

borisbstyle, 2015-07-14:
3S or 4S? The reason I ask is because I can get it up to 3's, but the motor sound can become noisy, while there are no visible oscillations.

swilkens, 2015-07-14:
@borisbstyle 
3S. I forgot to mention that LuxFloat had some bad vibrations on the default PID tuning in combination with the default looptime. Once I dropped the looptime down to 2000 all was well. I've been down to 1500 / 1200 but that didn't really improve things.

The only other adjustment I did was bump up the acc_lpf_factor for angle mode, but that shouldn't influence acro mode - right? She's fine in acro - I'll try and get a video of a punchout once my new props come in.

borisbstyle, 2015-07-14:
@swilkens 
That makes sense. Once you go lower with looptime the looptime variations/jitter gets worse and that affects PID's. Even if luxfloat is scaling to looptime.
Video is actually only usefull with gopro on it. Other cameras can't deal with the noise on the quad and everything sounds the same :(

fakatwork, 2015-07-22:
I'm using Lux on a Arris X-Speed 250 frame currently (the one with dual layered arms) and the default PID parameters were indeed not flyable out of the box (crazy oscillations, had to do an emergency landing) I had to lower pretty much everything to start tuning it (don't have my figures in head unfortunately, could report later if needed)

I think we all agree the defaults are too high for 250 quads, but we need to have more feedback from people using larger aircrafts.

bdoiron74, 2015-07-22:
At silly low looptimes, it may not just be that loop time variation gets worse (but that is part of the problem). Gyro samples are produced (by the mpu60xx at least) at 1Khz with the default gyro_lpf and others unless you disable it to produce them at 8Khz. 

looptime should be locked to some multiple of this frequency (1000us, 2000us, 3000us...). If the gyro is outputting at 8kHz and your looptime isn't (get me?) you're losing values. If it's 1kHz and your looptime is below 1000 you're using values twice, at gyro_lpf = 42Hz @ 1kHz and looptime 2000 you're losing every 2nd value but it's smoothed so the accumulated error is small. 

Since we're using a varying cycletime in the calculations and jittery sub-sampling of a fixed frequency signal, it's no wonder tightening up these loops can cause shakey results with filtering layers being developed to counter the issues. RC inputs are a whole other kettle of fish since they are updated slowly in comparison and also not synchronous.

swilkens, 2015-07-22:
That might explain why my 1500 looptime is stable with relatively high P terms. I'll try dropping the looptime silly low to evaluate the real world effects in 1.9.0 - no new filtering.

fakatwork, 2015-07-23:
@borisbstyle : that's precisely the area I'd like to contribute to, but if I'm not mistaken @ledvinap is already working on it in his local branch. Even if Lux, Harakiri and Rewrite are cycletime-aware and thus more immune to loop jitter, they all would benefit from a more stable and regular "heartbeat". For the other PIDC it's even more important...(but should we care about them if they get eventually removed in the end is another question....)

So I know this is absolutely not the right place to discuss this topic, but I'd like to engage a discussion around that. As you noted, it would be best if the loop was tied to the gyro sampling rate (or multiples of) both for the reasons you cited AND for the aforementioned PIDC performance enhancement (the more regularly they are called, the more stable they get)

Maybe #1053 is the correct place to talk about it?

bdoiron74, 2015-07-23:
Agreed - #1053 is the place for IMU sync discussion.

swilkens, 2015-07-31:
Just to add to my previous comments:

Since my last comment, I've switched to DL mode on the same ESC which has forced me to dramatically lower the PIDs on LuxFloat. It seems the slower response time of my previous setup (no DL or COMP_PWM) allowed for the default LuxFloat PIDs (~3) to be functional.

That is far from the case with DL enabled. Quad was unflyable and had to be re-tuned dramatically.

This makes sense. If the quad is able to respond quicker, lower values of the PID terms are required to stabilize the system and achieve proper damping. 

So in short: defaults are ballpark for non-DL / COMP_PWM setups. Far too high for DL setups.

dgordon, 2015-08-01:
I don't know if this is useful to this discussion but, here is my config, setup, and a video... I like the lux but, think there is some very fast oscillations in flight. I think it can be seen during playback of video when looking at distant objects (tree line, or etc : 1:25 in linked video)

I am also under the current belief the P is at its max with this setup but, I'm still rather green.

**setup**: qav, 3s 1300 60c lumenier, 2300 lumenier, gemfan 5030

**tune**:
"PIDs": [
        [1.5,0.045,12],
        [2.5,0.05,20],
        [3.5,0.045,50],
        [5,0,0],
        [0.15,0,0],
        [3.4,0.14,0.053],
        [2.5,0.33,0.083],
        [5,0.03,75],
        [4,0,0],
        [12,0.045,1]
      ],
      "RC": {
        "RC_RATE": 1.07,
        "RC_EXPO": 0.35,
        "roll_pitch_rate": 0,
        "roll_rate": 0.8,
        "pitch_rate": 0.8,
        "yaw_rate": 0.4,
        "dynamic_THR_PID": 0,
        "throttle_MID": 0.4,
        "throttle_EXPO": 0,
        "dynamic_THR_breakpoint": 1500,
        "RC_YAW_EXPO": 0.35
      }

**looptime** : 1200
yaw_deadband = 3 *<-- fix yaw drift*

https://www.dropbox.com/s/x14cm7m1jzea37d/vid124-residential.mp4?dl=0

I'm a *noob* but, want to help if I can.

borisbstyle, 2015-08-01:
your pids look too high from what I can see. Did you try to lower them?

dgordon, 2015-08-01:
Haha.... To answer your question. Yes, I've had many tunings on controllers 1 and 2. Like I mentioned before. I'm stilling learning how to tune and trying to figure out what 'feels good'. 

I don't want to hijack this discussion on tuning my quad . But if I can be of help trying to figure out where the 'base line' is for PID2. I can keep posting PID settings and videos of my various tunes.

I'll certainly checkout the the result of dropping some of my PID values.

borisbstyle, 2015-10-21:
solved

closed 2015-10-21 by borisbstyle

inciteman, 2015-11-03:
Has anyone installed Cleanflight on a CC3D Board with the Arris X-Speed 250 Frame?

fakatwork, 2015-11-03:
I had a Arris X-Speed previously but with a Naze not a CC3D. Why? Is this a general support request or is it related to this precise issue?

brannan, 2016-01-11:
Is anybody still interested in this? I put a Naze32, built from GitHub checkout on 2015-01-05, on a wood h-quad that weights about 1kg and has a wheelbase of 460mm. The default PIDs were way, way too high for me as well. Yaw turned out to be the biggest change. I can post results later today if anybody cares.