cleanflight/cleanflight #1102: GUI Sensor Section - Kinematic

mehmetyldz87, 2015-07-09:
Dear Hydra, 

Could you add Kinematic Values [Roll , Pitch, Yaw (deg )] into Sensor Section.

mehmetyldz87, 2015-07-15:
It will be useful for understand the board directions and also for test bench PID tunning

jaapp, 2015-07-15:
You should look into the Blackbox features. Much more useful.

mehmetyldz87, 2015-07-15:
Dear jaap,

Also we can use logging section , but Angle can be seen simultaneously

jaapp, 2015-07-15:
I know these values are already available via MSP, which tools like EZ-GUI Ground Station can pick up an display.

mehmetyldz87, 2015-07-28:
I am trying to obtain step response of my system on a test bench , so I apply some test and logged data, before star the test I want to measure the angle via GUI .

thenickdude, 2015-07-31:
@mehmetyldz87 if you use Blackbox to record your test data, it will give you angle information from the IMU simultaneously (use the `--simulate-imu` command-line option to `blackbox_decode`)

closed 2015-07-31 by thenickdude