cleanflight/cleanflight #1105: max_angle_inclination = 900 flight not stable

cfischer9009, 2015-07-10:
I am building a rocket ship for nationals here in the US and am having a horrible time getting it tuned for racing.
Here is the quad I'm working on.
http://www.rcgroups.com/forums/showthread.php?t=2454440

We have it shooting around no problem in rate mode but I want to race in angle with 90 degrees max inclination. 

I have scoured the internet trying to find info on tuning or understanding how angle mode works. I believe it is a really big deal for racing.

I feel like the issues we are having are hardware related. As if the acc cant actually perform under aggressive movements or high angles. There are many symptoms. First there is lots of slipping out of attitude and lots of correcting needed from the pilot. Second is the inconsistent angle the quad will reach. Sometimes its 90. Sometimes its much less. Sometimes its more than 90 and it tumbles all over itself. Third is the worst, I will maneuver a roll hard and immediately after the center stick is now reset at some percent of that roll and full stick the other way barely keeps me level. Its as if the acc resets during the maneuver. This will wear off after a few seconds of flying level. 

Reading up on the controllers it seems that the acc is polled or averaged after some time and used as a check against the current attitude. Is this at all correct?

We as a team have experienced this on multiple quads now all running the afromini which has the same hardware as the naze32. 
This is on 4s 6" 6045 sunnysky 2300kv @ about 400g AUW so lots of power to weight.
The issue is less so on 3s.
Any help would be greatly appreciated.

henn1001, 2015-07-10:
I´m not sure if this is addressed in cleanflight but problems at 90° may be related to [Gimbal Locks](https://www.youtube.com/watch?v=zc8b2Jo7mno)

cfischer9009, 2015-07-10:
Thats interesting. Feels like a possible fit to my problems. I wonder if that is the cause then maybe we could rotate the origin of the loss of the degree of freedom to a pitch angle behind the quad, allowing the same system to work with at least 90 in forward flight.

digitalentity, 2015-07-10:
@cfischer9009 or move to quaternion math for defining orientation.

cfischer9009, 2015-07-10:
^ is that part of the pid controller? Do we have a choice for this currently?

digitalentity, 2015-07-10:
@cfischer9009 no, that's quite alot of coding. I was suggesting the way of solving a gimbal lock problem.

thenickdude, 2015-07-10:
@cfischer9009 I think the issue that you're having is that CF figures out which way is down by looking at the accelerometer. When the accelerometer's reading about 1G of acceleration, CF assumes that direction must be downwards. However, if you're simultaneously rolling and freefalling, the accelerometer could measure 1G of acceleration outwards, making CF think that that direction is "down".

I can't think of a good way of fixing this apart from increasing `gyro_cmpf_factor` from the default of 600 towards the maximum of 1000. At the high end of this value, the accelerometer has less influence on the attitude calculation, instead the gyroscopes are replied upon.

cfischer9009, 2015-07-10:
This is great feedback! I will try this today. Is this independent of PID controllers? Are there PID controllers that process the acc differently, possibly leaving one or two as a better option for this? Thanks so much for the feedback guys!

-Chris

thenickdude, 2015-07-10:
This is independent of the PID controller, it tunes the IMU which is the
code which estimates the craft's attitude using gyroscope and accelerometer
data.

The PID controllers don't address the accelerometer directly, but rather
just use the attitude that the IMU calculates.

ctzsnooze, 2015-07-16:
I always set gyro_cmpf_factor = 1000 and acc_smooth = 0, this was necessary to stop noise aliasing in acc wrecking the level hover point.  

See this long discussion on how attitude is maintained in CF, #791 it took me a long time to understand how the levelling process works.

Also an independently people are reporting max_angle_inclination is being ignored #1072

ctzsnooze, 2015-07-20:
I've identified the cause of the bug that limits max angle in Level mode to 50 degrees.  

Not sure how best to fix it.  Would be grateful if Boris or any other coders could propose a robust solution to what I think has been a long-standing bug affecting most controllers   :-)

Please see:   https://github.com/cleanflight/cleanflight/issues/1072

closed 2017-02-25 by hydra