cleanflight/cleanflight #1106: Remove MSP telemetry.

hydra, 2015-07-10:
Does anyone use MSP telemetry?  I've a feeling no-one does.  I also feel that LTM #1010 would be better instead.

Considering removing it for 1.11.

digitalentity, 2015-07-10:
I second this, LTM and/or MAVLink are better options for telemetry. MAVLink protocol has two-way support, LTM is usable at very low bitrates.

Arakon, 2015-07-10:
I think quite a few people use mwosd with cleanflight, and I believe that does use msp?

misakm, 2015-07-10:
I use mwosd but not sure about communication protocol.

digitalentity, 2015-07-10:
@Arakon MSP protocol is not going anywhere. mwosd is using polling (bi-directional communication) to get required data from flight controller. MSP telemetry is uni-directional stream of data - probably not used by anybody. mwosd users should be fine if MSP telemetry is removed.

Arakon, 2015-07-10:
Ah, great.

sppnk, 2015-08-28:
I think many of the multiwii pilots, using mwosd are swapping the osd modules bettween models, as they are compatible with CF. 

I use the android EZIO apk to comunicate with the plane, (with BT or using openlrsng receiver). I dont know if mavlink is better, but AFAIK, MSP is an interesting protocol that is not dead.

digitalentity, 2015-08-28:
Nobody is going to remove msp protocol, this issue is about removing msp
telemetry that is never used even by OSDs - they use polling.

tracernz, 2015-08-30:
I'm pretty certain MWOSD can listen to MSP telemetry as I have used it in the past... In fact it definitely can https://github.com/ShikOfTheRa/scarab-osd/blob/master/MW_OSD/Serial.ino#L721

Boomer91, 2015-09-01:
I'm using the unidirectional MSP telemetry to drive arduino based ledstrip animations using speed etc. What alternatives can I use if this would be deleted?

thenickdude, 2015-09-01:
Bidirectional MSP.

commit 1e098bf1 (2015-12-14): Delete MSP telemetry. Saves ~350 bytes of code space on F1 targets and
reduces ram usage. MSP telemetry was not used by many people.  MSP is
not considered a good telemetry protocol - it is bandwidth heavy when
compared to LTM, Mavlink, etc.  Closes #1106.
closed 2015-12-14 by hydra