cleanflight/cleanflight #1110: Derivative Kick

borisbstyle, 2015-07-10:
I opened this issue to continue discussing theories about preventing the "Derivative Kick" on RcCommand handling from some other PR's.

This is what we are talking about. Slower sampling of Rcdata is causing "infinity" in derivative. This is even filtered data, which smooths it out a bit already!
[image]

This is currently not an issue, rather than a cosmetical thing in my opinion. There are no negative effects in flight, but it would be neater to find a solution where we can smooth it out a bit even if it is preventing ramping a bit to prevent huge rises in D. We don't want to add too much delay preferably

borisbstyle, 2015-07-10:
@ledvinap 
Let's continue here.

I try different filter approaches in octave from my real data.

ledvinap, 2015-07-10:
Can you please link corresponding .LOG file?

borisbstyle, 2015-07-10:
it's added. It's a random log I have. It is visible in any existing log where quick sharp stick movements occure like flips and rolls
https://dl.dropboxusercontent.com/u/31537757/cleanflight/blackbox_many_flights.TXT

ledvinap, 2015-07-10:
Here is graph with only gyro signal filtered:
[image]

Not only D-term but also P term is strongly affected by RC setpoint. I'll definitely try RC filter on RC input to PID, with trying few different cutoffs. 20Hz may be good candidate for 20Hz update rate - last trace

borisbstyle, 2015-07-10:
I made a few graphs as well today and pretty much any filter is better than now :dancer: 

I will start with 25HZ and see how it goes. Thanks for thinking.

borisbstyle, 2015-07-10:
This is what I will test later today. Just a quick implementation for poll, pitch and yaw.
https://github.com/borisbstyle/cleanflight/commit/82ef24a5b67ba04b076a45841f4bfa20ef1d0269

borisbstyle, 2015-07-10:
Ok just modified and tested. Works like a charm! Don't notice a difference on the stick and the graph is smooth!
Off to more testing.

borisbstyle, 2015-07-10:
Some screens of testing. Derivative is still sensitive on the edges.

no smoothing
[image]

22hz
[image]

15hz
[image]

borisbstyle, 2015-07-10:
I guess for more effect I need to go even lower.

ledvinap, 2015-07-11:
Maybe 10Hz, but filtering delay starts to get non-negligible (research would be necessary, quick googling:  in FPS games < 100ms is great, 200ms ok, 300ms almost unplayable).

What about testing FIR (averaging) filter?

borisbstyle, 2015-07-11:
Yeah sure can you provide some working FIR averaging code and I will try it
out.

I still have to do some more testing with delays. It was too dark yesterday
to do some fpv flying. Thats where you also have some additional delay from
the video equipment as well.

ledvinap, 2015-07-11:
Delay on RC channel is in 'human' loop - perception delay/reaction time is added to it. Human brain is extremely good at compensating this delay. My wild guess is that anything under 50ms has negligible effect (but I am quite sure there are purists that can tell 1ms difference .. and connect their TX with Monster ethernet cables ...)

borisbstyle, 2015-07-11:
yes....it all comes down to what you are used to.
I work in telecom industry and I have made some satellite solutions in the
past for voice and data.

You even can adapt having voice conversations up to 1sec delay once you get
used to it.

But there are guys out there who chose to fly with serialrx as they say the
feel the difference.

ledvinap, 2015-07-11:
Maybe it would be worth implementing blind-testing into firmware - choose tested parameter randomly, make notes during flights and compare it to blackbox log afterward .. placebo effect works great ...

joshuabardwell, 2015-07-11:
>But there are guys out there who chose to fly with serialrx as they say the
feel the difference.

If serialRX has a higher update frequency, then rcCommand should have more discrete steps, therefore derivative kick should be reduced. Maybe that's what they're feeling.

Zestforlife808, 2015-07-11:
I did some relatively competitive FPS gaming prior to turning to Mini Quads.  I can tell you there is most definitely a perceptible difference between even 50 ms and 10 ms.  There is an ever evolving line of gaming monitors and input peripherals aimed at reducing latency because of this.  If you were to play any modern FPS game with a mouse with 20 ms of input lag, you would have a bad time.  Similarly, it's common knowledge that a person playing with 10 ping has an enormous advantage against someone with 100 ping.  

I don't know how this translates to Quads where inputs are processed and limited by the hardware capabilities of the quad itself.. but I fully agree with those who claim for example that they feel the difference between sbus and ppm.

borisbstyle, 2015-07-11:
@joshuabardwell 
Faster update will hae more steps and more kicks ;)

@Zestforlife808 
The rc filter only adds 1-3ms delay. That is not really noticaeble at all.
But I even felt the difference between effio FPV cam and sony 600tvl. The difference is like 50ms and huge difference in flight.

borisbstyle, 2015-07-11:
12hz seemed fine
[image]

@ledvinap 
I tried your suggested mod applied to rcData, but that gave me huge delay. Not sure what went wrong and didn't have much time to debug

ledvinap, 2015-07-12:
@borisbstyle : 

The code using rcCommand is OK. Using rcData would be more complicated.

Did you try to disable D-term averaging? 

Is there some difference in energy consumption?

borisbstyle, 2015-07-13:
@ledvinap Not yet. There was too much rain and I didn't get to test it all properly. My test list got long lately. I will stop changing other things to complete my test list.
My initial impression is that it didn't make much difference, but needs more serious testing. I will measure motor temperature.....do onboard 120fps recordings and also analyze sound of the motors from the recordings together with logs once I get to it.
My energy consuption was much better yesterday anyway, but I also lowered the gyro_cut_hz and was also using luxfloat.....so it's no real comparisent yet.

I will also test level mode where the stick input is much larger. That's where I expect bigger differences.

olelille, 2015-07-13:
Isn't filtering the RC input related to reducing the PID factors: 
https://en.wikipedia.org/wiki/Step_response

What about using dynamic PID gain profile based on RC input ?

Qopter, 2015-07-13:
Hi Boris,

maybe I can help you. I already have a solution to minimize the derivative kicks. The best solution is to interpolate the steps of rccommand. This can be easily done by dividing the large step by (repetition cycle of rcdata / looptime) and add this small step each looptime cycle.

Here my code snippet running at start:

	if (mcfg.looptime > 1200) {
		maxfactor = (40000 / mcfg.looptime + 1) / 2;
	}
	
Here my code snippet running in mw.c each looptime cycle:

	static int8_t factor;
	static int16_t lastcommand[2] = { 0, 0 };
	static int16_t deltaRC[2] = { 0, 0 };
	int16_t setpoint[2];
	
	if (isRXdataNew) {
		deltaRC[ROLL] = rcCommand[ROLL] - lastcommand[ROLL];
		deltaRC[PITCH] = rcCommand[PITCH] - lastcommand[PITCH];
		lastcommand[ROLL] = rcCommand[ROLL];
		lastcommand[PITCH] = rcCommand[PITCH];
		isRXdataNew = false;
		factor = maxfactor - 1;
	} else {
		factor--;
	}
	// interpolate steps of rcdata
	if (factor > 0) {
		setpoint[ROLL] = lastcommand[ROLL] - deltaRC[ROLL] * factor/maxfactor;
		setpoint[PITCH] = lastcommand[PITCH] - deltaRC[PITCH] * factor/maxfactor;
	} else {
		setpoint[ROLL] = rcCommand[ROLL];
		setpoint[PITCH] = rcCommand[PITCH];
		factor = 0;
	}

Unfortunately I don’t use cleanflight. I use my own software based on baseflight. So you have to modify my code that it fits to cleanflight. But I think it’s not a real problem for you. 

This interpolation is similar to a FIR filter. It adds a delay of half a cycletime (10ms for PPM). You can compensate this additional delay by reducing the moving average on rcdata (4 -> 3).

Hope this helps.

borisbstyle, 2015-07-13:
@Qopter 
Thanks I will definately try it once I prove that the derivative kick has negative influence.
That should fit in the code. I don't see declaration of "isRXdataNew", but I get the picture.

ledvinap, 2015-07-13:
@borisbstyle : Here is FIR implementation, but not yet finished/tested:
https://github.com/ledvinap/cleanflight/commit/cf5b4cdf59acf25f2268d6d7347633eb9bf0ab6d

Longer filter in filterRCconfig would be probably necessary  and initialization should be added ...

Qopter, 2015-07-13:
@borisbstyle
"isRXdataNew" is a boolean which ist set true every time rcdata is updated.

borisbstyle, 2015-07-13:
@Qopter 
I just tried your approach on my bench and it totally smooths out the rcData.

May I ask how I should interprate this maxfactor? And why are you limiting it to 1200 looptime?
           maxfactor = (40000 / mcfg.looptime + 1) / 2;

borisbstyle, 2015-07-13:
Here quick flight on my basher LOS quad outside. 
https://dl.dropboxusercontent.com/u/31537757/cleanflight/RC_smooth.TXT

@ledvinap 
Can you have a look in my local if thi could be a good candidate for removing averaging. I am now filtering rcData and this should also prevent problematic behaviour on bad RX frames etc.
https://github.com/borisbstyle/cleanflight/commit/74f5854f257f8485947119e0fa0ac5cfb886dcd8

Qopter, 2015-07-13:
@borisbstyle
maxfactor is the number of intermediate steps. The formula is a simple rounding for integer. I limited it to 1200 looptime because my naze32 can not run faster. By now there are faster FCs on the market so you should limit it to a lower value. But you have to prevent for too low values. Set the default to a medium number, for example maxfactor = 16.

borisbstyle, 2015-07-13:
@Qopter 
Yes I figured that out. You can run faster when disabling accelerometer. My naze can go down to 300.
I protected that part in case looptime is set to 0. It uses cycleTime to get some maxnumber.
It works really well. Thanks for helping out.

ledvinap, 2015-07-13:
@borisbstyle : maxfactor is number of loop periods to to transition from old to new RC value (+-1). I don't know why its rounded to even value ...
with 40000 this is 20ms, standard 50Hz PPM frame time.
This works great if actual frame time is greater or equal to expected value. With shorted frame time, there will be residual step when new data arrives.

Personally I would prefer FIR approach - its mathematically simpler/cleaner and easier to encapsulate/hide implementation details. But there is overhead, all filter taps need multiplication, my very rough estimate is 2% CPU time for 20 tap at 1kHz (* 3 RC channels) ... Good for testing, but production code may need some optimization ...

borisbstyle, 2015-07-13:
I saw your code, but this approach seems more efficient and the graphs look very smooth a you see in blackbox data.

Perhaps the amount of factoring would need to be more dynamic? Maybe some timing built in.
Also there should be statement for maxfactor selecting between serial and analog to assign correct factoring for least delay.

ledvinap, 2015-07-13:
@borisbstyle : It may be enough to use last interpolated value when computing delta - that way ramp will go from last position to newly received value, avoiding discontinuity if smoothing time is too large.
Then simply hardcode reasonable smoothing value for each receiver type, it won't hurt if the value is not spot on.

ledvinap, 2015-07-13:
@Qopter: Is there any reason for / 2 in  maxfactor calculation ?

ctzsnooze, 2015-07-14:
The ideal filter for these kick type spikes is a linear moving average, not exponential moving average, and that's exactly the result here with qopter's code.  

Even if the moving average is not the whole PPM width, but just say half i.e. 10ms, the size of the spikes will still be markedly reduced probably to insignificance, and there will be a bit less delay.   

Perhaps the code could set the duration of the average to see if it is PPM or SBUS input and choose a timeframe to suit?  Or if the user can set it, ideally in ms, they might use 10-15 for ppm 3-6 for sbus, code might adapt that setting for loop time....

Qopter, 2015-07-14:
@ledvinap
The reason for / 2 in maxfactor calculation is rounding. You can use <pre><code>maxfactor = 20000 / mcfg.looptime;</code></pre> without rounding. I got better results with rounding therefore I implemented it.

borisbstyle, 2015-07-14:
yes I will check for different receiver types. I actually believe there is only 10hz from softserial and 50hz from ppm. Just do this check:
          if (isRxDataDriven) 
                  10hz
           else 50hz

I will also rethink how to better apply timing for those cases when looptime isn't stable. I think the best idea is to average a few looptimes all the time to have the most accurate timing. I think when we average 5 loops and use that for maxfactor would be most suitable and covers any scenario.
That would also prevent smoothing time being too large as well.

Also some extra conditional protection for PWM/PPM
If there is a failsafe kill switch or disarm on PWM/PPM something more would need to be done perhaps like filtering those 2 additional channels as well. 

Another thing I would also consider is integrating it into annexCode and rework the logic there to prevent repeating calculations for rcCommand to make the code more efficient. A lot of things are happening in annexCode what don't need to run each cycle.

ledvinap, 2015-07-14:
@Qopter: Thanks, i overlooked that ...

ledvinap, 2015-07-14:
@borisbstyle : sbus is 11ms between frames ans is data driven ...

borisbstyle, 2015-07-14:
@ledvinap 
Thanks.

borisbstyle, 2015-07-14:
@ledvinap Where in the code can you find that SBUS is being refreshed more frequently than other data driven channels? I see just 2 definitions in rx.c

ledvinap, 2015-07-14:
rxDataReceived is set on each serial frame.

Qopter, 2015-07-14:
I would suggest to count the loop cycles during a RX frame. This gives you the number of intermediate steps (maxfactor). The problem could be, that the last frame will be measured and that number will be applied to the next frame. If there is a jitter it could be worse than with hard coded RX frame length. Otherwise if the jitter is less than 10% you can cut down maxfactor by 10% without getting great drawbacks.

borisbstyle, 2015-07-14:
To prevent jitter effect I will remove the multiplier by 2 for rounding. That will truncate some jitter. It is already average from 5 loops now constantly updated. I can also indeed add a little 10% tolerance.

ledvinap, 2015-07-14:
IMO there is no need for maxfactor to be precise. Just change code so that ramp is generated from last position:
```
if(factor > 0)  factor --;
for(axis) {
  if (isRXdataNew) {
    deltaRC[axis] = rcCommand[axis] -  (lastcommand[axis] - deltaRC[axis] * factor / maxfactor);
    lastcommand[axis] = rcCommand[axis];
    factor = maxfactor - 1;
  }
  // interpolate steps of rcdata
  setpoint[axis] = lastcommand[axis] - deltaRC[axis] * factor/maxfactor;
}
isRXdataNew = false;

```

If maxfactor is larger than rcData period, setpoint will smoothly change slope so that it follows new target (new rcCommand). Only penalty is small delay.
If maxfactor is smaller, then interpolation will be qual to `lastcommand` until new data arrive. Derivative will fall to zero in this interval, it will cause step in D-term, I'am not sure how important it will be ... but it should be quite easy to make sure that maxfactor is large enough ...

borisbstyle, 2015-07-14:
What do you suggest as a maxfactor choice for any looptime configuration?

One time 5 looptime average? on that way you will always get approximatelly good value unregarding of what configuration is used. 
If I give it a tolerance of let's say 30% you make sure that it's not changed just like that?

ledvinap, 2015-07-14:
I would still calculate maxfactor from (configured) looptime, using selected RC input to select appropriate delay and add some reserve (30%?). For SBUS (9ms frame update on frSky) that would be about 10-12ms (and rounding up when calculation maxfactor). Some googling should supply values for other supported receivers.

But now it is probably best to use CLI parameter for maxfactor, so that this whole idea can be tested.

borisbstyle, 2015-07-14:
Well my goal is to get the least delay possible for all, but I will add both for testing. When cli maxfactor disabled dynamic setting derived from cycletime or looptime and otherwise it will use the cli setting.

hydra, 2015-07-14:
Would suggest thinking of a slightly better name than maxFactor.  Its a bit vague.  Looking forward to reading about your testing results and comparisons too.

ledvinap, 2015-07-14:
rcRampPoints?

borisbstyle, 2015-07-14:
I already formalized some of them in my code over here. 
rcInterpolationFactor ???

borisbstyle, 2015-07-14:
Draft code
https://github.com/borisbstyle/cleanflight/commit/e77beb6feb6bb2dbe73953997d33a8bce7ab0094

Zestforlife808, 2015-07-14:
@borisbstyle 

I downloaded the naze hex from your dropbox yesterday and didn't notice any shark fins on the dterm while using luxfloat.  Is smoothing enabled on that version?

borisbstyle, 2015-07-14:
Ok did some cleanups in the code and retested.
@ledvinap I also tried your suggested approach, but that somehow didn't end well.

Final trace from my mini h after a lot of testing with my small mini x.
https://dl.dropboxusercontent.com/u/31537757/cleanflight/luxfloat_minih.TXT

I also found out that this works extremally well with these filters. Incredibly locked in
gyro_lpf = 0
gyro_cut_hz = 60
pterm_cut_hz = 40
dterm_cut_hz = 15

borisbstyle, 2015-07-14:
By the way I am getting impression that this definately improves tuning and flying characteristics. I now also think that derivative kick wasn't usefull at all as we first thought. Derivative kick was actually lowering response as it was working in the opposite direction than P

ctzsnooze, 2015-07-14:
Boris, filtering RC does not remove derivative kick. I have tried to explain this in the past, I try again.

Let's look just at pterm trace.  It is error based and has no dterm kick, but it has sawtooth, because the quad responds within the PPM period and the error is reduced as the quad turns.  If we smooth RC so the input PPM change is not a step, but a straight line, we we lose the abrupt change in error, so no more sawtooth on pterm and the gyro follows smoothly.

So the RC filter removes saw-toothing in quick-responding quads.

In Dterm when targeting error we also get saw-tooth, but here it is due to the sudden change in derivative and is due to kick.  Now let's say that for some filtering reason, maybe smoothing in RC, we have a kick 100 units high spread over 4 cycles.  The area under that curve represents an overall change in Dterm.  Now if we smooth it to be 50 units high over 8 cycles, we same have the same 'effective' change in Dterm.  The effect of the derivative kick on Dterm over time does not change.  If we make it 1 unit high over 400 cycles, the effect is the same.  

If we smooth RC so that the derivative kicks are no longer visible, the 'energy' in the derivative kick remains.  It is just 'invisible' now.  It has the same effect on Dterm as before, but it is just not visible anymore.  

You will notice that if you go back to original lux dterm behaviour (not target error) but retain RC filtering, the dterm curves will be different from the new error targeting method, proving that the effect of the kick remains.

Another way of thinking about this is that derivative kick is due to the setpoint change.  At some rate, when the derivative is determined by error, every time you change setpoint, whether the change is fast or slow, there is a change in derivative (either abrupt or gradual) and that change ends up in dterm.  

Let's be clear for once and for all, smoothing RC does not remove the beneficial effect of derivative kick.  It only smooths it out, like it smooths out the sawtooth in pterm.

ctzsnooze, 2015-07-14:
PS that mini-H trace is pretty amazing, isn't it!  How good is that!  :-)

The gyro trace looks almost precisely the same as the stick trace - like the quad is EXACTLY following the stick input.  And there is no yaw pterm sawtooth in the motors anymore, in fact no sawtooth in the motors anywhere.  

Its kind of curious that there is wobble in pterm and motors on sharp inputs, but the gyro trace is so perfect and does not show those wobbles.  I can't really explain why the motors must wobble to make the gyro trace so rock solid.  Anyhow the gyro trace is the proof of the outcome I guess and it is - perfect!

Did the motors sound different?  I think that sometimes on hard inputs they used to make a kind of harmonic overtone noise  that I rather liked... maybe now they just sound boring???  ha ha  :-)

PS - A person on RCgroups asked if the gyro recordings in BlackBox are affected by the sw filters, I think they are only affected by gyro_lpf?

Zestforlife808, 2015-07-15:
I tried the latest hex if anyone wants to see my results.  This is just some light flying LOS with a 2 pack tune.  Haven't had a chance to do any flips or rolls.  There's some slight bounce on sharp roll inputs but I don't want to damp the stick feel excessively.      

https://www.dropbox.com/s/8mwq1es34oiwmu9/LOG00239.fixed.TXT?dl=0

2015-07-13 @ 16:38:40 -- Running firmware released on: Jul 13 2015 21:49:19

cobra 2300kv 
5x45 
KISS
blackout mini h

gyro_lpf = 0
gyro_cut_hz = 60
pterm_cut_hz = 40
dterm_cut_hz 15

luxfloat 2
looptime 600
roll 1.0 .030 22
Pitch 2.3 .040 30
yaw 3.6 .060 15

borisbstyle, 2015-07-15:
@ctzsnooze
Well lets say it doesn't remove the energy, but it wastes more enegy over the same period. The motors are going to try to deaccelerate and accelerate, but the energy during the kick will only be wasted as motors will never be able to keep up with that.
At the end the total consumed energy for the same move will be more.

The short mini h flight was actually a LOS flight outside in the dark. The roll is completaly mistimed as I was flying on LEDs.
Still need to do a final tune, but it feels nice and responsive and it is easy to keep the quad on same spot.
The yaw was the most dirty axis and rc smoothing seemed to help it together with lowered vut frequencies the motor sound is pretty much like on the bench.

The roll/flip sound also got more silent. The violent sound of hard flips and rolls is more gentle and quite.

Another thing:
PID1 always seem to have weird level mode. I never could get a grip in it. The problem was basically that it could be tuned to be super snappy and stable, but as soon as you would make sharp stick moves it would oscillate.
The level mode uses bigger stick movementd and that causes more ramping in normal operation. That is fixed now as well.

borisbstyle, 2015-07-15:
@Zestforlife808
If it slightly bounces on roll you would need more D I would say.
Can you also try the latest hex so we can compare?

borisbstyle, 2015-07-15:
@ctzsnooze
gyro traces in blackbox are affected by both. gyro_lpf and gyro_cut_hz

[8 more replies]

closed 2015-10-21 by borisbstyle