cleanflight/cleanflight #1116: Sonar connected - altitude sticks to 0 as soon as sonar is out of range

kaefert, 2015-07-11:
I've got a sonar connected to PWM port 7 & 8 (enabled PPM input)

My sonar looks like its kind of working when connected to the cleanflight configurator and looking at the sensors page.
The problem is, that my sonar will say the distance is 0 if the ground is out of range - and the barometer will be overwritten by this 0 it seems. Therefore on my OSD when I fly higher than 2 meter I will always only see 0 meter altitude.

hydra, 2015-07-13:
that's currently how it works.  no your sonar is not faulty.  the code does ignore invalid readings from the sensor.

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/altitudehold.c#L268-L277

closed 2015-07-13 by hydra

kaefert, 2015-07-13:
hmm. well, either I did not understand your message correctly or there is some inconsistency here.
With my sonar activated, my MinimOSD will always show me a flight height of 0meters, unless I'm close to the ground I sometimes get readings of 2 meters, and often readings of 1 meter.

If I understand you and the code block you referenced correctly, this should not happen (and also from a user standpoint this behaviour is of course not desireable).

So the question is, why does my quadcopter behave like that?
There are two possible explanations I can think of:

1.) My sonar does not really give 0 (or smaller) but something a little bit over 0 when it does not see the ground anymore. - so that the codeblock `sonarAlt > 0` does match and the sonar is used instead of the barometer.

2.) The code does not really do what it was meant to do or at least not what it looks to me like it was meant to do.

skaman82, 2015-07-16:
I've noticed this behavior also when I played with the sonar function. The altitude shown in the OSD is then always 0. It should display the baro altitude if the sonar is out of range. Can someone take a look at this? Why is this issue closed? Looks like a bug to me.

kaefert, 2015-07-22:
@hydra could you take another look at this issue please? I think it has been closed mistakenly due to an misunderstanding between the two of us.