cleanflight/cleanflight #1117: Vibrations or High-G Maneuvers cause Throttle Cut / Disarm

AustinSaintAubin, 2015-07-13:
Airplane |  Configurator: 0.65.0 | CLFL, version: 1.9.0 | Board: AFNA, version: 1
When making high-g maneuver or airframe experiences vibrations System seems to disarm cutting throttle which is unable to be rearmed until plane is grounded.

This has resulted in several crashes during windy conditions.
Cannot find any documentation on how to disable or tune what seems to be an automatic disarm when the flight controller "detects a crash".

thenickdude, 2015-07-13:
There isn't such an auto-disarm system. Either your flight controller has a shaky power connection or maybe you're executing a disarm stick command without realising it? (low throttle, full left yaw). You can rule out the accidental stick disarm by assigning an arming switch to that function instead.

joshuabardwell, 2015-07-13:
That was my first thought to. An airframe that goes into failsafe during vibrations likely has a faulty connection either on the throttle pin (for PWM input) or on the signal pin (for PPM input). Another explanation would be a loose power or ground lead going to the receiver.

The described condition could also be the result of a loose power connection to the Naze itself, since the Naze shutting down would look like a zero-throttle failsafe. After the crash, the copter would come back to life, but wouldn't allow itself to rearm until the gyros were stable and could be calibrated.

Another possible explanation might be an under-specced battery experiencing voltage sag under heavy draw, resulting in brownout to the receiver or Naze. This is somewhat less likely, IMO, but not impossible.

SteveAmor, 2015-07-13:
Check it's not going into failsafe due to extremes of stick throw causing out of range reading.

ledvinap, 2015-07-13:
Blackbox log would be great to diagnose the problem ...

hydra, 2015-07-13:
If it's a motor stall desync issue try raising minthrottle.  I think this is a non-issue and likely mis configuration.

Check your stick channel endpoints go from exactly 1000-2000. (+/- 10)

Please provide a blackbox log and screenshots of the receiver tab when the sticks are at full throw and minimum throw.

AustinSaintAubin, 2015-07-15:
After working with the issue for the last few days.
Looks like it was an issue with the ESCs BEC overheating because of the amount of power being pulled by the servos during high frequency oscillations. The ESC was performing a thermal shutdown cutting power momentarily. This power cycled the flight controller thereby bring the motor to a disarmed state as part of the startup sequence for the flight controler.

Dont have a OpenLogger or Flight Control with BlackBox functions onboard. Tested my theory on ESC under different power loads. The more load and heat on the ESCs BEC, the more frequent the occurrence of my "disarm" issue.

Thanks for all your guys input. I need to get an [OpenLog](https://www.sparkfun.com/products/9530).

closed 2015-07-15 by AustinSaintAubin