cleanflight/cleanflight #1118: Three beep beep beep sound after power mean?

yuan0226, 2015-07-13:
Hi,I would like to know a beep beep beep sound after power is what it meant, thank you!

thenickdude, 2015-07-13:
Three beeps means that the gyroscope calibration has completed successfully. It's important that the quadcopter stay still on the ground until these beeps sound so that the gyroscope calibration isn't thrown off (if you move it substantially during calibration, then it'll automatically start calibration again and the three beeps will be delayed).

yuan0226, 2015-07-13:
Thank you for your reply, I understand.

hydra, 2015-07-13:
A documentatiob update with this info would be useful

joshuabardwell, 2015-07-13:
Done.

https://github.com/joshuabardwell/cleanflight/pull/1

hydra, 2015-07-13:
The PR is against the wrong fork, I will merge anyway.

closed 2015-07-13 by hydra

joshuabardwell, 2015-07-13:
Sorry. I'm still figuring out Git. I thought I was supposed to make a new fork for my changes, vs. branching the master. Is that not right?