cleanflight/cleanflight #1121: The voltage value of PWM1 ~ 6 of NAZE32

yuan0226, 2015-07-14:
Hi
I would like to ask a question, voltage values are initialized at power after 1.3V NAZE32 PWM1 ~ 6, and this is 1.3V RMS or what? Thank you.

JohnieBraaf, 2015-07-17:
Seems pretty much identical to CJMCU hardware, sharing the schematics would be a first step I guess.

yuan0226, 2015-07-20:
[image]
[image]

JohnieBraaf, 2015-07-20:
Pretty nice that is has an integrated Bluetooth module

yuan0226, 2015-07-21:
First, thank you for the compliment. Piece Bluetooth chip is used in conjunction with mobile communications, it is available through Android phones APP control flight control. But our flight control code that is not very good, not very stable flight, and can not be completed at the stunts.

hydra, 2015-08-20:
@yuan0226 Hi

Can you provide the developers with sample hardware?

Do you have links to the product so that people can buy the boards?

Are they in production right now?

If yes to all then yes, adding support should be possible.  It seems like a nice platform for developers to test indoors.

Do all the the processors in the boards have 128K or 64K?

goebish, 2016-02-23:
https://github.com/goebish/cleanflight/tree/crazepony
https://github.com/cleanflight/cleanflight/compare/master...goebish:crazepony

chr0n1x, 2016-03-05:
@goebish this is AWESOME. Thank you!

Were you planning on opening a PR anytime soon?

goebish, 2016-03-05:
Yes, meanwhile you can build it from sources or follow instructions in the rcg thread.

closed 2017-02-25 by hydra