cleanflight/cleanflight #1123: Is a HMC5883L != HMC5883 ?

oxivanisher, 2015-07-14:
Hi There

I tried using a Ublox Neo-7M with cleanflight (naze32) and i have no luck on the Z axes of the magnetometer. I even let hobbyking me sent a replacement for the module and tried both of them with a arduino to read the I2C bus. The conclusion I came to, was that it's possible that the two sensors do not work the same way. Can anyone please enlighten me?

In cleanflight i only see movement on the Y and X axes. I also set mag_hardware to 2.

I only have a arduino for testing, sadly no oscilloscope.
With this https://learn.adafruit.com/adafruit-hmc5883l-breakout-triple-axis-magnetometer-compass-sensor/wiring-and-test I get no Z axis movement like in cleanflight and with this http://shop.boxtec.ch/pub/iteadstudio/LIB_IM130918001.zip I get a heading (sometimes it is sending "0", but only for 1 tick). I am really confused. It is also possible, that i just got two deffective Ublox Neo-7M from hobbyking...

oxivanisher, 2015-07-14:
I know that answering and not editing is probably not the best way, but i think i found the root of the problem by testing multiple arduino libraries with this chip. Sadly, I am not able to fix it by myself (my C ninja is way too bad):

I think in [compass_hmc5883l.c](cleanflight/cleanflight/blob/master/src/main/drivers/compass_hmc5883l.c) on line 310+ (void hmc5883lRead) we have probably to add a fix like [HMC5883L_processing.ino](jarzebski/Arduino-HMC5883L/blob/master/HMC5883L_processing/HMC5883L_processing.ino) line 72+.

I googled several hours to try to make sense of the behavior and it seems to be a more or less known problem of some ICs..

The working library is located here: https://github.com/jarzebski/Arduino-HMC5883L
One of the "not working" is located here: https://learn.adafruit.com/adafruit-hmc5883l-breakout-triple-axis-magnetometer-compass-sensor/wiring-and-test

I hope this explains my problem more.

timkostka, 2015-07-30:
It's a bit unclear what the actual issue is here, but a short look at the datasheets for the two definitely shows they are different.  In particular, the allowed supply voltages are different and the pinout can be different.  The output rate is also different which may be causing some timing issues.  Just use the proper chip.

oxivanisher, 2015-07-30:
Thanks for looking into this.

* I use the chip the library file is named after (HMC5883L)
* I use a "officially supported" [U-blox Neo-7M w/Compass](https://github.com/cleanflight/cleanflight/blob/master/docs/Gps.md)
* The voltage and pinout can not be the problem or else I could not get any data
* I can say nothing to the output rate, but i get most of the values (correctly) just some headings seem to be wrong. this is why i think a fix like this could resolve the problem [jarzebski -> HMC5883L_processing.ino](https://github.com/jarzebski/Arduino-HMC5883L/blob/master/HMC5883L_processing/HMC5883L_processing.ino):
```arduino
  // Fix HMC5883L issue with angles
  float fixedHeadingDegrees;
 
  if (headingDegrees >= 1 && headingDegrees < 240)
  {
    fixedHeadingDegrees = map(headingDegrees, 0, 239, 0, 179);
  } else
  if (headingDegrees >= 240)
  {
    fixedHeadingDegrees = map(headingDegrees, 240, 360, 180, 360);
  }
```

My conclusion is, that I am using a proper chip. But the chip factory might have different versions internally of this chip as discussed in [this thread](http://forum.arduino.cc/index.php?topic=55159.0). As I mentioned before, i already tried it with multiple units, so that it should not be because i have a defective one.

timkostka, 2015-07-31:
How is that module wired up to your FC?  Did you solder the compass wires to the SDA/SCL pads?

oxivanisher, 2015-07-31:
Yes, i soldered it to the SDA/SCL pads. I soldered a cable with two connectors on there so i could also connect the baro. But i don't matter if i have the baro connected or not. And because i have the same problem on the Arduino, it is most likely not the way i hooked it up which generates the problem.

Yes, i saw the same symptoms. Thats why i initially hooked it up to the arduino in the first place. I have no screenshot handy at the moment, but the "red line" did show almost no change if i rotated it. But this cloud also be, because the problem already shows up on the calibration and so the graphical representation might be wrong from the start. But I surely can make a screenshot and post it here later.

That is also what i think, it might even be better because it is far away from any power lines this way.

I have a not finished [wiki page for the batbone](http://rc.oxi.ch/index.php/Flite_Test_Bat_Bone#GPS_und_Magnetometer) (german), maybe there is something for you to see on these pictures.

timkostka, 2015-08-01:
Well, it definitely sounds like you just have a bad chip then.  

As far as I know, the X Y Z values reported in the configurator are raw values from the sensor, so the calibration should not affect them.  If one of those isn't changing during rotation, that's a bad chip.  They certainly all respond correctly on my quad.

I don't think the "// Fix HMC5883L issue with angles" code you show would actually fix the problem either.  A magnetometer needs 3 axis to be working correctly.  If only 2 are working, it's just not going to do the job well.  No amount of code is going to fix that.

oxivanisher, 2015-08-01:
This is what bugs me ... i have a second U-blox Neo-7M w/Compass and it behaves exactly the same. I would like to know if there is someone with a working U-blox Neo-7M w/Compass mag on the current cleanflight release.

xqoperator, 2015-08-07:
I also have problems with HMC5883L breakout board I wired to my CC3D board (along with BMP180 which works ok)

The issue is that I am getting very noisy readings from the compass.
Setup tab shows a quad randomly rotating in yaw axis, heading jumps randomly 360 deg.
Sensors tab shows magnetometer reading jumping around -30..30 range which was surprising at first because chrome app divides the 16bit integer by 1090 and later by 100 to get fixed point two decimal places value before displaying. value of 30 would mean cleanflight sent a 'RAW' value of approx 3,000,000 which is beyond 16bit range...
Let me know if I missed something here.

Also it is not true that cleanflight reports raw values read from HMC5883L chip because immediately after they're read they get multiplied by magGain which (probably) cannot be debugged without modifying & reflashing. Later magADC is fed to alignSensors and magZero is applied. Only after that the magADC is available through MSP_RAW_IMU response.

Next step for me will be to connect the same sensor to my RPi which gave me no such issues when I first tried another IMU + custom python code.
I will also flash cleanflight modified to actually output magGain somewhere.
HMC5883L saturation detection code checks for values below -4096 while datasheet says return values will be in -2048..2048 range and I'm not sure this is correct saturation test.

let me know what you think

oxivanisher, 2015-08-07:
Finally I have good news (at least for me):
My magnetometer is now working as expected. This explanation might help other people with the same problem, or help them getting on the right track.

First my problems / fails:
* I did not understand how a magnetometer works. It is not like a gyro or accelerometer and it is absolutely normal that the Z axis does not return a curve on rotation. HobbyKing does not know that either btw. The heading gets calculated from the X and Y sin with the `atan2f` function in [src/main/flight/imu.c](https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/imu.c).
* I have as it seems a magnetic field on my floor in the living room ... yes I was able to prove that with the compass in my phone. It could be that my neighbors a floor down have a strong transformer in their ceiling lamp or something like this (just a wild guess).

My problem was finally cleared with the new knowledge that it is very important how the magnetometer is aligned (it can be flipped also). A read up on the function of the magnetometer is very helpful at this time! Then, these jumps cleanflight showed in the GUI suddenly made sense. Pleas read up on the `align_mag` variable in [docs/Cli.md](https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md) from the start and you don't have to reverse engineer it too and realize "oh, there is already a variable to configure this". :smile: 

Also, the chips HMC5883 and HMC5883L only differ very slightly (high speed I2C) and it should not matter which one you use to refer the original topic of this issue.

I made a pull request for a slightly updated documentation for the UBLOX Neo-7 on how to set it up correctly.

Many thanks to @westlicht for an evening full of debugging and explanations!

@xqoperator We probably do not have the same problem. But since mine is fixed, I hope somebody with a deeper knowledge can help you out! You have to reopen a new issue with your problem since mine is fixed. (At least you can just copy/paste your text :smile:) Good luck!

closed 2015-08-07 by oxivanisher

xqoperator, 2015-08-07:
just found spare sensor board - this works as expected.

PR #1193 (2016-06-06): Align_mag addition for UBLOX Neo-7 in the docs
I added a align_mag addition for UBLOX Neo-7 in the docs.
This resolves #1123