cleanflight/cleanflight #1124: Make roll/pitch/yaw rates a profile variable

MallyAzure, 2015-07-14:
(Unfortunately I don't have any coding experience)
I noticed the rates are a global variable, and would prefere it to be on a profile to profile basis, mainly because the rates on Harakiri is very different to the other pid controllers, but also for the ability to change to say a profile with a more calmer response without going in and changing the rates manually.

Thanks!

lichtl, 2015-07-14:
On 1.9 all rates and switches are already part of the profile

MallyAzure, 2015-07-14:
Hmm I think I might have run across a bug of some sort, the rates seems to follow from profile 1 to 2, but 3 is unaffected, made a lil video here: https://dl.dropboxusercontent.com/u/34685402/20150714_191050.mp4

MallyAzure, 2015-07-14:
Upon closer inspection, profile 1 and 2 seem to have the same "rateprofile 0" in the cli dump while profile 3 has "rateprofile 2",

lichtl, 2015-07-14:
Hmm what board and version are you using? I've just flashed a flip32 I have laying around to 1.9. stable and set different rates for all 3 profiles (10%, 20%, 30%). Either in the dump or in the cleanflight GUI while switching between the profiles all rates are shown as expected.

https://dl.dropboxusercontent.com/u/9237831/cleanflight_1.9.naze.profile-dump.txt (see rateprofile 0 to rateprofile 2 section)

MallyAzure, 2015-07-14:
Also using a flip32,
Did a full reflash, and copied the full CLI dump (minus rateprofile) from profile 1 and 2 over, the problem seemed to persist. Did a reflash again and copied only the pids over from profile 2 and the problem is gone.
Here is the dump of the 2 profiles (minus rateprofile): https://dl.dropboxusercontent.com/u/34685402/profile-dump.txt

lichtl, 2015-07-14:
did you copy and past the entire profile? try to paste it in segments or even easier change them via the GUI (change to profile + save then modify rates and save ones more. that should work (I also sometimes notice if i paste too much not all entries are being processed.

MallyAzure, 2015-07-14:
Yes, that might been what caused the issue, copying only in segments like you said solved it. Thanks for the assistance anyways!

closed 2015-07-14 by MallyAzure