cleanflight/cleanflight #1125: Implemented support for FlySky i-Bus telemetry

smiling-Jack, 2015-07-15:
Hello
is something planned or in progress for FlySky i-Bus telemetry ?

SteveAmor, 2015-07-15:
#983 

Please can you close this issue if the above PR answers your question.

smiling-Jack, 2015-07-15:
i think ist not the same issue or ?

I mean send telemetry data back to the radio. 
e.g. altitude, ....

Fheilmann, 2015-08-20:
I'd be willing to have a more detailed look at this, however there is absolutely no information on the web on what the sensor part of the i-bus protocol looks like.

dixiak, 2015-09-08:
I've monitored the connection between an FS-IA6b receiver and an FS-CVT01 add-on voltage sensor. 
The data packets do not look too complicated. However, data transfer is bi-directional on a single wire. Could this be handled on supported unmodified hardware or would we need to add a diode between Rx and Tx? Transfer rate appears to be 115200 so software serial is not likely to work too well.

dixiak, 2015-09-08:
Most of the time, here's what the two devices say to each other repeatedly.
Receiver says 
04 A1 (5A FF)
Sensor says 
06 A1 xx xx (58 FF)

The bit in brackets is the packet checksum, calculated the same way a regular iBus i.e. 0xffff minus each individual byte value, stored little-endian.

04 A1 is probably a request for a voltage reading. 
06 A1 probably means "here's a voltage reading". The next two bytes are a signed 16-bit integer representing hundreths of a volt. i.e. F401 would be 5.00V (stored little-endian)

dixiak, 2015-09-08:
There's also a setup sequence, which I have no idea how to interpret.
Receiver repeatedly says
04 81 (7A FF)
04 81 (7A FF)
approx. every 7ms

When you plug in a sensor it will reply with
04 91 (6A FF)

Receiver:
06 91 03 02 (63 FF)

Sensor:
04 82 (79 FF)

Receiver:
04 A1 (5A FF)
Sensor: 
06 A1 xx xx (58 FF)
At this point we have fallen into the pattern described in the previous post, repeated approximately every 6ms.

Fheilmann, 2015-09-08:
The two-way communication makes this difficult indeed. Maybe someone "official" can chime in about whether we can do 1-wire bidirectional communication on a (soft)-serial port?

ledvinap, 2015-09-08:
FrSky smartport is bidirectional. It works both on softserial (with heavy changes from master and untested lately) and F3 UART (flying that) in my 'local' branch.
Smartport needs inverted mode (only F3 UART supports this), I-bus should be even easier ...

Fheilmann, 2015-09-08:
So it may be possible to use the FrSky telemetry port on, say, a NAZE32 to accomplish this? That would be awesome!

ledvinap, 2015-09-09:
@Fheilmann : There is some SW support for bidirectional communication in CF (on UART), using it for Ibus telemetry should be quite easy

Fheilmann, 2015-09-09:
@ledvinap Sorry for my stupid questions, I am still trying to get an overview of cleanflight and how everything works.

What kind of port is the special "1-wire frsky telemetry" port that you can find on most naze32 boards?

ledvinap, 2015-09-09:
IMO it's just transistor-inverted TX signal with pull-up ... Works with FrSky telemetry port (single-direction, just inversion is needed)

twokaff, 2015-09-13:
I second the request of iBus compatible telemetry support.  LiPo voltage without an additional sensor would be nice.

hydra, 2015-12-14:
flysky RX support now merged into CF 1.12.0 development versions.

CraigJPerry, 2016-01-22:
Hey Guys,

I've started working on this tonight. I've proven it works on an F3 discovery (checkout the ludicrous voltage report in this photo - i was just counting up in a loop to test it):

[image]

Here's a trace of a higher voltage from the logic analyser:

[image]

The protocol works like this:

1. The rx queries: 0x04, 0x81, 0x7A, 0xFF
2. I respond: 0x04, 0x81, 0x7A, 0xFF
3. The rx queries: 0x04, 0x91, 0x6A, 0xFF
4. I respond 0x06 0x91 0x00 0x02 0x66 0xFF
5. At this point the rx seems to be looking for more sensors but i've not explored that yet
6. A few more ms passes then the rx queries 0x04, 0xA1, 0x5A, 0xFF
7. I respond with a value and a checksum, probably easiest expressed in the code:

    static void sendFlySkyVoltage(uint16_t voltage) {
      uint8_t first = voltage;
      uint8_t second = voltage >> 8;
      serialWrite(smartPortSerialPort, 0x06);
      serialWrite(smartPortSerialPort, 0xA1);
      serialWrite(smartPortSerialPort, first);
      serialWrite(smartPortSerialPort, second);

      uint16_t checksum = ~(0x06 + 0xA1 + first + second);
      serialWrite(smartPortSerialPort, checksum & 0xFF);
      serialWrite(smartPortSerialPort, checksum >> 8);
}

Now that i've proven it works, i'll put together a proper patch with test coverage. I just wanted to share that some progress has been made. I've been checking this issue for about a month now - since i started flying small quads, hoping someone would submit a fix!

/me -> bedtime, work in 7 hours...

CraigJPerry, 2016-01-22:
Here's the temperature gauge protocol kind of working. It displays some data, but not what i was expecting! Anyway, seems it's pretty hot in here so time to logoff for sure!

[image]

cklam2, 2016-01-22:
Over here Dave Borthwick posted some pic code that emulates a sensor for the i6. Maybe can check what he already figured out:
http://www.rcgroups.com/forums/showpost.php?p=33365816&postcount=152

CraigJPerry, 2016-01-22:
Nice one! He's documented the fields and the bit I was struggling with - the timing parameters.

I'll get some more time on this over the weekend. It's not a complicated protocol, the code to support it is pretty easy but this is my first stm32 project so I'm sure I'll run into some issues! Just learning about the stm32 support for bi directional half duplex serial has been cool. I thought we'd have to connect the tx / RX pins with a resistor then plug that into the ia6b but there's no need, the stm32 supports this natively. Cool little micro, wayyyy more guts than the arduino's I've been playing with til now!

CraigJPerry, 2016-01-25:
Hi Guys, can some folks please test PR #1723

Once you've flashed the hex and restored your settings, go to the config tab in the configurator and enable telemetry.

Set the function bitmask to 256 for your chosen serial port. Baud rate is ignored because ibus always runs at 115200. E.g. i run on serial port 1 so i enter this at the cli:

`serial 1 256 115200 57600 115200 115200`

Lastly, to connect it, just use the tx pin of your uart to connect to the ibus sens port. The rx remains unconnected.

Please feedback values you get for "Tem3".

Cheers

cklam2, 2016-01-26:
Amazing, very excited. 
Trying to make it work but I'm stuck at the last step. Flashed your hex file and in Cleanflight config (ver 1.2.0) trying to set the serial but somehow it doesn't save it. 

```
# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# serial 0 257 115200 57600 115200 115200

# save
Saving
Rebooting
```
When checking again it shows this
```
# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
```
And also lost my serial on UART2. Any tip on this?

CraigJPerry, 2016-01-26:
Aha IBUS sits and blabs away all the time so if you port share it with MSP like that i think it will cause problems so I didn't enable this sharing for that reason. However i don't know it's a problem for sure i just guessed so we should test it to see, it could be fine to share these. MSP may be robust enough to handle the collisions.

I'll upload a new hex tomorrow night when I'm next on a computer.

Thanks for trying so far! Appreciate it.

CraigJPerry, 2016-01-26:
If you want to build a new hex yourself before then I also updated the dev docs so that might help some folks too. Also if you have something other than a base. Although, there are much easier ways to build a hex than installing eclipse but this might interest some all the same, help get more contributors to the project!

https://github.com/cleanflight/cleanflight/blob/72e1fab1749ec2a7a14791a27b8fd0f2cb0b6fe9/docs/development/Building%20in%20Eclipse.md

Fheilmann, 2016-01-26:
I would recommend you get in touch with the people in this rcgroups thread:

http://www.rcgroups.com/forums/showthread.php?t=2486545

They have cracked the Firmware of the FlySky i6 transmitter wide open and I am pretty sure they can help you get meaningful telemetry data onscreen of this TX (they can rename the temperature sensor to something else for example)

cklam2, 2016-01-26:
I tried that too, built a custom hex file but I see the same thing happening. I actually don't do anything with MSP but when using 256 instead I see the same result. Also when giving up the iBUS rx on UART2 (just want to see telemetry working) and running "serial 1 256 ..." it also resets after I save it. So not sure what I'm doing wrong. Will try again tonight.

CraigJPerry, 2016-01-26:
Hey @cklam2 

Yeah the serial 0 port being reconfigured to default settings was expected with the last hex, i'd deliberately not marked ibus as shareable. Here's a hex with it shareable:

[cleanflight_NAZE.zip](https://github.com/cleanflight/cleanflight/files/105750/cleanflight_NAZE.zip)

Can you test a) that it works but also b) that the configurator works as normal with this in place. I.e. no random disconnects etc. Make sure there's power to the rx when testing this (i.e. battery plugged in).

Cheers!

CraigJPerry, 2016-01-26:
@Fheilmann yeah i definitely would, even just to say thanks for some of the stuff those guys have made because it's been awesome for me. However, rcg won't let me login. My original account there (i used to fly rc planes and heli's years ago) got deleted / disabled / disappeared.

My old posts are still on the forums but the account doesn't exist. Super weird. So i created a new account and they blocked me as a spammer! I mailed them for help and i submitted a message via their online form a couple of months ago but never heard anything back.

cklam2, 2016-01-28:
Hi @CraigJPerry. Sorry, been busy on other things lately. 

I flashed your hex and this is much better, doesn't overwrite the serial 1 when setting the serial 0 for MSP+IBUS telemetry. In the configurator I don't see any random disconnects either.

However, no telemetry info on my TX (flashed with 10ch 1.5 firmware). In the Sensors list I see IntV, Tx.V and Err only. Just to make sure that I followed the right steps:
- [ ] UART1 TX pin (= Naze32 rev5 FrSky [telemetry TX](http://www.processio.com/wp-content/uploads/2015/02/naze32GPS1.png)) connected to FS-iA6B SENS signal pin
- [ ] In configurator, CLI, entered serial 0 257 115200 57600 115200 115200 and then save (verified with serial that setting is indeed saved)
- [ ] In configurator, Configuration, enabled Telemetry + Save and reboot.

Also tried to use UART2 instead by connecting SENS to pin3 and configured serial port 1 but still no luck. In the TX under RX Setup I see a "i-BUS Setup" and it's set to channel CH1, no idea what it means, I left that as-is.

Btw, saw that you successfully posted on RCg with your new account. Hope that Dave or povlhp can provide you valuable information on the iBUS protocol.

CraigJPerry, 2016-01-29:
Thanks for testing this @cklam2

Can you send me a dump of your CLI please. You've done everything right, it behaves like the pin isn't connected but you're on the correct pin on both the naze and the RX.

cklam2, 2016-01-31:
Hi @CraigJPerry, Here is the dump of the cli settings ([clidump.txt](https://github.com/cleanflight/cleanflight/files/111291/clidump.txt))

I restored the connection to the UART1 and I connected it as shown in the picture. It's the black wire, bit hard to see. I hope that someone else can try your firmware cause I'm not trusting my iBUS sensor port and configuration anymore. 

[image]

CraigJPerry, 2016-02-01:
Your wiring looks good. The only real difference between our setups is you're using IBUS serial RX and I'm stuck on PPM for now. I have a naze32 in the post to me just now so I'll be able to exactly duplicate your setup and test soon!

cklam2, 2016-02-01:
Well, well @CraigJPerry, I finally got some telemetry data on my RX display :)

Tried everything I could think of but your last post gave me the clue: I only get the i-BUS Telemetry working when configuring PPM instead of i-BUS RX and connecting SENS to pin 3 of UART2. I guess that the UART1 is limited in its use. Or something else need to be modified in the code in order to make it work on this particular port. 

So for now I need to choose between telemetry and 10 channels.

[image]

CraigJPerry, 2016-02-02:
Nice one @cklam2 ! I suspect this is might be a hardware (naze specific) thing for uart1 since i run both ibus serial 10ch AND ibus telemetry together on an STM32F3DISCOVERY. I'll know more once my naze arrives.

ExtV on sensor 2 shows as 0v for you in that screenshot. I just want to make sure you have vbat monitoring disabled? Or does the battery status show correctly in the cli when you do a "status" command?

Thanks for testing this btw! There's one other person about to test on an fs-i10 then i'll crack on with augmenting the data we can send back.

All the telemetry implementations in cleanflight duplicate their metric gathering logic so I'll re-factor out a common "telemetry source". As well as eliminating that duplication of code it'll make it much easier to write tests if i can inject different scenarios from the tests.

cklam2, 2016-02-02:
The displayed value is correct cause I did not yet connect the main battery to the vbat connector :) Right now using an external battery monitoring device but I'll switch over to the naze once I have a buzzer.

goldjacket, 2016-02-20:
From what I have read about the naze32 you can't connect with USB and use UART 1 pin 1 rev 6 board at the same time. You should try this again in ibus mode not ppm to see if this works while not connected to USB at the same time. Use your Lipo only for power.

goldjacket, 2016-02-20:
Pin 1 is the TX pin uart1 just to clarify

AtomoxS3R, 2016-04-02:
Hi @CraigJPerry, probe the firmware with the telemetry. And it worked the first time. I have a Skyline32 as flight controller.
No option to upgrade to the latest version of naze32 with telemetry option?
And I wanted to ask a question, do not know much about it, but can be placed in the telemetry option softserial? The only problem is that the port operates at 19200 max.
regards
Sorry for the translation, my language is Spanish and would translate through Google translator.

CraigJPerry, 2016-04-02:
Hey @AtomoxS3R thanks for testing and i'm glad it worked for you!

>> No option to upgrade to the latest version of naze32 with telemetry option?

I've just changed PC but as soon as i get everything re-installed, i'll make a new build.

>> can be placed in the telemetry option softserial?

No, it can't work on softserial unfortunately. ibus is fixed speed at 115200.

All the best,

Craig

hopoffbaby, 2016-04-05:
Hi,

First post here. I have just flashed cleanflight onto a SPRacingF3 and set up Serial RX using IBUS. I have the TGY i10 tx and TGY ia10 receiver. I'd like to test your telemetry code too if possible. Could you please point me in the right direction?

Also, do you know if the ia10 is capable of 14 channels? It is supplied with a TGY-AEV01 IBUS module which according to the manual "will allow the user to add 4 more channels if the channels of the receiver is not sufficient". It also suggests that up to 4 modules can be daisy chained to provide an additional 16 channels.

Thanks,
Neil.

Kayouu, 2016-04-15:
Were you able to get ibus talking to cleanflight with the ia10 rx? If so how did you connect it? @hopoffbaby

Arise, 2016-04-15:
Do I have to use the second IBUS port on receiver (the IN port)?

hopoffbaby, 2016-04-15:
@Kayouu I connected it by using the iBUS OUT port and connecting that to the boards input as described in the documentation. It required me to cut the cable on one of my rx sensor modules and solder to the SPRacingF3 harness. I then configured SerialRX and set it to iBUS in the configuration.

@CraigJPerry I assume that for the telemetry I will need to connect to the iBUS IN in the same way?

Kayouu, 2016-04-16:
@hopoffbaby and by doing this you were able to control your quad using serialrx \ ibus? Do you enable sbus in the RX settings on the i10? And are you running CF firmware or betaflight? Thanks!

Arise, 2016-04-16:
@Kayouu already explained this to you here: http://www.rcgroups.com/forums/showpost.php?p=34530160&postcount=2120

inode-, 2016-04-26:
I'm tryning the NAZE32 hex that you provide, I got the telemetry on the UART2, all values are 0 like into the @cklam2 photo, but I will check that later. The main problem for me is that currently when you set the serial port with that mask you remove all settings from the port, so it's not possible to use the same port for the RX. It's possible to setup that in the correct way to have the IBUS send and IBUS receive on the same port? Ex. PIN3 connected to SENS, PIN4 connected to SERVO?

In your opinion will be possible to receive GPS cords via ibus telemetry?

Arise, 2016-04-26:
Yes, I would +1 the above reply. Need telemetry on some shared port or on same port as IBus.

CraigJPerry, 2016-04-28:
Ibus telemetry is bi directional. Although it connects with only 1 wire at
the receiver, it's using 2ports at the stm32 - both tx and RX pins are
used. The stm32 internally connects tx and RX pins if you set bidi mode on
the serial port. So although there's only 1 wire connection to the stn32,
internally it's duplexing tx and RX outputs to the one pin. There is no
spare RX pin on the serial port, it is being used by ibus telemetry even if
there is only 1 wire.

CraigJPerry, 2016-04-28:
@Arise already shareable with msp on the serial port

Arise, 2016-04-28:
Ok, thanks for some clarifications.

However on naze32 the MSP is set on UART1 and the receiver is connected via the UART2/pin4 (single wire).

Is it possible to move receiver on UART1 pin? So I can share the receiver (rx part and telemetry) with the MSP port?

inode-, 2016-04-28:
@CraigJPerry do you think that the two ibus are sharable on the same port?

Another thing that I saw, others telemetry code send the data only when the drone is armed, in this way doesn't create problem when the drone is connected to cleanflight. Maybe this can be used as a workaround if it cause problem on the UART1 port.

CraigJPerry, 2016-04-29:
I don't think both ibus's can share 1 port. You'd need some changes on the
software side but I think that could be done. Electrically though on the
hardware side of things - I'm really not sure how you would connect the
extra transmitting node without risking damage. You couldn't just use a
resistor it'd skew the signal. It'd need to be some transistor based
switching you put in the middle.

AtomoxS3R, 2016-05-07:
Hello @CraigJPerry, applying some new telemetry to the most current version of cleanflight for Naze. There is no way it is applied standard for future versions of the .hex
And it could be applied telemetry code in the code of 10ch to unify the tx and rx in ibus.
I have no idea of programming, sorry for this last question if it is a little strange.
We depend on you, we do not know the programming. Thank you.

Arise, 2016-05-09:
@CraigJPerry  I took a look over your commits and I can see:
```
 * Due to the limited telemetry data types possible with ibus, we
 * simply send everything which can be represented. Currently this
 * is voltage and temperature.

...

static const uint8_t SENSOR_ADDRESS_TYPE_LOOKUP[] = {
    IBUS_SENSOR_TYPE_INTERNAL_VOLTAGE,  // Address 0, not usable since it is reserved for internal voltage
    IBUS_SENSOR_TYPE_EXTERNAL_VOLTAGE,  // Address 1, VBAT
    IBUS_SENSOR_TYPE_TEMPERATURE,       // Address 2, Gyro Temp
    IBUS_SENSOR_TYPE_RPM                // Address 3, Throttle command
};

```
What happens if we add more sensors? Could there be hidden sensors that could be processed?

Is it possible to send data like the Altitude or GPS coordinates? Maybe send them unethically disguised as RPM/Voltage/Temperature? Even if the TX will report them as wrong, we could know what their meaning is...

hopoffbaby, 2016-05-09:
Hi,

Does anyone know how I can compile this to work with SPRacingF3 or point me to some documentation?

Thanks,
Neil.

CraigJPerry, 2016-05-21:
Hi @Arise, you can freely add more sensors. Addresses 0-15 are supported by the protocol. There are no hidden sensors.

It's possible to send other data types, i asked if the guys doing the FSi6 firmware patches over on RCGroups would mind labelling a few new datatypes like GPS co-ords, height, distance from home etc. or even just a generic "data" type.

I don't have time to chase that up right now, i just occasionally reply on Github notifications. When i get back to rc quads i expect to have more time for this.

Cheers,

unitware, 2016-07-11:
Hi, I've made some more work based on @CraigJPerry :s pull request. baiscally rebased it on the release v1.13 and added unit tests. Now I'm working on supporting daisychained sensors between the RX and the flightcontroller. However this is a bit if blind flying for me as I do not have the sensors myself. I have made the assumption that every unit in the chain only resends the adresses not used further down the line, which might or might not be valid.

Could anyone help me out and test this code on a Naze32? (other targets can be built on request of course) 

set serial 512 1 115200 57600 115200 115200

If it does not work then it would be very helpful to get a serial dump from the signal line at various points in the daisy chain.

(Do not fly this one, I have not. And make a backup as always):
https://www.dropbox.com/s/d8r04bb36318hi2/cleanflight_NAZE_daisy_chainable.hex?dl=0

Arise, 2016-07-11:
Almost forgot about iBUS extension, I can give it a try tomorrow. What exactly should I test? You want to daisy chain real sensors or to just see if the TX can see the voltage?

unitware, 2016-07-11:
Real sensors in daisy chain with the fc at the end.

Arise, 2016-07-13:
Can you explain how you wired (connected) the sensors and the FC?
Usually for naze32 the receiver connects to the PIN4 but how do you daisy chains the rest of the sensors?

Source code so I can try your builds?

unitware, 2016-07-13:
Hi, the code is here:
https://github.com/unitware/cleanflight/tree/ibus-telemetry

I do not possess any sensors, hence the request for a test run.
My understanding is that the signal normally going to pin4 on FC should be routed via the pyshical sensors - ie daisy chaining. so the RX in one end, connected to one side of the sensor, then the other side of the sensor to next sensor, and then finally to the FC pin 4.

unitware, 2016-07-30:
@Arise did you have any luck testing this?

[59 more replies]

closed 2017-02-25 by hydra