cleanflight/cleanflight #1126: Battery Voltage Scaling of PID Roll/Pitch/Yaw output.

bdoiron74, 2015-07-15:
Ever notice how your optimal PID values change between full and flat batteries? This is due to the battery voltage drop essentially causing a system gain change - about 20% between full and flat. 

Here's a code snippet of what I did - and I really like it:

        // scale to have constant gain through battery sag
        float batt_scale = feature(FEATURE_VBAT) ? constrainf(batteryWarningVoltage / vbat, 0.75f, 1.0f) : 1.0f;

        // Find roll/pitch/yaw desired output
        for (i = 0; i < motorCount; i++) {
            rpy[i] =
                axisPID[PITCH] * currentMixer[i].pitch +
                axisPID[ROLL] * currentMixer[i].roll +
                -mixerConfig->yaw_motor_direction * axisPID[YAW] * currentMixer[i].yaw;

            rpy[i] = rpy[i] * batt_scale;
        }

        for (i = 0; i < motorCount; i++) {
            motor[i] = rpy[i] + rcCommand[THROTTLE] * currentMixer[i].throttle;
        }

Note that you have to separate out the throttle portion and scale the RPY portion first, then add in the throttle. I'm doing other things to evenly scale the RPY and adjust throttle to fit with feedback to the PID on saturation, but I won't cloud this thread with that.

jaapp, 2015-07-15:
That is an interesting idea.

Though, vbat is more influenced by current draw than by drainage. For as for as I can see, this will negate TPA and might even create a feedback loop (more current draw by motor -> more sag -> even more current draw -> even more sag) etc.

Do you have any logs with and without this change to quantify the improvement you noticed?

joshuabardwell, 2015-07-15:
At the very least, you must apply some smoothing to VBAT. Blackbox shows that it takes some pretty extreme and high-frequency jumps during flight.

ledvinap, 2015-07-15:
Just guessing, but as long as battery can supply enough power, there won't be any instability related to battery resistance. Maximum output power is when load resistance is equivalent to source resistance, and this limits theoretical maximum output power to (U/2)^2/R. Another guess - this limit can't be reached without greatly exceeding battery C rating (unless wire resistance is unrealistically high).

ESC is probably better place to do this compensation. Is there any experience with RPM-controlled ESC with internal PID?

joshuabardwell, 2015-07-15:
@ledvinap BLHeli's closed-loop mode will do exactly what you're saying, but I've never tested it. My gut feeling was that it was better to let CF's PID loop do the work, as opposed to having one loop inside another. But it never occurred to me that closed-loop mode might eliminate the effect of battery voltage on throttle response.

ledvinap, 2015-07-15:
@joshuabardwell : My understanding is that inner loop is OK as long as it's time constant (bandwidth?) is short enough compared to outer loop.

ESC PID may be run at higher frequency and has almost delay-free propeller speed feedback. This is definitely worth testing (but the results may be poor) ...

joshuabardwell, 2015-07-15:
Ugh. Just the thought of another PID loop to tune gives me cold sweats.

ledvinap, 2015-07-15:
It should not be that bad - you can tune on bench and tuning should be quite simple - motor with propeller is quite simple system. Also it is done only once for ESC/motor/prop combination ...

Displaying speed response (measuring actual RPM) may be the harder part ..

joshuabardwell, 2015-07-15:
The ESC knows the RPM. Is it reported to the FC?

ledvinap, 2015-07-15:
Standard servo protocol is one-way. It would be nice to have bidirectional communication, but keeping this the protocol simple is probably better option ..

joshuabardwell, 2015-07-15:
Some ESC's support I2C, don't they? Maybe that is the wave of the future. But for now, I suppose we will have to use a tachometer.

bdoiron74, 2015-07-15:
You don't want closed loop rpm control (static thrust vs dynamic). Maybe closed loop power control? Back in my APM days I measured a few motor/prop combinations for static thrust vs control input (% voltage), and they were surprisingly linear due to the work being done by the motor - if the control input were rpm, it would certainly not be (more like quadratic), and the motor wouldn't naturally spin up as the air speed increased with motion. 

I have a note in my code to put time constant filtering on the battery voltage, instead of sample bank averaging. I don't think the current sample bank is long enough (false low battery beeps on aggressive maneuvers, even without this), which is probably why you see lots of high frequency jumps in the log. Although, I haven't noticed any ill effects yet - maybe someone skilled at log analysis can check it out? 

I haven't analyzed any logs (acro naze32 board), but I can say that my spare quad (virtually identical to the first) felt very twangy with a fresh battery and smoothed out as the battery got spent - then I realized I hooked the voltage up to the OSD and which meant I wasn't doing this compensation on that one. 

Again back in my APM days, the main reason I did this was to maintain a constant hover point across battery droop, and it worked great (throttle x == voltage y to the motor) - and as a side effect I noticed the quad responded more consistently throughout the battery. For mini quads, I deliberately took the throttle out of the equation because I want the thrust punch of a new battery. 

How do you mark this 'issue' as a feature/discussion instead of an issue?

borisbstyle, 2015-07-16:
This is not a bad idea if done properly. It should not cause jitter in the pids as VBAT is really jumpy.

@bdoiron74 
Make a PR so we can discuss.

bdoiron74, 2015-07-16:
Do you have any logs showing the jumpiness of VBAT? I would argue that if VBAT is simply sagging due to current draw, this would do exactly what it's supposed to do. If it's noise introduced by improper grounding of the VBAT measurement, you'd want to filter that out. However, too much lag in the measurement and I think it could potentially oscillate or react out of phase. 

I ran into a bad grounding case on my Indy250 because the motor return current was lifting the ground seen by the Naze board - the fix was to take the VBAT +/- right off the batter terminal. Again, I haven't looked at a log, but I'm just going by the numbers I'd see in my OSD.

I'll see what I can do about a PR - it's kind of wrapped up in my Roll/Pitch/Yaw scaling mod, and I'm pretty green in terms of proper git etiquette.

joshuabardwell, 2015-07-16:
This is typical of my results. The battery is a 3S 1500 mAh Turnigy nano-tech 35-70C.

[image]

You can see that it's not just sag. It's lots of high-frequency variance.

This is a log from somebody online, using a 5S battery. It's much less variable, but I don't know if that's the higher cell count or something else.

[image]

borisbstyle, 2015-07-16:
Yeah definately some good low delay filtering should be applied to vbat.
I believe its even averaging at the moment.

I agree that a multirotor with half empty battery can handle higher P, but
you would have to detect it accurately and sample over a longer period to
be certain.

bdoiron74, 2015-07-16:
That _is_ a lot of noise in the first case and likely a grounding issue (you really do want the vbat +/- coming straight off the battery). Do you know what the scale/range is on that graph? Looking at blackbox.c, it looks like blackbox logs the raw adc values (vbatLatestADC), and not the smoothed vbat signal. That would be completely unusable for an OSD or warning beep check. You'd have to apply the FC's 8 sample wide moving average filter to see how much of that noise is removed. 

The 2nd is much more reasonable, especially if the averaging was applied.

joshuabardwell, 2015-07-16:
The first example is from my quad. Vbat is coming straight off the PDB. However I don't have a separate ground wire run to the Vbat terminal, so possibly that is affecting the results.

bdoiron74, 2015-07-16:
I have a CC3D running cleanflight 1.9.0 (with a few tweaks), so I thought I'd try enabling the blackbox to have a look at mine...

feature BLACKBOX
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = 0

I tried both blackbox_device=0 and blackbox_device=1, nothing. I'm using switch arming, could that be the problem?

bdoiron74, 2015-07-17:
So I tried the latest master (1.10) - and something is very very wrong... doesn't seem like throttle is getting passed through correctly, wobbles, drops out of the sky if you angle it over.... Maybe I messed something up porting my changes into that revision.

So I reverted back to my original modded 1.9 branch (which flies great!) - and the blackbox worked this time. Just waiting for it to offload.

Anyone know why offload is excruciatingly slow? Port utilization 8%... ugh.

joshuabardwell, 2015-07-17:
Offload from BB is slow because it's happening at 115,200 baud. Get an OpenLog with a SD card and never look back.

bdoiron74, 2015-07-17:
Hmm - scaled and filtered appropriately, I'd say the VBAT noise isn't bad at all on my setup. The upper lines are the a couple of throttle punches. The pink is what was logged and the blue is the vbat as seen by the flight controller. 

[image]

Here's a zoomed out image (I started with a half flat battery). I'd say the battery sag is much more significant than the noise...
[image]

I see why they're working on rc command filtering and PID term filtering...
[image]

bdoiron74, 2015-07-21:
I cleaned this up and posted a pull here: https://github.com/cleanflight/cleanflight/pull/1152.

closed 2015-07-21 by bdoiron74