cleanflight/cleanflight #1129: Horizon flip

nivaldin, 2015-07-15:
Hi, after a susseful flashing of cc3d with cleanflight 1.9.0, I flew well, but when I change the flight mode to horizon (tested in the software if it is changing) the quad won't flip, it stop flipping at 90º. 
The max throw of aileron is reaching 1902mu.

What I can check now?

Thank you

stronnag, 2015-07-15:
This is a support question, not an issue. Please consider asking such questions in a support environment such as RC Groups or #cleanflight, rather than diverting developer time. For the record, my cc3d machines flip just fine in horizon mode, so it is not a software issue (bug).

nivaldin, 2015-07-15:
Ok, sorry for that...

SteveAmor, 2015-07-17:
@nivaldin please can you close this issue.

closed 2015-07-17 by nivaldin