cleanflight/cleanflight #1130: Roll/Pitch/Yaw Scaling

bdoiron74, 2015-07-15:
This is another feature development I've been toying with, but it may not be for everyone. I'm just throwing it out there for discussion. It currently doesn't handle 3D and you'll see my battery sag compensation down there in the code too...

In the stock code, when the roll/pitch/yaw/throttle gets calculated and saturates a motor, all motors get reduced equally - and if any motor motor goes below minimum, then it gets clipped off. Say you had your throttle just above minimum and tried to pull a quick roll - since two motors can't go any lower, you essentially get half the power (half the PID gain) to do the roll as you would with your throttle up. Or if your current pilot inputs -> PID asked for more than 100% difference between the high and low motors, the low motors would get clipped, breaking the symmetry of the motor balance by unevenly reducing the gain. 

What I've been doing is determining the requested motor spread and calculating a factor by which the roll/pitch/yaw inputs get scaled to be less than 100% motor difference, then adjusting the throttle down - or up - so that it fits. This scaling factor is also fed back to the PID to adjust the pilot input and error generation so it doesn't keep saturating. I scale the roll/pitch/yaw matrix output evenly so that coordinated requests stay coordinated - just slower using as much motor as available.

What I like is that the quad responds the same to stick banging with the throttle up or down - I wrote it in response to a high rate death tumble that didn't end well because I tried to upright the copter near the ground before punching the throttle... What you won't like is that once you take the throttle off minimum, the effective throttle will rise higher that you expected due to yaw drift or the quad not being level (if level enabled). I haven't been bothered by this, but when I arm and lift the throttle, I don't dilly dally before getting into the air to avoid yaw drift accumulation. 

Here's my mixTable, with scaling enabled by defining RPY_SCALING:

    void mixTable(void)
    {
        uint32_t i;
    
        if (motorCount >= 4 && mixerConfig->yaw_jump_prevention_limit < YAW_JUMP_PREVENTION_LIMIT_HIGH) {
            // prevent "yaw jump" during yaw correction
            axisPID[YAW] = constrain(axisPID[YAW], -mixerConfig->yaw_jump_prevention_limit - ABS(rcCommand[YAW]), mixerConfig->yaw_jump_prevention_limit + ABS(rcCommand[YAW]));
        }
    
        // motors for non-servo mixes
        for (i = 0; i < motorCount; i++) {
    #ifndef RPY_SCALING
            motor[i] =
                rcCommand[THROTTLE] * currentMixer[i].throttle +
                axisPID[PITCH] * currentMixer[i].pitch +
                axisPID[ROLL] * currentMixer[i].roll +
                -mixerConfig->yaw_motor_direction * axisPID[YAW] * currentMixer[i].yaw;
    #else
            // TODO - deal with FEATURE_3D
            // TODO - Consider the fact that a yaw burst might lift the copter by boosting throttle - see how this feels. 
    
            int16_t rpy[MAX_SUPPORTED_MOTORS];
            int16_t rpy_max = 0; // assumption: symetrical about zero.
            int16_t rpy_min = 0;
    
            // scale to have constant gain through battery sag
            float batt_scale = feature(FEATURE_VBAT) ? constrainf(batteryWarningVoltage / vbat, 0.75f, 1.0f) : 1.0f;
    
            // Find roll/pitch/yaw desired output
            for (i = 0; i < motorCount; i++) {
                rpy[i] =
                    axisPID[PITCH] * currentMixer[i].pitch +
                    axisPID[ROLL] * currentMixer[i].roll +
                    -mixerConfig->yaw_motor_direction * axisPID[YAW] * currentMixer[i].yaw;
    
                rpy[i] = rpy[i] * batt_scale;
    
                if (rpy[i] > rpy_max) rpy_max = rpy[i];
                if (rpy[i] < rpy_min) rpy_min = rpy[i];
            }
    
            // Scale roll/pitch/yaw uniformly to fit within throttle range
            int16_t rpy_range = rpy_max - rpy_min;
            int16_t th_range = escAndServoConfig->maxthrottle - escAndServoConfig->minthrottle;
            int16_t th_min, th_max;
    
            if (rpy_range > th_range)
            {
                for (i = 0; i < motorCount; i++) {
                    rpy[i] = (rpy[i] * th_range) / rpy_range;
                }
                th_min = th_max = escAndServoConfig->minthrottle + (th_range / 2);
            }
            else
            {
                th_min = escAndServoConfig->minthrottle + (rpy_range / 2);
                th_max = escAndServoConfig->maxthrottle - (rpy_range / 2);
            }
    
            // Now add in the desired throttle, but keep in a range that doesn't clip adjusted 
            // roll/pitch/yaw. This could move throttle down, but also up for those low throttle flips. 
            //
            // TODO: handle the case when motors don't all get the same throttle factor... 
            //       too lazy to sort out the math right now.         
            for (i = 0; i < motorCount; i++) {
                motor[i] = rpy[i] + constrainf(rcCommand[THROTTLE] * currentMixer[i].throttle, th_min, th_max);
            }
    
            // adjust feedback to scale PID error inputs to our limitations. 
            rpy_limiting = constrainf(((rpy_limiting * th_range) / rpy_range), 0.1f, 1.0f); 
    #endif
        }
    
        if (ARMING_FLAG(ARMED)) {
    
            bool isFailsafeActive = failsafeIsActive();
    
    #ifndef RPY_SCALING
            // Find the maximum motor output.
            int16_t maxMotor = motor[0];
            for (i = 1; i < motorCount; i++) {
                // If one motor is above the maxthrottle threshold, we reduce the value
                // of all motors by the amount of overshoot.  That way, only one motor
                // is at max and the relative power of each motor is preserved.
                if (motor[i] > maxMotor) {
                    maxMotor = motor[i];
                }
            }
    
            int16_t maxThrottleDifference = 0;
            if (maxMotor > escAndServoConfig->maxthrottle) {
                maxThrottleDifference = maxMotor - escAndServoConfig->maxthrottle;
            }
    #endif
    
            for (i = 0; i < motorCount; i++) {
    #ifndef RPY_SCALING
                // this is a way to still have good gyro corrections if at least one motor reaches its max.
                motor[i] -= maxThrottleDifference;
    #endif
                if (feature(FEATURE_3D)) {
                    if ((rcData[THROTTLE]) > rxConfig->midrc) {
                        motor[i] = constrain(motor[i], flight3DConfig->deadband3d_high, escAndServoConfig->maxthrottle);
                    } else {
                        motor[i] = constrain(motor[i], escAndServoConfig->mincommand, flight3DConfig->deadband3d_low);
                    }
                } else {
                    if (isFailsafeActive) {
                        motor[i] = constrain(motor[i], escAndServoConfig->mincommand, escAndServoConfig->maxthrottle);
                    } else {
                        // If we're at minimum throttle and FEATURE_MOTOR_STOP enabled,
                        // do not spin the motors.
                        motor[i] = constrain(motor[i], escAndServoConfig->minthrottle, escAndServoConfig->maxthrottle);
                        if ((rcData[THROTTLE]) < rxConfig->mincheck) {
                            if (feature(FEATURE_MOTOR_STOP)) {
                                motor[i] = escAndServoConfig->mincommand;
    #ifdef RPY_SCALING
                                // TODO - clear the PID accumulators when throttle low
    #endif
                            }
                            else if (mixerConfig->pid_at_min_throttle == 0) {
                                motor[i] = escAndServoConfig->minthrottle;
                            }
                        }
                    }
                }
            }
        } else {
            for (i = 0; i < motorCount; i++) {
                motor[i] = motor_disarmed[i];
            }
        }
    ....

Then in the PID, I do things like this:

    #ifdef RPY_SCALING // rate request
        rcCommandRate = rcCommandRate * rpy_limiting;
    #endif

    ...

    #ifdef RPY_SCALING // angle request
        error = error * rpy_limiting;
    #endif

borisbstyle, 2015-07-16:
Although I am still thinking if this is really needed actually, but it's worth testing. It's better to make a cli command for enable/disable to be able to test or perhaps use in the future.

Anyway I don't think motor saturating on low or high throttle is an issue with rolls and flips as the multirotor can flip/roll much faster than we use? I would say for roll/flip you usually use like 20-30% throttle and this is never a problem.
Also simillar approach is done to dynP8,dynI8 and dynD8 where stick is being scaled into PID's for some pid controllers. I am not sure if you need to multiply error and rccommand if you would work out the table for dynP,I and D?

bdoiron74, 2015-07-16:
I notice a huge difference in responsiveness at low throttle, say punching up and cutting throttle to do some quick tricks at the apex. Also, if you're deliberately dropping out of the sky at very low throttle, your level gain would normally be cut in half and wobble... but those are less to do with the scaling, and more to do with the throttle lift to avoid clipping off the low side motors.

I think the scaling comes more into play with combination maneuvers, especially if you have a responsively tuned yaw - and I like my yaw snappy. Do the math on what gets passed through the motor matrix when you pin the stick for max pitch + max roll at standstill with an rc_rate of 2.5...

borisbstyle, 2015-07-16:
I actually don't have the problem that I would get wobble with drops. Maybe my machines are overpowered to notice that :D. Perhaps if my rotation rate would get faster it might be noticed perhaps?
I can go literally 0 throttle freefall without any wobble or negative impact. (using PID1 and PID2)

I know other pid controllers have this dyn PID scaling going on which I hate.

maca55, 2015-08-21:
This is a great idea that I would like to test this an optional feature. I came to the same sort of conclusion recently after analysing my blackbox logs and realising that my motors are reaching max/min quite often. This happens It is often at the start/end of a flip/roll, and definitely an issue during maximum stick yaw spins. As Boris says, it may not be necessary for everyone's quad, but I feel like it would be a big improvement for many. 

So this will takes care of multiple motors maxing out, as well as the multiple motors reaching min. I think it will also help auto optimise yaw rate. For those who set the yaw rate higher than the quad/esc/motor/prop combo can physically deliver, I guess it will also help prevent two motors at max and two at min and the resultant loss of stability in this state?

How does it affect performance when throttle is at MIN? Currently the I terms are zeroed at MIN_THROTTLE and your quad can drift significantly on a long decent.

RCMaca, 2015-08-21:
An example of how unstable my quad gets if I push yaw rate too high. 2 Motors go to min and the roll pitch axis unstabilise. I realise I should drop my yaw rate, but it would be cleaver if the code simply recognized the limitations of my setup and prevented this loss of control from occurring:
[image]

bdoiron74, 2015-08-22:
As is, this scaling might cause stuttering if your yaw tuning is creating
large spikes (basically squashing the pitch/roll reqs to nearly zero).
Reducing the yaw jump prevention limit can help control this. I'd like to
look into implementing a softer limit someday.

Where I found it useful for me was on the low side, which wasn't the
scaling per se, but the fact that it allowed the average throttle to rise
to avoid low side clipping.

bdoiron74, 2015-12-13:
So @borisbstyle - AIR MODE looks a lot like a cut and paste of this code eh? With the necessary safety improvements and such of course. Getting good feedback?

borisbstyle, 2015-12-13:
The mixer part is a proof of concept from someone else indeed, but the AIR
mode itself is a configurable mode on switch what activates pids from 0
throttle + the different mixer

bdoiron74, 2015-12-13:
Myself, I prefer having throttle 0 stop the motors, but I think it's great this code leaked out in to the betaflight.

borisbstyle, 2015-12-13:
oh i didnt realize it was you....lol!

That happens when answering github messages in email.

First of all I have to excuse myself that I discarded your code in the
past. But I guess things change and we learn new things :).

The code is excellent. At first I didnt seem to be bothered, but when I
started on my Idle up/ air mode feature to improve inverted moves etc I
started to realize it cant be done with the default mixer code.
Your mixer concept really seems to make a huge difference with 0 throttle!

Just one question about one of your remarks. Why did you put a comment
about unequal throttle factors to the motors?

bdoiron74, 2015-12-13:
I've had mixers in the past (APM days) that ran different throttle ranges for a PVC-VTail quad (That I really should rebuild.. very fun flight dynamics!). The code here assumes all motors are getting equal throttle, which typically is the normal case so not that big a deal.

RCMaca, 2015-12-13:
@bdoiron74 Yes, Boris' recent Mixer rewrite has made a huge difference, just like you said it would 5 months ago. My flying ability, as well as my two flying buddies, has instantly improved 100%. It's a real  shame your code got initially dismissed by so many.

@borisbstyle Thanks for daring to rewrite the Mixer in Betaflight. I really thought @bdoiron74 was on to something amazing, but no one else seemed to care back in August. I use AIR MODE as default and we don't bother with the old crappy mixer anymore. I have been calling it the BORIS MIXER REWRITE rather than AIR MODE, but perhaps it be renamed to the BDOIRON MIXER, just like we have LUX FLOAT, to give credit to original code??

borisbstyle, 2015-12-14:
@RCMaca  It's 2 separate things. Air mode is intended to have full PID interaction at 0 throttle on better way than pid_at_min_throttle and does more thaghn mixer, but that one is needed to overcome zeroing of the motors. But as I worked on this I also realized it flies much better throughout entire range.
Perhaps this mixer could become default in the future if it all turns out

RCMaca, 2015-12-14:
@borisbstyle @bdoiron74 well I can say with the new mixer code as default (AIR MODE ALWAYS ON) things are turning out really well for me so far. My flying mates and I are doing acro maneuvers that previously were only in our dreams!  After 10 packs, I've only had one strange unintended spin. I'm analyzing the BB logs now and will feed back the cause if I can pin point it.

Read my write up on AIR MODE on your rcgroups thread - you haven't replied to it by the way - I described the Mixer code changes as 'bloody genius'. But since it is a copy and paste of the @bdoiron74 's mixer code then I feel a sorry that he doesn't get any credit. Is there a way we can give him the proper acknowledgment and credit he deserves?

ProDrone, 2015-12-14:
@RCMaca but this idea is an open issue and is not dismissed (like you say on rcgroups), it is a code idea/proposal that was just sitting here and needed testing (though @bdoiron74 already knew it worked very well :grinning: ). Maybe it was picked up sooner if it had been a pull request instead of a code proposal in an issue?

The way i see it, is that @borisbstyle exposed the code for testing to the biggest beta testgroup there is, which is a good thing imo. It is by the way not the first piece of code-art from bdoiron74 that went into betaflight. His quaternion based IMU code also made a big difference, it made the shorter loop times possible.

It is a nice thought to make a list of credits.

RCMaca, 2015-12-14:
@ProDrone Thank you for all your hard work making sure no one else gets hurt by CF or BF! You've done an outstanding job and I feel much safer around my quads now.  Yes, Issue is still open, but the idea was, for all intents and purposes 'dismissed', or as Boris puts it 'discarded'.

The mixer code has gone from just a few lines to over 100 lines. This is a significant amount of original logic and thought to be copy and pasted with out any form of attribution. I have no idea how @bdoiron74 feels about it, but if it was my code that was used as "the best discovery so far in betaflight.", then I would like to be credited for it, rather than just be referred to as 'somebody'. 

But this isn't my code, so I'm going to stay out of this now and let @bdoiron74 speak up if he is in anyway unhappy. I'll go back to flying with this brilliant new code, and continue to help out @borisbstyle  by testing and finding the bugs in his latest releases :grinning:

borisbstyle, 2015-12-14:
@RCMaca I hope you realize what you are saying/claiming?
First of all. If you read my description of the mixer idea you would probably realize that I already stated that the idea was from someone else and that I used it for totally different purpose than it was indended by bdoiron. I only didnt meantion his name as I didn't even know it was him as the code I had locally was an old branch. There was no PR about this at all, cause otherwise I would have found the person behind it!
And I want to tell that it's definately not copy/paste as you say it is as I really wanted to use the mixer with 0 throttle situatluons mainly where it wasnt even designed for as there was an Iterm windup issue which I resolved and I added many safety features to be able to use it with 0 throttle. Don't meantion the amount of props and time for testing/coding/debugging I spent on this project. 
And third thing you are saying that we should have been flying this 5 months if I had not rejected it is again completaly wrong total bullshit and nonsense. There was never, but never any PR made by bdoiron to add this code! Only PR's can get into the code who are accepted by Hydra and not me or someone else. So you do realize that this code would have never been discovered otherwise?

Can you next time do some more research before you claim things all around the places which are totally not true!

bdoiron74, 2015-12-15:
Nope, I didn't make a PR with this since there didn't seem to be any interest. Been flying it myself along with the battery scaling mod ever since. I fly with an ARM switch, MOTOR_STOP enabled and full braking in the ESCs because I really do want the motors to stop as fast as they can when I zero the throttle. I do handle handle clearing the PID error when the throttle hits zero, but I didn't post it here.

Some credit for the mixer code would be nice, but @borisbstyle has obviously put more effort into testing the limits and adjusting the surrounding code for mass consumption.

The battery scaling mod (https://github.com/cleanflight/cleanflight/pull/1152) might help people with tuning issues since it deals with the fact that the battery voltage directly affects the control loop gain by about 22% from full to empty. IIRC, cleanflight moved on while that PR was outstanding and changed the way the battery is measured or filtered and I haven't bothered to try and port it forward.

borisbstyle, 2015-12-15:
I will put your name into the code with one of the next vommits and in the
description @bdoiron. You do deserve that.  Normally if there is a PR for
something I merge that in betaflight and the name of the coder gets into
betaflight commit history, but in this case there isnt. If you want feel
free to contribute to betaflight. That is a better playground than
cleanflight and things can be discussed in larger public. Once we see
public wants something we can add it to cleanflight. The feedback there
seems better as more folks are able to test.

As I said I spent a lot of time with 0 throttle. After I while I also
realized that the mixer also flies better. The air mode has become my
standard mode for now :).
There are still some optimalisation I want to do, but need to get some time
for it.
I want to separate the feedback to the pid controller by each axis. At the
moment if there is a one bad axis it will take away too much error on other
axis and destabilize those as well. Need to test with that to find the best
way.

Yes I tested your battery scaling too in the past. The idea is great, but
the battery read as it is is not reliable enough to use it for that. If the
battery reads were faster this would work great.
How is youd experience so.far?

borisbstyle, 2015-12-15:
@bdoiron74
Can you do a new PR for your voltage scaling so we can work on that
together.

Also feel free to controbute to the current code wherever you see an
improvement.

bdoiron74, 2015-12-15:
Thanks boris - I was pretty surprised to find my mixer changes at the core of this Air Mode everyone was raving about.

The feedback was intended to keep things scaled evenly so the intended rotational movement is maintained - but I don't imagine it likes crashes or yaw spikes very much. Softening the limiting through prediction might help, but some sort of crash detection might also be needed?

As for the voltage scaling, I'll take a look at what betaflight is using for voltage measurement. You can see the way I used it in the pull request to cleanflight I mentioned above (https://github.com/cleanflight/cleanflight/pull/1152). I upped the sampling rate and put a time constant filter on it and, for me, it has been working great. My tune and hover point feels constant across the whole battery (3S) and I imagine anyone flying 4S would notice it even more. 

We just entered the rain/snow season here in Halifax NS, so I won't get many opportunities to actually test things myself for while :(

borisbstyle, 2015-12-15:
@bdoiron74 
Yes it became a key part of the code to prevent motors going flat on low throttle.

I know that feedback should keep parts evenly, but I did see cases where one axis eats so much from authority that 90% reduction removes all error on some axis. I think that also integer truncating in PID1 is responsible for that as well on this particular case.

I already have some users what can reproduce issues on very high rates. I was also trying to figure out some kind of "algorythm" for crashing or spiking and wanted to limit the error of that particular axis so the maxumum PIDsum gets limited. There should be some theoretical maximum in normal operation.....like some kind of derivative for PIDsum.

My fear with battery scaling is introduced jitter. But I guess we need to test with more users and that's where betaflight is for

Feel free to propose your changes into a PR for both battery scaling and potential prediction algorythms, which I could merge after some testing so more users can provide feedback.
https://github.com/borisbstyle/betaflight/pulls

borisbstyle, 2015-12-16:
@bdoiron74 
I added your name in the comment
https://github.com/borisbstyle/betaflight/commit/83e932d3a33398fd9f7436c7501f0b398f56c9b2#diff-95e70d63cf1e155a276b2ad2a8e7a947R764

I also changed the logic when mixer goes into saturation. I found some issues with your method causing infinite rolls when rates high in acro and when 1 of the motors would show cogging.

bdoiron74, 2015-12-16:
Thanks for the comment. 

Hmm - cogging is a bad thing indicating min_throttle is too low right? That'll definitely throw everything off.

At saturation, your change looks like it retains the rpy mix scaling to the full throttle range (at least one motor at max and one at min), but then opens up the allowable throttle range. I'm not quite sure why this helps... if throttle is above 50%, it'll be reduced to 50% just as before the change; if below 50%, it'll cause the rpy mix to get clipped off on the low side upsetting the yaw balance. 

In the code I didn't post, I used the rpy scaling factor to also reduce the commanded **input** (AngleRate) to the PIDs - basically slowing down the pilots request to something achievable. Also, the output of each PID axis should really be constrained such that axisPID[X] * currentMixer[i].X doesn't exceed 1/2 the throttle range. You'll never get any more than that out of the motors, so no one axis should ever ask for more than that (this could be per axis limiting). If set up that way and all axes wanted full motor power each, the rpm mix would be reduced by 1/4 and none would get completely crushed.

borisbstyle, 2015-12-16:
@bdoiron74 
In this case it was indeed ESC min throttle deviating from others, but that doesnt matter it revealed this where the error would grow faster than the limit allows it. There were some users who reported this to me before I reproduced it.
I think that limiting of error is not the right approach. With my current approach it only limits the upper throttle on a simillar way like the old mixer. When 1 motor would reach maximum the rest will be scaled to it and it wont get higher than that. On that way there would still be stabilisation on maximum throttle and full authority and response on lower throttle. It feels damn locked in now and I was not able to upset it anymore.

I tried many approaches and also limiting the controller input, but that felt weird. With higher rates you would literally feel rolls and flips stalling a bit. 
I think any way where reduction of error would be applied is prone to extreme edge cases where a lot of error is generated like crashes etc. Think about spazzing out. I think my approach to keep the limitation within the mixer only is the best.

bdoiron74, 2015-12-16:
Your upper throttle (>50%) side is fine, yes. The scaled rpy mix will force the throttle down to exactly 50% retaining the new mixer's performance. On the low throttle side, if you're rpy mix saturates the motors, then you'll be clipping the lowest motor(s) - immediately changing the control vector. Basically, I don't think you need the line 788 change (https://github.com/borisbstyle/betaflight/commit/83e932d3a33398fd9f7436c7501f0b398f56c9b2#diff-95e70d63cf1e155a276b2ad2a8e7a947L788), unless your intent was to be able to pull the throttle to zero and limit the fastest motor to 50% when you crash. I'd blackbox the rollPitchYawMixRange variable to see how big the rpy mix request is getting before it is scaled.

borisbstyle, 2015-12-17:
> borisbstyle@83e932d#diff-95e70d63cf1e155a276b2ad2a8e7a947L788

Yes exactly! My idea and findings are that on lower throttle reaching the
max of motors is really exceptional. It only occurs shortly during begin or
end of some hard flips and rolls where clipping of the lower motors is even
desirable. Only other occassions where motors would start clipping in the
low range is during crashes.
If you can try this code. You will now feel the max authority on 0 throttle
manouvres. You obviously need to disable motor stop this time ;p

bdoiron74, 2015-12-17:
Great for crashes I guess - but I don't see why it would be desirable at the end of hard flips. It'd go something like this at 0 throttle:

 - You mash the stick in the opposite direction of the current motion
 - When the rpy mix grows until just before saturation, you get exactly what you want: 100% motor spread with at least 1 motor at nearly max and at least 1 at min. The is achieved by the mixer lifting the throttle to 50% to center the rpy mix in the available motor space.  
 - The rpy mix then just touches saturation and this code suddenly removes the throttle lift - throttle returns to the commanded point of 0% and the rpy mix gets chopped in half with the motor spread reduced to 50%. So when you need it most, you lose the 'authority' in a very discontinuous way. This will keep you in saturation longer. 
 - Then, as the copter catches up (using only 50% of the available motor spread) and the rpy mix shrinks to just below saturation, the mixer will now jump the throttle back up to 50% giving back the 100% motor spread to the rpy mix, discontinuously.

I discontinuous reduction in the capability of the quad just when it needs it most is not what I would want....

borisbstyle, 2015-12-17:
@bdoiron74 
In my tests that approach was unstable and erratic. When no error reduction used destabilasation of 1 of the 3 axis would occur when saturation gets big enough.
For example the end of hard flips would result in some roll and I mean really fast flips. With error reduction or/and commanded rate reduction it gets better, but than still it doesnt cover all of the scenarios.
With too much reduction spazzing and infinite rolls/flips possible and with value too high you still get the chance to cause destabilisation of other axis.
So at the end it would be a copter specific reduction percentage and people would have to tune it.
On the way I did it now I achived my goal where I have the maximum authority with low trottle in inverted drops....low throttle flips....rolls etc and I still maintain the stability on high throttle range on the same way like the old mixer does.
I don't really care about the dynamic situations where you would erratically move the stick left/right like you say, which is anyway usually not resulting in situations or not a long one anyway. There would be some softening, but yet that makes it more natural.

bdoiron74, 2015-12-17:
Forget about the error reduction, keep what you have in the pid. I stand by my reasoning that the line 788 change in the mixer is not a good one. It counters what you are trying to do and increases the saturation if triggered at low throttle.

borisbstyle, 2015-12-17:
Your line 788 would actually just set min and max to 50%. You do realize
that the few lines of code above that in the if statement dont do anything?

In my solution dynamic scaling for motors will happen when possible and
when not it will just use what is there available without forcing it. I
dont think destabilisation would be more than with error feedback.

I am going to collect some bb logs.

borisbstyle, 2015-12-17:
@bdoiron74 
I actually just did some flights trying to reproduce scenarios on low throttle where motors would max out and I was not able to find one on this particular and typical 4S mini setup. Even on flips and rolls there is just a short burst of 70% throttle, what seems fine for my rates. This would probably change when someone would have even more ridicoulus fast rates.
I am still with my changes. I don't see any issues. I also introduced some crashes and that scenario works great as well. Every time you crash and tumble over the ground at least one of the motors is maxed out. This is a great crash detection as the other 3  will always zero out.

Even quick and sharp left to right moving (opposite direction) the motors dont even reach the saturation point. Dont underestimate the rotation force it can be generated with low power

bdoiron74, 2015-12-17:
At 3S, you'd be hitting 93%. If you hit it in flight you'd notice. Someone will.

Yes, good for crashes, but presumably you could detect the crash based on the acc/gyro/pid behavior leading up to this condition instead of leaving a lurking glitch in the code... Discontinuities are bad.

borisbstyle, 2015-12-19:
@bdoiron74 
You may be right about 3S and even worse for 2S setups.
The thing is that I also found that your way was causing power limiting as well. 
 Remember my main motivation in all this was to improve 0 throttle behaviour in any possible way. I think my personal goal is reached. Also your implementation would cause much more instability/shakes during some manouvres.

If someone feels like further improving the mixer .... be my guest.

bdoiron74, 2015-12-19:
I think, as-is, it covers a balance between flight and crashes, but I suspect the transition into and out of limiting will be messy in flight if someone hits it. 

I'm hoping I can dig in to this over xmas vacation, but the weather could be a problem... The original intent was to scale all axes evenly so a maneuver maintained to intended 3D direction, but I've seen stuttering from yaw on my setup before I limited the yaw output (one axis dominating). I think I want two layers - a per axis limiting in the pids, so no one axis can dominate, and what I have now. What it amounts to for normal flight is probably acceleration limiting on the commanded input for each axis to avoid asking for more than it can do - which makes crash detection pretty easy. I've done this in the past by passing in a limit for each axis, but I'd rather it was dynamic so it auto adjusts to battery level / cell count.

closed 2017-02-25 by hydra