cleanflight/cleanflight #1132: Feature: Ability to reverse channel direction in cleanflight

11digits, 2015-07-17:
A simple CLI command would be ok for starters, so we can invert the direction of the channels - there are many rx - tx out there than can't do that.

JohnieBraaf, 2015-07-17:
Why would you need it. For RSSI there is already an CLI command to do so: https://github.com/cleanflight/cleanflight/pull/816

11digits, 2015-07-17:
1. I don't have RSSI
2. As stated, there are many TX-RX out there that are not configurable (channel direction)

Arakon, 2015-07-18:
Those "many" wouldn't happen to be 20$ toy transmitters?
Even the most basic RTF transmitters I've seen from many years ago had a channel inversion switch, either in software or a physical switch.

iforce2d, 2015-07-24:
I second this, and would also love to be able to trim the rc values. For example, suppose your pitch channel is reading 1485 with the stick centered - you could just add 15 to trim it to 1500.

I don't know about toy transmitters, but all my TXs so far are arduino based with no fancy features like trims or channel fiddling options. It would be much easier to change things in the Cleanflight config than to recompile and upload my arduino sketch just to trim the channels. OpenPilot config has these settings (and also endpoints) which makes setup easy for us 'toy' users.

borisbstyle, 2015-07-29:
@iforce2d 
FC should not be responsible for correcting the trims from the TX. I know about openpilot having that tx wizzard. Could be a nice feature indeed to add in configurator in rx tab something like calibrate tx.

hydra, 2015-08-20:
latest code already has rxrange, i just tested removing the check for min being lowerthan max and used `rxrange 3 2000 1000` and the throttle channel could be reversed.

hydra, 2015-08-20:
The great thing here is that code was deleted to add this feature :smile:

closed 2015-08-20 by hydra

digitalentity, 2015-08-20:
@hydra it's funny that I've devised exactly the same solution, but didn't have a chance to post a PR :smile:

commit 71872950 (2015-08-20): Allow FC to reverse input channels by using rxrange.

Just reverse the min/max arguments.

Unit tested.

Fixes #1132.