cleanflight/cleanflight #1137: Advice Request: Best way to get flash memory chip id

therippa, 2015-07-18:
Hi all (even though I'm assuming @hydra  will be the best to answer this),

In my efforts to get blackbox data without having to use an external OpenLog (and clog a UART port), I did a lot of research and found a flash memory chip that shares all the same pins/voltage size/specs/characteristics of the chips listed as supported in the blackbox README - you can view it [here](http://www.digikey.com/product-detail/en/S25FL132K0XMFI041/1274-1134-ND/4692909).

I chose this chip because it has the same footprint as the 2mb (even though listed as 16mg) chip on the naze32 full, but has 32mb of storage which will give roughly 32 minutes of data at 1200 looptime (and I've never had a problem that was outside that time window).  The other chips recommended/supported in the docs do not have the same footprint, so reflowing them to the naze32 will not be trivial.

I soldered the chip on, and it didn't work (somewhat expected).  Digging through the cleanflight source code, I see that the cause is that the chip id function cannot identify it.  I'd like to get the hex code for this chip and add it, but am wondering what my options are.  So far, this is what I can think of, and I was wondering if someone has a better option...

1.  Buy a Cortex debugging pad and debugger, but this isn't ideal because I'm not looking to spend money to get the one value I need
2. Modify the cleanflight code to add a cli command that returns the result of the chip id function (most ideal as far as I can see)
3.  Send a flash chip (I ordered extras) to someone who already has the debugger and is willing to solder it on and update the flash_m25p16 file.

Like I said, I'm probably going to go the #2 route, but if anyone has a simpler solution, please let me know.

I think it would be good to add this chip as "officially supported" since it's just an easy swap with the existing 2mb one on the naze (and various other flight controllers) we all use.

Arakon, 2015-07-18:
Size of that chip is 32 mbit, that's 4 mbyte.. that won't even last for 5 minutes of recording at 1600 looptime. Biggest they offer is 64 mbit.
The device IDs it sends are actually described in the datasheet under 7.1.6.

therippa, 2015-07-18:
Welp, my bad.  Even after all these years I still muck up mbyte/mbit.  Thanks.

closed 2015-07-18 by therippa