cleanflight/cleanflight #1138: Beginners problem

fuxen, 2015-07-18:
I am building my first quad and have run into some problem of which I can not find answers on.

The hardware is a stock TX turnigy 9X with stock receiver (9x 8C V2) with RXC 12A ESC, DYS BE1806, and FLIP32+ FC.

My problems are with the throttle I believe. 
1) There is a very narrow window from hoover to full throttle, extremely hard to fly
2) When armed, the motor do not spin up, MOTOR_STOP not ticked
3) When I turn the quad upside down with  no throttle, the motor spins like it wants to level it - no throttle used. So when I had a small crash it was spinning full upside down and destroyed propellers, and an enginge was effected. I had no throttle at this time and switch TX off without failsafe effected throttle.

I can not figure out what is going on?

Best Regards,
fuxen

Zero9r, 2015-07-18:
OK, I have same radio although i upgraded the firmware and added a DMS2/DSMX module. first few things you need to check.

1. Look at the receiver Tab.  see what values each channel has for Min and max throw.  It should be about 1000 min  1500 in the middle and 2000 max.  somewhere around there doesn't have to be perfect.  IF it is lower or higher it can cause some issues.  this is fixed in the radio by adjusting end points.  I can't remember how to on a stock 9x but with ER9x firmware it is easy.

2. Now you calibrated your ESC in the CF GUI. (go to motor tab, check the checkmark and slide all motors to top, then plug in battery and quickly slide all motors to bottom)  If you have don it correctly slowly rise the slider till motors spin (you can use up/down arrow to fine tune it) write down what value the motors start to spin at.  they should be all the same if not keep calibrating.  for example mine are all 1063.

3.  After all that  you go to the Configure tab.  you want to put the minimum Trottle to what you motors spin up on.  Remember I said mine was 1063.  set you mid throttle to what center was (should be left at 1500 if possible)  Max throttle I set mine to what my Max level is in the reciever tab on full throttle.  Min Command leave it at 1000.

That should help you out.  you have to tell the FC what value to start to spin you motors or when you arm it may not as it is sending the default of 1000, which will be too low.  I hope this helps you.

Zero9r, 2015-07-18:
Set you fail safe to 1000 also so motor stop

joshuabardwell, 2015-07-18:
I'm glad that @fuxen got some help here, but is this really a bug that deserves a github issue opened for it?

For the record, the cleanflight rcGroups thread is a better place to ask questions like this.

http://www.rcgroups.com/forums/showthread.php?t=2249574

SteveAmor, 2015-07-18:
@fuxen could you please close this issue.

fuxen, 2015-07-18:
Thanx for the replies, now I have a clue to continue, and so sorry to post on wrong thread :)

closed 2015-07-18 by fuxen