cleanflight/cleanflight #1140: New servo settings (smix) not persistent across all profiles

stronnag, 2015-07-18:
The new tri yaw configuration `smix reverse 5 2 r` appears not to be persistent across profile changes, leading to unexpected violent rotation on take off and potential aircraft damage.

0. Have profile 0 and profile 1 identical other than some trivial PID settings.
1. Power on with profile 0 and reversed yaw (assumed now to be global, from `smix reverse` rather than per profile as previous `servo` stanza). Fly around a bit. 
2. Land, use the stick command to switch to profile 1.
3. Attempt to take off again. Despair as the craft rotates uncontrollably.

It would appear that the new 'global' servo settings are not persistent across profile switches. This is sub-optimal. The new  `smix` setting should be consistent across all profiles, as they are now global.

stronnag, 2015-07-19:
Even if one powers on in profile != 0, one ends up in landing gear destroying rotational hell, alas.

ethomas997, 2015-08-05:
@hydra This issue still exists with the current master as of 8/3/2015 (I just tested it).  Definite bug that needs to be fixed.

--ET

hydra, 2015-08-07:
Agreed, will look into this

@stronnag thanks once again for excellent testing feedback and bug reporting - much appreciated - I owe you a beer at some point.

closed 2015-09-05 by hydra

sppnk, 2015-09-05:
When trying to reverse servos in custom airplane I get "parse error":

smix reverse 4 2 r
smix reverse 4 2 r
Parse error

May be because of PR #1283 Not sure

commit 84ae209c (2015-09-05): Update smix reverse documentation, closes #1140