cleanflight/cleanflight #1141: gimbal_flags no longer recognized

imagebuff, 2015-07-19:
Running Boris CF 1.10.0 Jul 16 2015 00:29:22 it seems the CLI refuses to even recognize gimbal_flags. I have a cam tilt that was making use of gimbal_flags=4 with CF 1.9 (early version).

set gimbal_flags=4 
Invalid name

joshuabardwell, 2015-07-19:
This is not a bug.

https://github.com/cleanflight/cleanflight/blob/783a4c4bfa095e41ba234c5a74ad56739fbb17bd/docs/Migrating%20from%20baseflight.md

stronnag, 2015-07-19:
Yes, well documented change:

commit 7b91524ba2ba23647be98fbff1aa0be142f81794
Author: Dominic Clifton <me@dominicclifton.name>
Date:   Wed Jul 8 16:41:29 2015 +0100

    Replace gimbal_flags with FEATURE_CHANNEL_FORWARDING and gimbal_mode.
    
    Gimbals are only one use of channel forwarding, PWM video switchers and
    PWM buzzers are others.

Hwurzburg, 2015-07-23:
care should be taken to document exactly how the current gimbal_flags=4 functionality could be set up in the release including motor shift and AUX1-4 to PWM5-8 mapping for hard wired installations using gimbal_flags=4 currently, to avoid system tear down....also to avoid a safety issue with someone upgrading without full EEPROM clear & restore, and missing that gimbal-flags=4 is invalid and then arming to find that two of his motors now arm with a midrc power level, throwing the quad across the room!

ethomas997, 2015-07-23:
What I believe is going on with the new CHANNEL_FORWARDING is that it maximizes the number of available servo outputs for any given board, taking into account the model-mix type, SERVO_TILT, and the particular board hardware. This provides a lot of good functionality, but because of all these variables it's hard to simply say how the channels will map out.

It would be great to have a full-fledged custom PWM mapping capability that could map any RC-input to any output -- then we could have "recipes" for reproducing the "gimbal_flags=4" mappings.  But I don't think that functionality exists yet.

Perhaps in the meantime there should be an option to make CHANNEL_FORWARDING do the same mapping as we've had with "gimbal_flags=4"?  Have it do that when "gimbal_mode=4"?

--ET

hydra, 2015-08-07:
yes, the resource allocation issues/prs will cover that @ethomas997 

Closing this

closed 2015-08-07 by hydra