cleanflight/cleanflight #1146: Flip32 test motor fail

madiron, 2015-07-20:
Hi, 

I flash my Flip32 with cleanflight. When go to Test Motor and start test the motors run all at the same speed when I moving the quadcopter.. it's normal?

joshuabardwell, 2015-07-20:
The Motors tab causes a fixed output signal to be sent to the motors. They should all spin at the same speed. It's normal.

madiron, 2015-07-20:
Thank's for response.. :)

in this video https://www.youtube.com/watch?v=J_NvkHAkyOw change spin when quad is moved.. why?

joshuabardwell, 2015-07-20:
Because he's spinning up his motors by arming the quad. You're spinning up the motors using the "test" sliders on the Motors tab.

madiron, 2015-07-20:
OK, Thanks Joshua ;)

SteveAmor, 2015-07-20:
@madiron as this isn't a bug, can you please close this issue.

closed 2015-07-20 by madiron