cleanflight/cleanflight #1147: Upside down orientation acc / gyro misconfigured?

NullVoxPopuli, 2015-07-20:
I have been having issues with the board orientation. In my WarpQuad, I have the board mounted upside down - so the rear of the board is still the rear, and the front is still the front, I just rotated the board 180 on the roll axis. 

The main side effect I see is that the board constantly tries to correct the yaw, but corrects in the wrong direction, and then tries switching yaw correction directions, and it just gets out of control pretty quickly (I always do initial configuration tests without props, and while holding the quad)

The setups I've tried:

I have Roll at 0 degrees, Pitch at 180, and Yaw at 180. But it wiggles due to trying to correct it's Yaw attitude. 

If I have Roll 180, Pitch 0, and Yaw 180, and it does the same wiggle.

If I do Roll 180, 0 Pitch, and 0 Yaw, Same thing.

If I do Roll 0, Pitch 180, and Yaw 0, Same thing.

If I just do Yaw 180, then there is no issue, but the Arming direction is backwards. (Left to arm).
 - this is how I currently fly, and as you can see from this video: https://www.youtube.com/watch?v=2JMkoJxr5QI the roll stick is backwards as well. 

Any ideas?

I've been sure to calibrate the accelerometer each time I change the board orientation.

joshuabardwell, 2015-07-20:
FWIW, accelerometer doesn't matter when you're flying in acro mode, which you probably are if you're flying a WarpQuad. Acro mode is based entirely on the gyro. Mis-calibrated accelerometer can prevent arming if the copter mistakenly interprets that it has exceeded small_angle, but that's about it.

NullVoxPopuli, 2015-07-20:
ah ok. that's good to know. Thanks for sharing the info!

so it's the gyro that's probably misconfigured.

SteveAmor, 2015-07-20:
Strictly speaking, this is a support question, not an issue. Please consider asking such questions in a support environment such as RC Groups or #cleanflight, rather than diverting developer time. 
However, try the following:

set align_acc = 5
set align_gyro = 5
set acc_trim_roll = 256

Please can you close this issue unless you really believe there is a bug. Thank you.

NullVoxPopuli, 2015-07-20:
Yes! sorry about that. I am very familiar with the support vs developers battle. 
I'll try those settings tonight. Also, joining #cleanflight

thenickdude, 2015-07-30:
I think you were changing the `yaw_control_direction `. This reverses your yaw radio channel which is almost never what you want to do. 

It doesn't change the direction that the FC tries to command yaw in order to maintain heading, so if that was the wrong way around then it won't improve it. 

The setting you'd want to set to -1 is `yaw_motor_direction` or `yaw_direction` (in earlier versions).

NullVoxPopuli, 2015-07-30:
I found out that the issue was actually that because my cleanflight board was also combined with the PDB, the motor mixing was backwards.

this is what I ended up doing: http://www.rcgroups.com/forums/showthread.php?t=2461436

    Roll: 180
    Pitch: 0
    Yaw: 0

Custom Motor Mixing:

    mixer custom // use custom mixer
    mmix reset // clear existing, just in case
    
    // we need to flip yaw and roll 
    //
    // mmix syntax:
    //   n THR  ROLL  PITCH YAW
    mmix 0 1.0  1.0,  1.0,  1.0   // (1) back right, now back left
    mmix 1 1.0  1.0, -1.0, -1.0   // (2) front right, now front left
    mmix 2 1.0 -1.0,  1.0, -1.0   // (3) back left, now back right
    mmix 3 1.0 -1.0, -1.0,  1.0   // (4) front left, now front right

closed 2015-07-30 by NullVoxPopuli