cleanflight/cleanflight #1148: quadcopter graphics

MichaelStanley34, 2015-07-20:
The quadcopter graphic that simulates the movement of the control board has turned from a quadcopter to a box with 'front' 'side' etc labels on each side of the box?

JohnOCFII, 2015-07-20:
That is usually due to an installation error of the Cleanflight Configurator software.  The configurator can not find the image file data.  I'd recommend a re-installation of the Cleanflight Configurator.

joshuabardwell, 2015-07-20:
Also, if you are using a custom mixer.

SteveAmor, 2015-07-20:
@MichaelStanley34 if a reinstall solves your problem, please close this issue.  If there is a bug, please raise as an issue under github.com/cleanflight/configurator
Thanks.

MichaelStanley34, 2015-07-20:
Not sure if I understand a custom mixer?

joshuabardwell, 2015-07-20:
https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md#custom-motor-mixing

MichaelStanley34, 2015-07-21:
reinstalled -all okay.

SteveAmor, 2015-07-21:
@MichaelStanley34 that's great news.  Please can you close this issue.

MichaelStanley34, 2015-07-21:
Can't work out how to close issue?

closed 2015-07-23 by thenickdude