cleanflight/cleanflight #1150: Luxfloat real PID's in configurator (MSP)

borisbstyle, 2015-07-21:
The current Luxfloat configuration is limited due to the current 8bit conversion from float in MSP commands.
It would be nice if the real PID's would be shown in configurator. There fore different approach for MSP_PID and MSP_SET_PID is needed for floating point pid controllers.
16bit integer would be needed in order to get the most precision.

borisbstyle, 2015-07-22:
https://github.com/cleanflight/cleanflight/pull/1156

mehmetyldz87, 2015-07-28:
Old method is like that; Am I right?

  GUI           /          Controller                       GUI Limits             /      Controller  Limits
------------              --------------                      ---------------                 ---------------------
Kp     2,5  ------ (x 1)------> 2,5                    Kp ----->[ 0  25]       /          [0  25 ] 

Ki     0,06 ---- ( x 10) -----> 0,6                    Ki ----->[0   0,25]     /          [0 2,5]

Kd     60 -------( / 1000) ---> 0,06                Kd ----->[0   100]      /         [0  0,1]

Kp_Level   5 ------( x 1 ) -----> 5                 Kp_Level---->[0    25]      /         [0   25]

borisbstyle, 2015-07-28:
@mehmetyldz87 
Yes that's the old method I believe.

closed 2017-02-25 by hydra