cleanflight/cleanflight #1154: Cleanflight support the throttle back in the remote control?

yuan0226, 2015-07-22:
Hi
I would like to know whether the code Cleanflight support the throttle back in the remote control, so set high hover? What still needs to be configured? Thank you for your answer.

yuan0226, 2015-07-23:
I was just wondering whether to support the throttle back in Cleanflight remote control, how is it so hard?

digitalentity, 2015-07-23:
What's "throttle back in the remote control" ?

yuan0226, 2015-07-23:
Throttle return the remote control.

yuan0226, 2015-07-23:
油门回中的遥控

yuan0226, 2015-07-23:
The throttle stick remote control can automatically reset to the middle position.

joshuabardwell, 2015-07-23:
You mean spring loaded throttle, like a DJI controller?

yuan0226, 2015-07-23:
Yes, as a novice, will be better adapted to the kind of remote control,cleanflight support that kind of remote control it?

thenickdude, 2015-07-23:
Yes, this is basically what the "BARO" altitude hold flight mode does. It requires a barometer to be present.

yuan0226, 2015-07-23:
What does this mean?It is the only barometer and set BARO mode together with DJI's remote control can achieve the kind of effect it?

joshuabardwell, 2015-07-23:
The throttle can automatically return to center, or not. Cleanflight doesn't care. If the throttle has a spring, then it will automatically return to center. If the throttle doesn't have a spring, you will move the stick to center with your fingers. Cleanflight doesn't care which of those things happens.

yuan0226, 2015-07-24:
Thank you for your help.

closed 2015-07-25 by yuan0226