cleanflight/cleanflight #1162: Serial RX settings are reset on power cycle

canphaz, 2015-07-25:
<b>CLEANFLIGHT VERSION:</b>
v1.90 (seems to also work on v1.81)

<b>QUADCOPTER:</b>
ImmersionRC Vortex

<b>RECEIVER:</b>
Spektrum Satellite

<b>PROBLEM:</b>
Setting up Serial RX on the Vortex is lost after unplugging / replugging power. It seems the defaults are restored because some combination is invalid.
 
 
<b>STEPS TO REPRODUCE:</b>
 
01. SETUP Tab:
 - Reset Settings
 
02. CLI Tab:
 - 'defaults'
 
02. PORTS Tab:
 - UART2:  Serial RX ---> Enabled
 - Save & Reboot
 
04. CONFIGURATION Tab:
 - Receiver Mode -> RX_SERIAL
 - Serial Receiver Provider -> SPEKTRUM2048
 - MOTOR_STOP -> Enabled
 - ONESHOT125 -> Enabled
 - Save & Reboot
 
05. Disconnect
06. Reconnect. 
07. Verify all of the above settings have indeed been saved:
 - RX_SERIAL is still selected
 - SPEKTRUM2048 is still selected
 - MOTOR_STOP is still enabled
 - ONESHOT125 is still enabled
 

<b>As you can see, it seems these settings work fine until the moment i pull out the power. 
After power cycle, these (and a whole bunch more) settings are reset to default values.</b>

08. Cycle Power (remove lipo in case of the Vortex)
 
09. CHECK PORTS Tab:
 - UART Serial RX is now disabled (checkbox unchecked)
 
10. CHECK CONFIGURATION Tab:
 - RX_PPM is now selected instead of Serial RX
 - SPEKTRUM1048 is now selected instead of SPEKTRUM2048
 - MOTOR_STOP is disabled now (checkbox unchecked)
 - ONESHOT125 is disabled now (checkbox unchecked)

canphaz, 2015-07-25:
My thread at ImmersionRC Support Forum: http://fpvlab.com/forums/showthread.php?42177-Vortex-RX-Settings-in-CF-are-reset-on-Power-Cycle&p=718901

thenickdude, 2015-07-26:
Cannot reproduce with the official Cleanflight 1.9.0 build on a naked Naze32 (no Rx).

Can you still reproduce it if your Rx is not connected to the system?

Do ImmersionRC provide a custom Cleanflight build?

canphaz, 2015-07-26:
Hey thenickdude. Correct it is a Vortex, it has a custom board, not the Naze32. I will try to see if i can reproduce when the satellite is not connected. Thanks. 

ImmersionRC uses the regular CF builds for the vortex, however they have an additional software layer that can be setup and upgraded via a windows tool called "ImmersionRC Tools".

The biggest thing i am missing in the setup they expect, is that i fon't have a Fatshark or something similar. So i can't go through the Setup Wizard that is displayed if you are using a fatshark. If i understand correctly, i don't need to run the wizard, i can set its configuration via the Tools app. But i can't verify :-/

SteveAmor, 2015-07-26:
Could be because of this #1002 ?

canphaz, 2015-07-26:
Ok, so i finally figured it out, with help from the cool folks over at fpvlab.com. 
The reset of settings was caused by the "ImmersionRC Tools" software that needed to be configured to use SPEKTRUM2048 instead of CPPM. Since I don't have FPV Goggles, I did not run through that wizard, and in the Tools I left it to be at CPPM. Changing it to the SPEKTRUM2048 fixed that problem. 
Apparently, the Vortex auto adjusts (or resets) cleanflight settings based on how you have gone through the OSD wizard.

closed 2015-07-26 by canphaz