cleanflight/cleanflight #1164: How to save and lock the throttle value

yuan0226, 2015-07-27:
Hi:
I do have a throttle back in the remote control, like DJI kind of remote control, let go will be back in the throttle stick. Flight Control is NAZE32 brush Cleanfligt firmware, I want to make to support this type of remote control flight control, when I let go of the throttle stick back in time, I hope to be able to save and lock the throttle value in the last, at this moment the flight control should holding a flight throttle value. what should I do? This code is needed it? There is no master willing to guide me, thank you.

joshuabardwell, 2015-07-27:
You should modify your transmitter to remove the spring-loaded centering on the throttle stick. This is why most transmitters do not have a spring-loaded throttle.

The spring-loaded throttle on DJI is because it is designed to be flown in altitude-hold mode. In this mode, centered stick means "maintain altitude". In any other flight mode, you are constantly adjusting the throttle up and down as you fly.

Naze32 full with barometer supports altitude hold mode, in which case the throttle will work mostly as you request. However, altitude hold mode on Naze is not very well implemented, and frankly it doesn't hold altitude very well.

yuan0226, 2015-07-27:
Thank you for your answer, I think the new fly hand may be more suitable throttle back in the remote control, the operation will be easier.

closed 2015-07-28 by yuan0226