cleanflight/cleanflight #1166: CC3D Autotune is not active

weud, 2015-07-27:
Hello,

I hate to open new topic but I am pretty sure there is now one asked this question before. I flashed my cc3d board with cleanflight v1.9.0 but there is no autotune mode? I couldn't find any information about it is not supported. I belive answer just in font of me but I cant see, it shouldnt be so hart to find such a basic info.

digitalentity, 2015-07-27:
Autotune is not supported on CC3D when you are using OpenPilot bootloader (when you flash CC3D using OpenPilot's GCS). Consider flashing a HEX-file (see https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20CC3D.md, `Flashing` section)

joshuabardwell, 2015-07-27:
AutoTune has been unofficially deprecated as far as I know, because it doesn't work.

weud, 2015-07-30:
I believe the document should be much more clear, I have not tested it is working or not but with cleanflight bootloader autotune selection has come.  thanks for the help.

hydra, 2015-08-07:
Please submit an update to the documentation as a PR if you think it needs improving.

martinbudden, 2016-03-25:
@hydra , this issue can be closed, since OpenPilot bootloader support for CC3D was removed in v1.11.0

closed 2017-02-25 by hydra