cleanflight/cleanflight #1168: unrecoverable failure of serial connection

oops47, 2015-07-27:
just added new esc w/ bec. board was fine till now

thenickdude, 2015-07-29:
Are you saying that you get this message upon attempting to connect to the board in the Configurator? Ensure that the correct port and baud rate are selected at the top.

Do the LEDs/buzzer light up properly and do their flashing during startup? (i.e. is the board still operational?)

borisbstyle, 2015-07-29:
I am not sure I get this one?

oops47, 2015-07-29:
Was flashing from BF to CF. Figured it out. Thanks anyway

closed 2015-07-29 by oops47

KillerUAV, 2016-12-18:
I have been having this same issue with the last three FC's on cleanflight and Betaflight.. I am getting the unrecoverable serial failure message and I have the fc communicating with the receiver but wont arm. It tries to arm when I reboot after setting up the sbus or serial rx but on other fc's I am getting no communication with the receivers.  I use Futaba and have been having a terrible amount of trouble wih this.  I had a drone working yesterday but it took many tries to get the serial rx setting to stick, it kept deselecting on reboot.  I have reinstalled drivers and tried another computer both with the same issue running win10.  I used to use naze fc's and never had a problem then the new betaflight came out and I have had numerous fc problems with Flip32, Lux, SPracing, an all having issues surrounding receiver communication.   I have tried using a sbus to cppm adapter and tried different receivers too. My latest receiver is a micro S-FHSS and my cleanflight has huge issues trying to save settings relationg to the serial rx or sbus.

joseadame, 2017-02-04:
I have the same problems. I can setup my receiver.