cleanflight/cleanflight #1169: Question about altitude hold the code

yuan0226, 2015-07-28:
Hi
Recent research Cleanflight altitude hold this part of the code, would like to ask some questions.For instance:
139 void updateAltHoldState(void)
140 {
141     // Baro alt hold activate
142     if (!IS_RC_MODE_ACTIVE(BOXBARO)) {
143         DISABLE_FLIGHT_MODE(BARO_MODE);
144         return;
145     }
146 
147     if (!FLIGHT_MODE(BARO_MODE)) {
148         ENABLE_FLIGHT_MODE(BARO_MODE);
149         AltHold = EstAlt;
150         initialThrottleHold = rcData[THROTTLE];
151         errorVelocityI = 0;
152         altHoldThrottleAdjustment = 0;
153     }
154 }
The above code comes from atitudehold.c .
Why errorVelocityl=0? altHoldThrottleAdjustment = 0? And why  initialThrottleHold = rcData[THROTTLE]?
I'm just a rookie.
I hope someone can help me.Thank you.

leothehuman, 2015-07-28:
This is not the code issue. Please, don't use ticketing system to read the code out loud. There's an IRC channel for that. Look it up on the cleanflight.com. If you're trying to start working on the code, looking at how these variables are used later might help you to understand why these are set up this way.

This is the setup code. At the start of the baro mode we set the accumulated error to 0, because we don't have any. We set adjustment to 0, as we don't need to adjust throttle yet and we also remember throttle stick position to compare to later.

SteveAmor, 2015-07-29:
@yuan0226 can you please close this issue

closed 2015-07-29 by yuan0226