cleanflight/cleanflight #1182: No channel 3 input

Nullstr1ng, 2015-08-04:
I just flashed my CC3D board I bought from banggood. 

first of all, all channels are working in GCS, after flashing to cleanflight, the channel 3 (THR) is not working even on channel remapping. 

I am using VFR-II RX.

I am sure that channel 3 is working because going back to OpenPilot, it shows that channel 3 is working. going back to cleanflight, none.

I hope you can fix this

joshuabardwell, 2015-08-04:
PPM or PWM input?

joshuabardwell, 2015-08-04:
This is probably a technical support issue, and not a bug. The IRC channel or the RCGroups Cleanflight thread is a better place to address it.

Nullstr1ng, 2015-08-04:
oh please consider this as invalid instead.

joshuabardwell, 2015-08-04:
You can close the issue yourself if you want to save an admin the trouble.

I'd offer to help you myself, but I don't know much about the CC3D board.

Nullstr1ng, 2015-08-05:
Ok well. I have worked it out now but still without the white wire. 
This is CC3D from Banggood. I'll just make this a reference for everyone instead.

This is how I got it to work:

My receiver is FrSky VFR-II
CC3D from Banggood

VFR-II Channels
CH1 = BLUE
CH2 = YELLOW
CH3 = ORANGE
CH4 = GREEN
CH5 = 
CH6 = BROWN
CH7 = 
CH8 = WHITE + RED/BLACK

Channel Mapping at CleanFlight
AERT1234

Yaw for Arming
Tx 3 way switch is used for Acro, Horizon, and Angle

closed 2015-08-07 by hydra